ifm/ifm3d #1: make check segmentation fault

MattyJKA, 2017-06-28:
When I run make check I get a segmentation fault. Here is the output I get:

~/ifm3d/build$ make check
[ 22%] Built target ifm3d_camera
[ 22%] Built target ifm3d_camera_shared
[ 38%] Built target ifm3d_framegrabber
[ 38%] Built target ifm3d_framegrabber_shared
[ 44%] Built target ifm3d_image
[ 44%] Built target ifm3d_image_shared
[ 50%] Built target gtest
[ 55%] Built target gtest_main
[ 66%] Built target ifm3d-image-tests
[==========] Running 3 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 3 tests from Image
[ RUN      ] Image.CloudMechanics
[       OK ] Image.CloudMechanics (1319 ms)
[ RUN      ] Image.XYZImage
Segmentation fault (core dumped)
make[3]: *** [modules/image/test/CMakeFiles/check_image] Error 139
make[2]: *** [modules/image/test/CMakeFiles/check_image.dir/all] Error 2
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
make: *** [check] Error 2

I'm using Ubuntu 14.04 and an O3X camera. I've already verified that the O3X is running firmware version 0.1.16.

I'm confused by the error path, modules/image/test/CMakeFiles/check_image, when I try to navigate to that directory it doesn't exist. I have path modules/image/test, but there is no CMakeFiles folder inside the test directory.

tpanzarella, 2017-06-28:
Please find the file `libifm3d_image.so` in your build directory. Then paste in the output of:

```
$ ldd libifm3d_image.so

( ... output omitted ... )
```

Unrelated, but, if possible, I'd recommend updating your camera to `0.1.20`. @GarrettPlace should be able to set you up with that.

MattyJKA, 2017-06-28:
Thanks for the advice.

Turns out I had opencv3 library on my computer but I told my cmake to expect opencv2. I decided to  uninstall opencv3 then I installed opencv2. That fixed my problem.

tpanzarella, 2017-06-28:
That is precisely where I was going with this.

closed 2017-06-28 by tpanzarella