ifm/ifm3d #2: There is no simple viewer application like in the libo3d3xx

graugans, 2017-06-29:
During a support call this morning I was not able to find a simple viewer application for the ifm3d like the one for [libo3d3xx](https://github.com/lovepark/libo3d3xx/blob/master/modules/image/src/bin/o3d3xx-viewer.cpp). Are there any plans to include this or is it supposed to have an separate project for this?

tpanzarella, 2017-06-29:
In conversations with GP several months ago, the thought was to just use rviz. A simple viewer could be added to this project. I'll put it on my near term TODO.

graugans, 2017-06-29:
I guess rviz is good, but some customers still do not use ROS. Maybe I'll find some time and an O3X to modify the old viewer...

tpanzarella, 2017-08-22:
Included in 0.3.0

closed 2017-08-22 by tpanzarella