ifm/ifm3d #3: cannot find -lopencv_dep_cudart

graugans, 2017-07-19:
Looks like there is an issue when one has the Nvidia CUDA libraries installed.

```
[...]
Scanning dependencies of target ifm3d_image_shared
[ 50%] Linking CXX shared library libifm3d_image.so
/usr/bin/ld: cannot find -lopencv_dep_cudart
collect2: error: ld returned 1 exit status
modules/image/src/libifm3d_image/CMakeFiles/ifm3d_image_shared.dir/build.make:103: recipe for target 'modules/image/src/libifm3d_image/libifm3d_image.so.0.2.0' failed
make[2]: *** [modules/image/src/libifm3d_image/libifm3d_image.so.0.2.0] Error 1
CMakeFiles/Makefile2:773: recipe for target 'modules/image/src/libifm3d_image/CMakeFiles/ifm3d_image_shared.dir/all' failed
make[1]: *** [modules/image/src/libifm3d_image/CMakeFiles/ifm3d_image_shared.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

graugans, 2017-07-19:
This issue is not ``ifm3d`` related and just for the sake of documentation. As described in this [issue](https://github.com/opencv/opencv/issues/6542) I was able to solve the issue by turning ``CUDA_USE_STATIC_CUDA_RUNTIME`` off.

```
cmake -DCUDA_USE_STATIC_CUDA_RUNTIME=OFF ../
```

closed 2017-07-19 by graugans

tpanzarella, 2017-07-19:
This is good info to have on the record. Thanks!