ifm/ifm3d #5: [Known Issue] XML-RPC issue when requesting unit vectors on O3X

graugans, 2017-07-27:
There is an issue when one tries to request unit vectors from an O3X which is configured to ``TriggerMode`` software trigger mode (``SW``) after a fresh reboot. To reproduce I've prepared a simple example:

```cpp
From e1eaa89116a1a34923d5b8723fcf55d49a30fd69 Mon Sep 17 00:00:00 2001
From: Christian Ege <christian.ege@ifm.com>
Date: Thu, 27 Jul 2017 09:05:00 +0200
Subject: [PATCH] Added a simple example to retrieve the unit vectors

This will grab the unit vectors from the camera if
supported.

---
 modules/examples/CMakeLists.txt      |  7 ++++++
 modules/examples/ex-unit_vectors.cpp | 41 ++++++++++++++++++++++++++++++++++++
 2 files changed, 48 insertions(+)
 create mode 100644 modules/examples/ex-unit_vectors.cpp

diff --git a/modules/examples/CMakeLists.txt b/modules/examples/CMakeLists.txt
index b436b1d..88aae01 100644
--- a/modules/examples/CMakeLists.txt
+++ b/modules/examples/CMakeLists.txt
@@ -67,3 +67,10 @@ target_link_libraries(ex-file_io
                       ${OpenCV_LIBRARIES}
                       )
 
+add_executable(ex-unit_vectors ex-unit_vectors.cpp)
+target_link_libraries(ex-unit_vectors
+                      ifm3d_camera_shared
+                      ${Boost_LIBRARIES}
+                      )
+
+
diff --git a/modules/examples/ex-unit_vectors.cpp b/modules/examples/ex-unit_vectors.cpp
new file mode 100644
index 0000000..471fbc1
--- /dev/null
+++ b/modules/examples/ex-unit_vectors.cpp
@@ -0,0 +1,41 @@
+/*
+ * Copyright (C) 2017 ifm syntron gmbh
+ *
+ * Licensed under the Apache License, Version 2.0 (the "License");
+ * you may not use this file except in compliance with the License.
+ * You may obtain a copy of the License at
+ *
+ * http://www.apache.org/licenses/LICENSE-2.0
+ *
+ * Unless required by applicable law or agreed to in writing, software
+ * distributed under the License is distribted on an "AS IS" BASIS,
+ * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
+ * See the License for the specific language governing permissions and
+ * limitations under the License.
+ */
+
+//
+// ex-unit_vectors.cpp
+//
+// Capture a frame from the camera, and write the data out to files. For
+// exemplary purposes, we will write the amplitdue and radial distance images
+// to PNG files and the point cloud to a PCD file.
+//
+
+#include <iostream>
+#include <memory>
+#include <opencv2/opencv.hpp>
+#include <pcl/io/pcd_io.h>
+#include <pcl/point_cloud.h>
+#include <ifm3d/camera.h>
+#include <ifm3d/fg.h>
+#include <ifm3d/image.h>
+
+int main(int argc, const char **argv)
+{
+  auto cam = ifm3d::Camera::MakeShared();
+
+  auto uv = cam->UnitVectors();
+
+  return 0;
+}
-- 
2.7.4
```

The camera's trigger configuration:
```console
$ ifm3d -ip=192.168.0.69 dump | grep Trigger
        "TriggerMode": "2",
```

graugans, 2017-07-27:
A workaround to this issue is executing a trigger before requesting the ``UnitVectors``

tpanzarella, 2017-07-27:
@graugans Did you want to commit this to the `examples` module? If so, just send the PR and I'll get it merged.

graugans, 2017-07-28:
@tpanzarella  before that I need to cross check with an O3D3xx. I pointed it out cause this error was reported by more than two guys. The idea was to give some advice how to do a workaround.

AFAIK this will stay in firmware releases ``0.1.x``, ``0.2.x``. There is a fix scheduled for firmware version ``1.0.x`` but this is not compatible with the current hardware.

tpanzarella, 2017-07-28:
I would expect similar behavior on the O3D3xx. Since we acquire the unit vectors over PCIC on the O3D, if the camera were to be set in SW trigger mode, it would need to see that trigger to deliver any data over the socket. However, in the case of the O3D, I don't think we would want to sleep, see [this](https://github.com/lovepark/libo3d3xx/issues/83) issue from `libo3d3xx`.

graugans, 2017-07-30:
For the workaround the sleep is needed cause the unit vectors are updated on each new frame. If the current frame is not yet processed you'll get an error. Tomorrow I'll work on the fix for the new firmware. Where like in O3D the vectors are updated on initializing the camera as-well. The O3D test is just for the sake of completeness. BTW this release will introduce completely new modes. Especially one up to 30m mode.

graugans, 2018-01-10:
This issue can be closed due to the fact the release version of O3X has a fw with the fix deployed.

graugans, 2018-05-02:
Due to the fact FW version 1.0.22 is available for download we can close this. For more details see [Issue #56](https://github.com/lovepark/ifm3d/issues/56#issuecomment-385892061)

closed 2018-06-19 by graugans