ifm/ifm3d #6: Missing debian dependencie

aslakjo, 2017-08-11:
Hi

We are trying to setup a minimal environment to use the PCL-python bindings with the o3d3 camera. However when we install the *.deb packages from the 0.1.0 release. We get the following error.

```
 ifm3d-image depends on ros-kinetic-opencv3; however:
  Package ros-kinetic-opencv3 is not installed.

```

Can you help us find the missing package? Or is there other workarounds to accomplish this?

Best regards Aslak Os, Rocketfarm

tpanzarella, 2017-08-11:
Hi Aslak,

I would suggest building the latest release from source. This will also build a deb (if you wish) and will introspect your system with the proper OpenCV dependency. 

The binary debs from 0.1.0 were there simply to accommodate very early adopter companies using the O3X. The expressed dependency that you are lacking is the OpenCV 3 package bundled with ROS Kinetic. If you build from source, that particular deb is not required. 

I hope this helps. 

Tom Panzarella
Founder & CEO
Love Park Robotics, LLC
2025 Washington Ave, Suite 217
Philadelphia, PA 19146 USA

mobile: +1 215 341 1754
tom@loveparkrobotics.com
http://loveparkrobotics.com

aslakjo, 2017-08-11:
Thanks for the quick reply, ill try that out.

closed 2017-08-11 by aslakjo