ifm/ifm3d #7: ifm3d 0.3.0 std::put_time and gcc 4.8

Arag24, 2017-09-06:
Hello gentlemen,

First, I'm a newbie with Linux platform...
Since a couple of weeks, I installed rev 0.2.0 on my Ubuntu 14.04, for running O3D and O3X and it was running fine.

Today, I pulled rev 0.3.0.

Here what I have when writing make command....
`/home/frlegrda/global_src/ifm3d/modules/tools/src/libifm3d_tools/time_app.cpp:67:16: error: ‘put_time’ is not a member of 'std'`

Just briefly having a look on gcc 4.8.X and `put_time` seems to not be implemented in gcc 4.8 (as a newbie I may have wrong for this point...)
Does it mean, we need a minimum gcc version and/or would it be possible to have a fix for 14.04 Ubuntu standard version?

Sincerely Yours

tpanzarella, 2017-09-07:
@Arag24 Thanks for the bug report. A new version has been pushed out (`0.3.1`) that should fix the issue you are having. Please let me know if this works for you.

Arag24, 2017-09-08:
@tpanzarella O3X running like a charm in 0.3.2.
Thanks Tom.

closed 2017-09-08 by tpanzarella