ifm/ifm3d #9: Building ifm3d with gtest

kennyboy55, 2017-10-18:
We are trying to use our IFM camera with ROS and are therefor trying to install this library. We are running into some issues though while trying to build the project. We have installed all the dependencies as far as I know including gtest. When running "make" on the project I get the following output: [output.txt](https://github.com/lovepark/ifm3d/files/1394572/output.txt).

Does anyone know how we might fix this?

Thanks in advance,
Kenneth

tpanzarella, 2017-10-18:
This seems to be an issue in building gtest. Particularly it looks like an issue with gflags -- perhaps it is missing? If you installed gtest via your package manager, it *should* have been expressed as a dependency and installed implicitly for you. However, if you manually installed gtest, you may have run into this issue. The above said, you should be able to install gflags on Ubuntu with:

$ sudo apt-get install libgflags-dev

This also brings up a few questions related to what your software stack looks like: which version of Ubuntu? gcc? ROS? etc. To potentially save some time here though, I refer you to our supported stacks with explicit instructions for installation:

1. [Ubuntu 16.04, ROS Kinetic](https://github.com/lovepark/ifm3d-ros/blob/master/doc/kinetic.md)
2. [Ubuntu 14.04, ROS Indigo](https://github.com/lovepark/ifm3d-ros/blob/master/doc/indigo.md)

We have a high preference toward the 16.04, Kinetic stack.

kennyboy55, 2017-10-18:
I did indeed install it manually before realizing there was a package. 
I am running Ubuntu 16.04 with ROS Kinetic. GCC version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.5).
Libgflags-dev is already installed on my system as well it seems.

I had not seen the installation instructions in the ifm3d-ros repository. I was trying to install all the dependencies manually if I couldn't find a package. I did install ROS before I tried to install this.

tpanzarella, 2017-10-18:
OK. Your stack should be fine. If you follow the instructions for 16.04 + Kinetic that I linked to above, you should be all set -- they assume a fresh install of the operating system and nothing else. If this gets you going, please report back and close out the issue.

ggupta9777, 2017-11-23:
I have been facing the same issue. I instead changed the path from /usr/src/gtest to /($PATH)/googletest in all test directories of all modules. 
Here $PATH is the directory of googletest clone from github. This made sure the build process went through. The make check however has been failing with this log.

> [  3%] Built target ifm3d_image
[ 19%] Built target ifm3d_camera
[ 23%] Built target ifm3d_camera_shared
[ 34%] Built target ifm3d_framegrabber
[ 38%] Built target ifm3d_framegrabber_shared
[ 42%] Built target ifm3d_image_shared
[ 50%] Built target gtest
[ 57%] Built target gtest_main
[ 69%] Built target ifm3d-image-tests
[==========] Running 3 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 3 tests from Image
[ RUN      ] Image.CloudMechanics
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.CloudMechanics (396 ms)
[ RUN      ] Image.XYZImage
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.XYZImage (314 ms)
[ RUN      ] Image.ComputeCartesian
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.ComputeCartesian (419 ms)
[----------] 3 tests from Image (1129 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 1 test case ran. (1129 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 3 tests, listed below:
[  FAILED  ] Image.CloudMechanics
[  FAILED  ] Image.XYZImage
[  FAILED  ] Image.ComputeCartesian

 3 FAILED TESTS
modules/image/test/CMakeFiles/check_image.dir/build.make:57: recipe for target 'modules/image/test/CMakeFiles/check_image' failed
make[3]: *** [modules/image/test/CMakeFiles/check_image] Error 1
CMakeFiles/Makefile2:870: recipe for target 'modules/image/test/CMakeFiles/check_image.dir/all' failed
make[2]: *** [modules/image/test/CMakeFiles/check_image.dir/all] Error 2
CMakeFiles/Makefile2:76: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:184: recipe for target 'check' failed
make: *** [check] Error 2

Now, I am not sure if these fails are a result of my tempering with gtest or it is another problem altogether. And no, I can't ping the sensor.

graugans, 2017-12-19:
@ggupta9777 is your camera connected to your host? Does it may have an IP address different than the *192.168.0.69* based on the latest changes I can run the camera unit test within my [ifm3d-docker](https://github.com/graugans/ifm3d-docker/blob/master/Dockerfile) container for Ubuntu 16.04.

```
fm3d@4b294cd3b8cf:~/build$ make -j12 check_camera
[ 28%] Built target ifm3d_camera
[ 42%] Built target gtest
[ 50%] Built target ifm3d_camera_shared
[ 64%] Built target gtest_main
[100%] Built target ifm3d-camera-tests
[==========] Running 17 tests from 3 test cases.
[----------] Global test environment set-up.
[----------] 14 tests from CameraTest
[ RUN      ] CameraTest.FactoryDefaults
[       OK ] CameraTest.FactoryDefaults (6840 ms)
[ RUN      ] CameraTest.DefaultCredentials
[       OK ] CameraTest.DefaultCredentials (17 ms)
[ RUN      ] CameraTest.ApplicationList
[       OK ] CameraTest.ApplicationList (46 ms)
[ RUN      ] CameraTest.SessionManagement
[       OK ] CameraTest.SessionManagement (762 ms)
[ RUN      ] CameraTest.CopyDeleteApplication
[       OK ] CameraTest.CopyDeleteApplication (14 ms)
[ RUN      ] CameraTest.CopyDeleteExceptions
[       OK ] CameraTest.CopyDeleteExceptions (16 ms)
[ RUN      ] CameraTest.CreateDeleteApplication
[       OK ] CameraTest.CreateDeleteApplication (20 ms)
[ RUN      ] CameraTest.CreateApplicationException
[       OK ] CameraTest.CreateApplicationException (19 ms)
[ RUN      ] CameraTest.ImportExportApplication
[       OK ] CameraTest.ImportExportApplication (1255 ms)
[ RUN      ] CameraTest.ImportExportConfig
[       OK ] CameraTest.ImportExportConfig (1040 ms)
[ RUN      ] CameraTest.ActiveApplication
[       OK ] CameraTest.ActiveApplication (27 ms)
[ RUN      ] CameraTest.ImagerTypes
[       OK ] CameraTest.ImagerTypes (49841 ms)
[ RUN      ] CameraTest.Filters
[       OK ] CameraTest.Filters (12540 ms)
[ RUN      ] CameraTest.JSON
[       OK ] CameraTest.JSON (12067 ms)
[----------] 14 tests from CameraTest (84504 ms total)

[----------] 1 test from Version
[ RUN      ] Version.Version
[       OK ] Version.Version (0 ms)
[----------] 1 test from Version (0 ms total)

[----------] 2 tests from Err
[ RUN      ] Err.LibraryError
[       OK ] Err.LibraryError (0 ms)
[ RUN      ] Err.SystemError
[       OK ] Err.SystemError (0 ms)
[----------] 2 tests from Err (0 ms total)

[----------] Global test environment tear-down
[==========] 17 tests from 3 test cases ran. (84504 ms total)
[  PASSED  ] 17 tests.
[100%] Built target check_camera
```

graugans, 2018-11-16:
@kennyboy55 and @ggupta9777 is this issue still relevant?

theseankelly, 2019-10-09:
Cleaning out old issues; feel free to reactivate if still relevant.

closed 2019-10-09 by theseankelly