ifm/ifm3d #10: Inconsistent Unit Vectors

cartesian-theatrics, 2017-11-08:
Hello,

I'm having an issue where I am occasionally getting inconsistent unit vectors out of the unit_vectors topic. I am able to resubscribe many times to the topic and get the same result out, but if I also restart the node I'll occasionally receive a result that is different than the previous one. This issue has showed up in the field and I've attempted to isolate the error to a small test case. I've attached the ifm3d dump and the test code:

[ifm_paramter_tests.txt](https://github.com/lovepark/ifm3d/files/1455508/ifm_paramter_tests.txt)
(rename to python file. Depends on python 2.x, ros, and nav_msgs)
[ifm3d_dump.txt](https://github.com/lovepark/ifm3d/files/1455504/ifm3d_dump.txt)

OS: Ubuntu 14.04 
ifm3d-version: 0.3.2

Here's the comparison failure output:

AssertionError:
Arrays are not almost equal to 6 decimals
(mismatch 99.9991348283%)
 x: array([[[-0.477243, -0.350969,  0.805651],
        [-0.473859, -0.351833,  0.80727 ],
        [-0.470467, -0.352692,  0.808878],...
 y: array([[[-0.344224, -0.838482,  0.422442],
        [-0.342063, -0.837794,  0.425552],
        [-0.339896, -0.837092,  0.428658],...
[ INFO] [1510176272.632392903]: Running dtors...
[ INFO] [1510176272.632457448]: Initializing camera...

Thanks in advance,
John

cartesian-theatrics, 2017-11-21:
Hello,

Following up on this issue. It's still coming up. I created a cpp test case that that I added to modules/image/test/ifm3d-image-tests.cpp that reproduces the issue for me. 

#define HOSTNAME "192.168.2.69"
#define PORT 80

TEST(Image, IFM)
{
  // Iteratively initializes an IFM frame grabber and
  // pulls the unit vectors. Asserts that successive returns
  // are equivalent.

  auto get_unit_vectors = []() -> cv::Mat {
    auto cam = ifm3d::Camera::MakeShared(HOSTNAME, PORT, "");
    auto im = std::make_shared<ifm3d::ImageBuffer>();
    auto fg = std::make_shared<ifm3d::FrameGrabber>(cam, ifm3d::IMG_UVEC);
    auto timeout = 500;
    EXPECT_TRUE(fg->WaitForFrame(im.get(), timeout));
    cv::Mat uvec = im->UnitVectors();

    // The issue only arrises if we subsequently make a frame graber
    // to start reading the distance image from the buffer.
    fg = std::make_shared<ifm3d::FrameGrabber>(cam, ifm3d::IMG_RDIS);
    if (fg->WaitForFrame(im.get(), timeout)) {
      auto distanceImg = im->DistanceImage();
    }

    return uvec;
  };

  cv::Mat first_uvecs = get_unit_vectors();
  // I'm getting a failure about every 50 iterations or so,
  // although the distribution doesn't seem very uniform.
  for (int i = 0; i < 500; i++) {
    cv::Mat next_uvecs = get_unit_vectors();
    cv::Size s = next_uvecs.size();
    std::cout << "Comparison #" << i << " ... width=" << s.width << ", height=" << s.height <<  "\n" << std::flush;
    bool isEqual = (cv::sum(first_uvecs != next_uvecs) == cv::Scalar(0,0,0,0));
    EXPECT_TRUE(isEqual);
  }
}

cartesian-theatrics, 2017-12-12:
As a followup, the failure rate seems to be highly dependent on the timeout value passed to WaitForFrame. If the timeout is set very low (100 ms), it fails almost immediately and even throws a floating point operation error. If it is set very high (5000 ms), it also fails very frequently. At 500ms, it fails relatively infrequently.

graugans, 2017-12-18:
So we checked the internal handling of the unit vector calculation and indeed found an issue. I will fix this and schedule a new release build.

graugans, 2018-05-02:
Due to the fact FW version 1.0.22 is available for download can we close this issue? For more details see [Issue #56](https://github.com/lovepark/ifm3d/issues/56#issuecomment-385892061)