ifm/ifm3d #12: Publishers Occasionally Hang

cartesian-theatrics, 2017-11-21:
Hello,

We've encountered anther issue in which we are unable to get any data out of the ifm publishers. I created some test code to repeatedly initialize the ifm3d-ros node and receive a distance message. Eventually, it'll hang waiting for the message. The test output looks something like this:

```
...
[ifm_left/ifm_left-41] process has died [pid 890, exit code -9, cmd /opt/shining_software/catkin_ws/devel/lib/ifm3d/ifm3d_node distance:=/ifm_left/depth_with_confidence __name:=ifm_left __log:=/home/brain/.ros/log/744bdbb0-cf05-11e7-9a7f-0010340b6800/ifm_left-ifm_left-41.log].
log file: /home/brain/.ros/log/744bdbb0-cf05-11e7-9a7f-0010340b6800/ifm_left-ifm_left-41*.log
process[ifm_left/ifm_left-42]: started with pid [937]
[ INFO] [1511305225.314623226]: Running dtors...
[ INFO] [1511305225.314707989]: Initializing camera...
[ INFO] [1511305226.327200369]: Initializing framegrabber...
[ INFO] [1511305227.381932637]: Initializing image buffer...
[ INFO] [1511305227.550609292]: Got unit vectors, restarting framegrabber with mask: 15
[ INFO] [1511305227.550649134]: Running dtors...
[ INFO] [1511305227.551026321]: Initializing camera...
[ INFO] [1511305228.563687840]: Initializing framegrabber...
[ INFO] [1511305229.253269551]: Initializing image buffer...
Received depth message. Shape:  0 0
[ifm_left/ifm_left-42] process has died [pid 937, exit code -9, cmd /opt/shining_software/catkin_ws/devel/lib/ifm3d/ifm3d_node distance:=/ifm_left/depth_with_confidence __name:=ifm_left __log:=/home/brain/.ros/log/744bdbb0-cf05-11e7-9a7f-0010340b6800/ifm_left-ifm_left-42.log].
log file: /home/brain/.ros/log/744bdbb0-cf05-11e7-9a7f-0010340b6800/ifm_left-ifm_left-42*.log
process[ifm_left/ifm_left-43]: started with pid [984]
[ INFO] [1511305231.366369760]: Running dtors...
[ INFO] [1511305231.366427715]: Initializing camera...
[ INFO] [1511305232.379364764]: Initializing framegrabber...
[ INFO] [1511305233.429639553]: Initializing image buffer...
[ INFO] [1511305233.586629652]: Got unit vectors, restarting framegrabber with mask: 15
[ INFO] [1511305233.586670046]: Running dtors...
[ INFO] [1511305233.587010254]: Initializing camera...
[ INFO] [1511305234.599095939]: Initializing framegrabber...
[ INFO] [1511305235.301061760]: Initializing image buffer...
Received depth message. Shape:  0 0
[ifm_left/ifm_left-43] process has died [pid 984, exit code -9, cmd /opt/shining_software/catkin_ws/devel/lib/ifm3d/ifm3d_node distance:=/ifm_left/depth_with_confidence __name:=ifm_left __log:=/home/brain/.ros/log/744bdbb0-cf05-11e7-9a7f-0010340b6800/ifm_left-ifm_left-43.log].
log file: /home/brain/.ros/log/744bdbb0-cf05-11e7-9a7f-0010340b6800/ifm_left-ifm_left-43*.log
process[ifm_left/ifm_left-44]: started with pid [1041]
[ INFO] [1511305237.421985366]: Running dtors...
[ INFO] [1511305237.422060482]: Initializing camera...
[ INFO] [1511305238.435424341]: Initializing framegrabber...
[ INFO] [1511305239.444415053]: Initializing image buffer... 
Traceback (most recent call last):
  File "ifm_parameter_tests.py", line 120, in <module>
    receive_depth_image_test(base_topic)
  File "ifm_parameter_tests.py", line 115, in receive_depth_image_test
    msg = next(depth_msgs)
  File "ifm_parameter_tests.py", line 61, in gen_msgs
    yield receive_one(topic, msg_class, 10.0)
  File "ifm_parameter_tests.py", line 54, in receive_one
    assert await_pred(container.deref, timeout=timeout), "Never received msg from %s after %s seconds" % (topic, timeout)
AssertionError: Never received msg from ifm_left/depth_with_confidence after 10.0 seconds
[ifm_left/ifm_left-44] killing on exit
[rospy.internal][WARNING][2017-11-21 23:00:47,406] Unknown error initiating TCP/IP socket to localhost.localdomain:58023 (http://localhost.localdomain:46241/): Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 557, in connect
    self.read_header()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 619, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosgraph/network.py", line 363, in read_ros_handshake_header
    d = sock.recv(buff_size)
error: [Errno 104] Connection reset by peer

[ifm_left/ifm_left-44] escalating to SIGTERM
```

It's pretty infrequent, but it doesn't consistently hang eventually on my machine. The process seems to hang after "Initializing image buffer...". Here's the test code: 

```python
#!/usr/bin/env python
import roslaunch
import rospy
import os
import signal
import time
from sensor_msgs.msg import Image
from multiprocessing import Lock

hostname = "192.168.1.69"
port = 80
base_topic = "ifm_left"

class Atom(object):
    # Don't trust assignment thread safety.
    def __init__(self, v):
        self._lock = Lock()
        self._value = v

    def reset(self, v):
        with self._lock:
            self._value = v

    def deref(self):
        with self._lock:
            return self._value

def kill_nodes(launcher, nodes):
    for node in nodes:
        os.kill(node.pid, signal.SIGKILL)
    time.sleep(2.0)
        
        
def await_pred(pred, timeout):
    start_time = time.time()
    while True:
        if pred():
            return True
        dt = time.time() - start_time
        if dt > timeout:
            return False        

       
def receive_one(topic, msg_class, timeout):
    """
    Creates a subscriber, waits to receive a message from the given topic/msg_class,
    then unregisters the subscriber. Returns the received message.
    """
    container = Atom(None)
    sub = rospy.Subscriber(topic, Image, container.reset)
    assert await_pred(container.deref, timeout=timeout), "Never received msg from %s after %s seconds" % (topic, timeout)
    sub.unregister()
    return container.deref()

def gen_msgs(topic, msg_class):
    while True:
        yield receive_one(topic, msg_class, 10.0)

def launch_ifm(launcher, base_topic):

    rospy.set_param(os.path.join(base_topic, base_topic),
                    {"frame_id_base": base_topic,
                     "ip": hostname,
                     "password": "",
                     "publish_viz_images": False,
                     "schema_mask": 15,
                     "timeout_millis": 500,
                     "timoute_tolerance_secs": 5.0,
                     "xmlrpc_port": port,
                     "assume_sw_triggered": False})

    node = roslaunch.core.Node(
        package="ifm3d",
        node_type="ifm3d_node",
        name=base_topic,
        namespace=base_topic,
        output="screen",
        respawn=False,
        required=False,
        remap_args=[['distance', "/%s/depth_with_confidence" % base_topic]]
    )

    return launcher.launch(node)

def receive_depth_image_test(base_topic):
    """
    1. Create a ros launcher instance.
    2. Repeatedly: 
         - Launch ifm node.
         - Create subscriber, receive one depth message, then unsubscribe.
         - kill ros node.
    
    This test will eventually fail with a timeout error waiting for the 
    depth message on my machine.
    """
    launcher = roslaunch.scriptapi.ROSLaunch()
    launcher.start()
    # Launch the ros node.
    node = launch_ifm(launcher, base_topic)
    topic = os.path.join(base_topic, "depth_with_confidence")
    depth_msgs = gen_msgs(topic, Image)
    msg = next(depth_msgs)
    while True:
        # Note: the first couple depth messages received after registering a 
        # subscriber are empty. This is another, probably separate, issue.
        print "Received depth message. Shape: ", msg.height, msg.width 
        kill_nodes(launcher, [node])
        node = launch_ifm(launcher, base_topic)
        msg = next(depth_msgs)

if __name__ == "__main__":
    rospy.init_node("UnitVectorsTest")
    receive_depth_image_test(base_topic)
```

You may also notice the first couple depth messages received after registering a subscriber are empty, but I believe that is a separate issue. This issue hasn't come up when testing the ifm3d module from cpp, so my guess is it's at the ros node level.

Thanks,
John

graugans, 2017-12-04:
@cartesian-theatrics Due to the fact I am a ROS noob, could you please be so kind and give me some instructions how I have to run your example from above. So I can check if it's an issue with the camera or with the upper layer. 

Thanks, 
Christian

cartesian-theatrics, 2017-12-04:
Hi Graugans,

No problem. Do you have ROS installed and a ROS workspace created with the ifm3d-ros module in it, i.e. are you able to launch the ifm3d-ros module? If not, you'll want to install ROS (we're running ros indigo) and go through this tutorial to create a catkin workspace:

http://wiki.ros.org/action/fullsearch/catkin/Tutorials/create_a_workspace?action=fullsearch&context=180&value=linkto%3A%22catkin%2FTutorials%2Fcreate_a_workspace%22

Assuming you do have ROS installed and are able to launch ifm3d-ros in some capacity, you should be able to runn the script above by:

1. In a separate terminal, start roscore: ./roscore
2. Set the 'hostname' and 'port' variables at the top of the script to match what you have configured or the ifm3d camera module.
4. run the script: python hanging_publishers_test.py.

tpanzarella, 2017-12-04:
[ifm3d-logs.zip](https://github.com/lovepark/ifm3d/files/1529062/ifm3d-logs.zip)
I spent the afternoon looking at this and I believe I know what is causing the issue(s).

The short version is, sometimes, the O3X seems to be returning an image chunk header (the leading 48 bytes) filled with zeros rather than the actual meta-information that we expect when fetching the Unit Vectors over XML-RPC. Because this header data was coming back as all zeros, the image chunk parser in the ifm3d framegrabber entered an infinite loop while trying to advance to the next image chunk and parse the rest of the data. I've pushed a new/debugging branch to ifm3d [here](https://github.com/lovepark/ifm3d/tree/brain-ros-issue-12) which now traps this error - i.e., it will no longer run in an infinite loop; however, when the unit vector image chunk header is all zeros, it will not know how to advance to the next image chunk and ultimately cause a framegrabber timeout. Looking at the [diff](https://github.com/lovepark/ifm3d/compare/brain-ros-issue-12) for the new branch should make it obvious as to where I am trapping this issue.

I've attached two log files with the IFM3D verbosity turned way up that illustrates this problem. In both the "OK" and "FAIL" logs, I have serialized the byte buffer representing the Unit Vectors. You will note on the "FAIL" log, the first 48 bytes (the v2 chunk header) is all zeros. In both logs the serialization of the unit vector buffer follows the message that says: "Unit vectors:" to make your searching easier. I will note that it appears (I did not validate the entire buffer) that except for the header info, the payload of the unit vectors seems to be consistent across the "OK" and "FAIL" logs.

@graugans Can you investigate on the camera side what would cause the Unit Vector image chunk header to come back as all zeros. The test script that @cartesian-theatrics sent triggers this error within a couple of minutes (at most).

@cartesian-theatrics Your test script was helpful to repeatably trigger this error and quickly find where the issue was. Thanks for that. Also, you noted in your message above:

> Finally, you may also notice the first couple depth messages received after registering a subscriber are empty, but I believe that is a separate issue.

I think you are correct. It is a separate issue. I believe I have that one resolved in the ROS node on [this branch](https://github.com/lovepark/ifm3d-ros/tree/brain-issue-12) Again, thanks for pointing this out.

Once we get this all sorted out: ifm3d, ifm3d-ros, camera firmware we will come up with the proper fix and merge all the changes in. However for now, you can feel free to clone my dev branch for this issue.

cartesian-theatrics, 2017-12-05:
@tpanzarella , glad you were able to get closer to the bottom of this. Thanks for working this. I know you said the unit vectors payload appears to be consistent, but have you had a chance to try reproducing the "Inconsistent Unit Vectors" test code? Still seems plausible that they are related.

tpanzarella, 2017-12-05:
I have not looked at the other issue. My comments regarding consistency were limited to the "OK" and "FAIL" logs I attached to my previous note. I was merely pointing out that the data payloads look correct while the headers are different. My intention was to point the firmware team in a direction they could look on the camera-side code (which I do not have access to) for the next level of insights on resolving this.

I agree, it is plausible the two issues are related.

graugans, 2017-12-05:
@tpanzarella thanks for investigating this issue. Could you and @cartesian-theatrics tell me which software version are you using on the camera?

tpanzarella, 2017-12-05:
```
$ ifm3d dump | jq .ifm3d._
{
  "Date": "Tue Dec  5 08:27:38 2017",
  "HWInfo": {
    "MACAddress": "00:02:01:40:54:09",
    "Mainboard": "#!03_M100_B01_12345678_008025483",
    "MiraSerial": "Not implemented"
  },
  "SWVersion": {
    "Algorithm_Version": "0.1.5",
    "Calibration_Device": "ff:ff:ff:ff:ff:ff",
    "Calibration_Version": "99.99.99",
    "ELDK": "GOLDENEYE_YOCTO_HARDFP-303-06d9c894636352a6c93711c7284d02b0c794a527",
    "IFM_Software": "0.1.20",
    "Linux": "Linux version 3.14.34-rt31-yocto-standard-00016-g5121435-dirty (jenkins@dettlx152) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Tue Mar 14 08:40:14 CET 2017",
    "Main_Application": "0.4.1122"
  },
  "ifm3d_version": 303
}
```

tpanzarella, 2017-12-05:
@graugans FWIW, I do not think ROS is needed to test this. You could probably write a shell script that calls an `ifm3d`-only piece of test code that will result in the same issue. Basically, create a framegrabber whose schema only grabs the unit vectors, grab the unit vectors, change the schema (to something else), grab an image/point cloud, restart the process. Keep doing this in a loop and eventually the behavior I saw was that I'd get the Unit Vector image chunk header come back as all zeros. The minimal test case is probably even simpler than what I describe above.

cartesian-theatrics, 2017-12-05:
@graugans , I produced the issue on both 0.2.0 and 0.3.2 of ifm3d, and 0.2.0/0.4.0 of ifm3d-ros.

graugans, 2017-12-06:
@cartesian-theatrics I am more interested in the camera fw version

cartesian-theatrics, 2017-12-06:
@graugans , Here's the output of *ifm3d dump*:
[ifm3d_dump.txt](https://github.com/lovepark/ifm3d/files/1534331/ifm3d_dump.txt)

Will that work?

graugans, 2017-12-06:
@cartesian-theatrics that's perfect. Thanks

graugans, 2017-12-06:
@tpanzarella your o3x is outdated and not supported anymore. You should have received a newer one a couple of month ago. Otherwise @garrettplace may can help

cartesian-theatrics, 2017-12-06:
@graugans, this issue has come up on all of the ifm units have received so far, including pre-production units with the old firmware and the newer dual-core units with new firmware. We currently use the default fw included with the device.

cartesian-theatrics, 2017-12-08:
Any update on the status of this?

graugans, 2017-12-08:
We are working on this. Due to the fact I am not in the office I am not aware of the current status.

marcinkajor, 2017-12-08:
Hi, I will try to work this issue out. @tpanzarella could you please shortly describe how I can activate the verbosity in ifm3d? It would be helpful for investigation of the problem on firmware side. I tried to add the env variable IFM3D_VLOG (I've seen it it the code) to my system but it did not help. Should it be set to any particular value? BTW. Is there any default location for generated logs in ifm3d?

As for reproducing of the defect my idea based on your description is to:
- call the framegrabber with IMG_UVEC 
- grab unit vectors 
- call the framegrabber with default schema 
- grab the XYZImage

repeat in a simple 10 minutes loop as a test case inside the ifm3d-image-tests.

Do you think I will be able to reproduce the bug is this way? 

Thanks,
Marcin

tpanzarella, 2017-12-08:
> I tried to add the env variable IFM3D_VLOG (I've seen it it the code) to my system but it did not help. Should it be set to any particular value? 

You need to set the verbosity sufficiently high. For example:
```
$ IFM3D_VLOG=30 ./your_exe
```
Would do the trick.

> Is there any default location for generated logs in ifm3d?

Yes. The logs are in `/tmp` by default.

In terms of your method to reproduce, it looks fine. However, just be sure that you actually kill the process and restart it. So, the best approach would be to write a simple C++ application that does what you say above. Then, write a shell script that wraps it in a loop to call it until you tickle the bug.

Also note, if you want the debugging info I uncovered in [this comment](https://github.com/lovepark/ifm3d/issues/12#issuecomment-349131440) you will need to use [this branch](https://github.com/lovepark/ifm3d/tree/brain-ros-issue-12) as it contains the code that will serialize the unit vector byte buffer whereby you can see that the chunk header contains all zeros rather than the expected meta-info.

marcinkajor, 2017-12-12:
Hi,
@tpanzarella thanks for your tips. The logger works fine. However I cannot reproduce this defect on my O3X. The header chunk is always filled up with proper bytes in my case in final log file (after 12 minutes of starting and killing the application). I attach my application and shell script so maybe you can have a look if I follow your suggestions. I will appreciate any advice.

Marcin

reproduce_iss12.cpp:
```c++
int main(int argc, const char **argv)
{
  std::chrono::time_point<std::chrono::system_clock> end;                                 
  end = std::chrono::system_clock::now() + std::chrono::minutes(5);                      
                                                                                          
  while(std::chrono::system_clock::now() < end)
    {                                                                                     
      auto cam = ifm3d::Camera::MakeShared();                                             
      auto fg = std::make_shared<ifm3d::FrameGrabber>(cam, ifm3d::IMG_UVEC);             
      auto im = std::make_shared<ifm3d::ImageBuffer>();                                   
      
    if (! fg->WaitForFrame(im.get(), 1000))
      {
        std::cerr << "Timeout waiting for camera!" << std::endl;
       // return -1;
      }

      cv::Mat uvec = im->UnitVectors();                                                   
      fg = std::make_shared<ifm3d::FrameGrabber>(cam);                                                                                                                         

    if (! fg->WaitForFrame(im.get(), 1000))
      {
        std::cerr << "Timeout waiting for camera!" << std::endl;
       // return -1;
      }                                                                                     
    auto distImg = im->XYZImage();                                                      
    }                

    return 0;
}
```

run_reproduce_iss12.sh:

```sh
#!/bin/bash

#remove previously generated logs
if ls /tmp/ifm3d.* 1> /dev/null #silence
then
  rm /tmp/ifm3d.*
else
  echo "No 'ifm3d*' files to remove"
fi

file="/tmp/append_all.log"
if [ -f "$file" ]
then
  rm /tmp/append_all.log
else
  echo "No append_all.log to remove"
fi

#time handling
testTimeS=720 #12m
end=$((SECONDS+$testTimeS))

while [ $SECONDS -lt $end ] 
do
  IFM3D_VLOG=30 ./reproduce_iss12&
  sleep 2s #allow fg to capture some frames
  killall -9 reproduce_iss12
done

#kill just in case
killall -9 reproduce_iss12

#append logs into one file
find /tmp -name '*.INFO.*' -exec cat {} + >> /tmp/append_all.log
```

cartesian-theatrics, 2017-12-12:
@markinkajor, I'll also take a look at this tomorrow to see if it reproduces the issue for me. In the meantime, can someone point me to the latest/most stable camera firmware version?

GarrettPlace, 2017-12-12:
Hi John,

A link to the current LTS O3X FW can be found below:

https://vault.ifmefector.com:443/home/link/browser?ll=AAAAAAAAAFYAAAy0DiXECk_yXJ8RIb8pWcpWgwdMaJk%3D

Best Regards,

Garrett

3D Imaging
484-343-5676

bsbc, 2017-12-12:
The [link above](
https://vault.ifmefector.com:443/home/link/browser?ll=AAAAAAAAAFYAAAy0DiXECk_yXJ8RIb8pWcpWgwdMaJk%3D) is for O3X_Firmware_1.0.62.swu, which is exactly what is running on the latest (production) O3X units.

tpanzarella, 2017-12-12:
Some follow up regarding [this analysis](https://github.com/lovepark/ifm3d/issues/12#issuecomment-349131440). I am continuing to use @cartesian-theatrics ROS launcher test script to drive this work as it seems to tickle the bug quickest. I am also using my outdated firmware and hardware per @graugans comment, however, for now, the behavior is consistent between what @cartesian-theatrics sees and what I see with my unsupported setup. I will move to a supported setup as soon as feasible. For now I wanted to test a theory...

I wanted to validate that the camera is in fact sending back the data I am logging (i.e., there is not a bug in a library we are using or some other data transformation deep in the stack). From my most recent test I can say with high confidence that the data I am seeing is coming back from the camera. Specifcially, I am talking about acquiring the unit vector data over XML-RPC and getting the 48-byte image chunk header sent back as all zeros. To test this, as I said above, I am using @cartesian-theatrics ROS script. I have turned logging up to "30" using the `IFM3D_VLOG` environment variable. Additionally, I captured output from `tcpdump` to validate the data using external instrumentation. Here are my findings from an exemplary test run...

In the ifm3d log, I am logging the raw byte buffer coming back from the camera which represents the Unit Vector packet (obtained over XML-RPC). Here is a *snippet* from my log file showing these data. Recall, these are the raw bytes and for our purposes here I am only pasting in the first 63 bytes of the payload data. As can be seen below, the first 48 bytes, the *chunk header* is all zeros followed by what appears to be meaningful pixel data (i.e., the actual unit vectors).
```
I1212 12:10:14.406785 31899 frame_grabber_impl.hpp:332] Caching unit vectors from xmlrpc...
I1212 12:10:14.693455 31899 frame_grabber_impl.hpp:351] Unit vectors: 
[00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,4e,be,f5,be,a3,70,bc,be,2a,de,4b,3f,b7,0a,f4
```

Correlating this back to the `tcpdump` wire capture I grabbed, here are some data showing the unit vector request and the beginnings of the response:

(NOTE: for reference, in this set up, my computer is 192.168.0.92 and the camera is 192.168.0.69)

```
12:10:14.406881 IP (tos 0x0, ttl 64, id 4877, offset 0, flags [DF], proto TCP (6), length 349)
    192.168.0.92.58712 > 192.168.0.69.80: Flags [P.], cksum 0x8341 (incorrect -> 0x9d70), seq 3595:3892, ack 2364, win 304, options [nop,nop,TS val 1795938 ecr 713320], length 297: HTTP, length: 297
	POST /api/rpc/v1/com.ifm.efector/ HTTP/1.1
	Host: 192.168.0.69
	Accept: */*
	Content-Type: text/xml
	User-Agent: Xmlrpc-c/1.33.14 Curl/7.47.0
	Content-Length: 131
	
	<?xml version="1.0" encoding="UTF-8"?>
	<methodCall>
	<methodName>getUnitVectors</methodName>
	<params>
	</params>
	</methodCall>
E..]..@.@......\...E.X.PM..'t......0.A.....
..gb.
.hPOST /api/rpc/v1/com.ifm.efector/ HTTP/1.1
Host: 192.168.0.69
Accept: */*
Content-Type: text/xml
User-Agent: Xmlrpc-c/1.33.14 Curl/7.47.0
Content-Length: 131

<?xml version="1.0" encoding="UTF-8"?>
<methodCall>
<methodName>getUnitVectors</methodName>
<params>
</params>
</methodCall>

12:10:14.407173 IP (tos 0x0, ttl 64, id 12807, offset 0, flags [DF], proto TCP (6), length 52)
    192.168.0.69.80 > 192.168.0.92.58712: Flags [.], cksum 0x9fe3 (correct), ack 3892, win 620, options [nop,nop,TS val 713320 ecr 1795938], length 0
E..42.@.@......E...\.P.Xt...M..P...l.......
.
.h..gb
12:10:14.568172 IP (tos 0x0, ttl 64, id 12808, offset 0, flags [DF], proto TCP (6), length 1500)
    192.168.0.69.80 > 192.168.0.92.58712: Flags [.], cksum 0x177b (correct), seq 2364:3812, ack 3892, win 620, options [nop,nop,TS val 713336 ecr 1795938], length 1448: HTTP, length: 1448
	HTTP/1.1 200 OK
	Status: 200 OK
	Content-Length: 616648
	Content-Type: text/xml
	Date: Wed, 24 May 2017 10:18:17 GMT
	Server: lighttpd/1.4.35
	
	<?xml version="1.0" encoding="UTF-8"?><methodResponse><params><param><value><base64>AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAATr71vqNwvL4q3ks/twr0vivUvL7XSUw/bVTyvmc3vb4ztUw/dJvwvlqavb5AIE0/xd/uvv38vb72ik0/YSHtvk5fvr5V9U0/SmDrvkvBvr5YX04/f5zpvvQiv77+yE4/+9Xnvj+Ev75BMk8/xgzmvjHlv74im08/2kDkvsJFwL6dA1A/N3LivvKlwL6sa1A/36DgvrsFwb5N01A/0czevh5lwb5+OlE/EPbcvhbEwb49oVE/mhzbvqAiwr6EB1I/bkDZvriAwr5RbVI/kGHXvlzewr6g0lI//n/Vvos7w75vN1M/vJvTvkGYw768m1M/yLTRvnj0w76A/1M/JcvPvjBQxL67YlQ/1d7NvmWrxL5oxVQ/2O/LvhYGxb6FJ1U/L/7Jvjxgxb4MiVU/3gnIvti5xb7+6VU/4BLGvuASxr5QSlY/QBnEvllrxr4IqlY//BzCvjzDxr4dCVc/FR7AvoUax76MZ1c/kBy+vjRxx75TxVc/bRi8vkPHx75tIlg/rxG6vq8cyL7Xflg/XAi4vndxyL6R2lg/dPy1vpbFyL6TNVk/+u2zvgkZyb7bj1k/8tyxvs1ryb5m6Vk/XMmvvty9yb4sQlo/RLOtvjwPyr41mlo/pJqrvt1fyr5x8Vo/hX+pvsOvyr7iR1s/7GGnvu3+yr6HnVs/2EGlvlJNy75X8ls/Uh+jvvKay75SRlw/XPqgvsrny750mVw/+9KevtYzzL6761w/NKmcvhJ/zL4fPV0/DH2avnvJzL6gjV0/iE6YvhATzb473V0/rR2Wvstbzb7rK14/gOqTvqyjzb6ueV4/B7WRvq7qzb5/xl4/SH2Pvs0wzr5dEl8/SEONvgd2zr5CXV8/DQeLvlm6zr4sp18/nsiIvsD9zr4X8F8/AIiGvjdAz74AOGA/OkWEvr2Bz77jfmA/VACCvlHCz77AxGA/pHJ/vuwB0L6PCWE/euB6voxA0L5RTWE/NUp2vjB+0L4BkGE/569xvtO60L6b0WE/nRFtvnT20L4dEmI/ZW9ovg4x0b6EUWI/T8ljvp9q0b7Lj2I/bB9fviaj0b7zzGI/ynFavp3a0b71CGM/ecBVvgQR0r7RQ2M/jQtRvllG0r6CfWM/ElNMvpZ60r4GtmM/HZdHvrut0r5b7WM/vdd[!http]
E...2.@.@.."...E...\.P.Xt...M..P...l.{.....
.
```
[... cut ...] so makes me think that we need to do some additional debugging on the camera/firmware side to get better insights as to why this is the case. I do *not* have access to the firmware and cannot really help in that department. If I can find the time, I will try to get a supported h/w and s/w stack up to see if this can all be duplicated with the latest stuff.

My full logs are attached for deeper analysis by interested/invested parties.
[1.zip](https://github.com/lovepark/ifm3d/files/1552790/1.zip)

bsbc, 2017-12-12:
Apparently we have a mixture of firmwares. We will make sure to update all units to 1.0.62 and re-evaluate the problem. Thanks

cartesian-theatrics, 2017-12-12:
Okay, the new fw version seems to fix this issue for me as well. Thanks @tpanzarella  and @marcinkajor for your work on this.

However, the "Inconsistent Unit Vectors" issue is still present (see https://github.com/lovepark/ifm3d/issues/10).