ifm/ifm3d #13: Timeout waiting for image buffer from camera

cartesian-theatrics, 2017-12-08:
Hello,

We're more issues with some of our new cameras. We are consistently failing to get any data out of some of our ifm3d cameras. Not sure what it is or how related to other issues it is. The /tmp/ifm3d.WARNING log repeatedly logs "Timeout waiting for image buffer from camera":

```
W1208 00:59:17.567520  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:18.068783  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:18.570011  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:19.071226  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:19.572422  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:20.073669  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:20.574818  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:22.267246  8242 camera.cpp:593] Skipping Imager, null value -- should be string!
W1208 00:59:22.340270  8242 camera.cpp:593] Skipping SpatialFilter, null value -- should be string!
W1208 00:59:22.340303  8242 camera.cpp:593] Skipping TemporalFilter, null value -- should be string!
W1208 00:59:23.059981  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:23.561238  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:24.062542  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:24.563859  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:25.065052  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:25.566350  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:26.067569  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:26.568842  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:27.070117  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:27.571358  8242 frame_grabber_impl.hpp:286] Timeout waiting for image buffer from camera
W1208 00:59:29.458178  8242 camera.cpp:593] Skipping Imager, null value -- should be string!
```

Here's the output of ifm3d dump:
[ifm3d_dump.txt](https://github.com/lovepark/ifm3d/files/1541093/ifm3d_dump.txt)

This is a major, blocking issue that we need to resolve as quickly as possible.

Thanks,

John

tpanzarella, 2017-12-08:
Could you bring up your camera as follows:

```
$ IFM3D_VLOG=30 roslaunch ifm3d camera.launch
```

 Let it run and timeout a few times as it is doing above. Then kill the process. Send over the output of `/tmp/ifm3d.INFO`.

NOTE: I won't be near a camera until about 10:30 AM EST tomorrow. Additionally, I am not sure I have access to the firmware version you are running, so, it may be hard (if not impossible) for me to duplicate your issue to take a look. None-the-less, if you can send over the verbose logging output I suggest above, I can do my best to make some inferences on what may be going on.

tpanzarella, 2017-12-08:
Also, I just wanted to double check, when the s/w is in this timeout loop, is the illumination unit on the front of the camera face pulsing or is it off?

cartesian-theatrics, 2017-12-08:
Here's the ifm3d.INFO log:
[ifm3d_info.txt](https://github.com/lovepark/ifm3d/files/1541351/ifm3d_info.txt)
 
The camera light is pulsing, although at a significantly slower rate then the working cameras.

cartesian-theatrics, 2017-12-08:
hmm, seems the log got overwritten somehow. Here's a better one:
[ifm3d_info.txt](https://github.com/lovepark/ifm3d/files/1541364/ifm3d_info.txt)

cartesian-theatrics, 2017-12-08:
One more for good measure:
[ifm3d_info.txt](https://github.com/lovepark/ifm3d/files/1541392/ifm3d_info.txt)

Realized I was running the camera node in a background process, so that was screwing up the logs. It's getting late over here...

tpanzarella, 2017-12-08:
So, giving this a quick look, it looks symptomatic of not getting pixel data back from the camera. With the verbosity turned up like you have it, we would see a bunch of chatter coming back from the frame grabber indicating that it is processing pixel data. 

As noted in an earlier comment, I do not have recent hardware nor firmware so I cannot really reproduce, but, to me it looks like the camera is thrashing while context switching between "edit mode" and "run mode". One thing you can try is to increase the timeout value. I think you have it set to 500 ms, maybe try 1000 ms -- I'm not optimistic that it will work, but it is a simple thing to try. My thinking is, it could just be the initial context switch from edit mode to run mode that needs a bit longer than .5 secs to bootstrap itself -- it is a shot in the dark. I'd expect subsequent calls to `WaitForFrame` to not timeout after .5 secs.

You can also try connecting to the camera with Vision Assistant and see if the pixel data are returned. If it works, which I expect it should, it can tell us that the usage paradigm on the camera side has changed since the last revision of ifm3d and ifm3d has to adapt. If VA does not work, then we can focus more on the camera side of the issue.

graugans, 2018-05-02:
Due to the fact FW version 1.0.22 is available for download can we close this issue? For more details see [Issue #56](https://github.com/lovepark/ifm3d/issues/56#issuecomment-385892061)