ifm/ifm3d #18: Publishers Occasionally Hang (Re-open)

cartesian-theatrics, 2018-01-08:
Hello,

Hate to re-open this issue, but the hanging ifm publishers issues are re-appearing on .62 and .64 when we run with this config:
[ifm_config.txt](https://github.com/lovepark/ifm3d/files/1612564/ifm_config.txt)

@tpanzarella , do you want to to see if you can reproduce with this config?

tpanzarella, 2018-01-08:
@cartesian-theatrics I can confirm that this behavior persists and is consistent with what we saw in #12 
 
Here is a snippet from my log file showing the "zero header info" on the unit vectors:

```
I0108 16:54:39.939026 28203 frame_grabber_impl.hpp:332] Caching unit vectors from xmlrpc...
I0108 16:54:40.150566 28203 frame_grabber_impl.hpp:351] Unit vectors: 
[00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,fd,a4,bb,be,76,4b,60,bf,c6,50,a0,3e,2e,9b,ba,be,42,2b,60,bf,54,38,a2,3e,6d,90,b9,be,fa,09,60,bf,5c,1f,a4,3e,a8,84,b8,be,9a,e7,5f,bf,16,06,a6,3e,c9,77,b7,be,17,c4,5f,bf,ca,ec,a7,3e,b7,69,b6,be,66,9f,5f,bf,be,d3,a9,3e,5b,5a,b5,be,80,79,5f,bf,2f,bb,ab,3e,a0,49,b4,be,5a,52,5f,bf,56,a3,ad,3e,.........
```

Interestingly, I ran the test script against the camera with "default settings" (I think ... I don't have a dump available), and I did not run into the issue. I ran the test for 45 minutes then manually stopped it. Then, I loaded your camera configuration on to my O3X, and got the error in 3.5 minutes.

Here is the info on my h/w and s/w stack:

```
$ ifm3d dump | jq .ifm3d._
{
  "Date": "Mon Jan  8 17:02:55 2018",
  "HWInfo": {
    "MACAddress": "00:02:01:40:67:3F",
    "Mainboard": "#!02_M162_B30_12345678_008026357",
    "MiraSerial": "Not implemented"
  },
  "SWVersion": {
    "Algorithm_Version": "0.1.7",
    "Calibration_Device": "00:02:01:40:67:3f",
    "Calibration_Version": "1.0.0",
    "ELDK": "GOLDENEYE_YOCTO_HARDFP-312-06d9c894636352a6c93711c7284d02b0c794a527",
    "IFM_Software": "1.0.62",
    "Linux": "Linux version 3.14.34-rt31-yocto-standard-00014-gb439e15-dirty (jenkins@dettlx152) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Thu Jul 13 10:49:08 CEST 2017",
    "Main_Application": "0.4.1382"
  },
  "ifm3d_version": 303
}
```

graugans, 2018-01-10:
I'll fire up a docker image with ROS and ifm3d to reproduce this on our side.

graugans, 2018-01-11:
A big thank you to @cartesian-theatrics for providing such a nice script to reproduce the issue. I have set-up a clean ROS + ifm3d-ros environment in a docker container. The Dockerfile is uploaded [here](https://github.com/graugans/docker_images/tree/master/ifm3d/ros/kinetic/xenial)  Plus this patch for reproducing this bug.

```diff
From fd0be831b65727f411cf0591ed31ae03d624607b Mon Sep 17 00:00:00 2001
From: Christian Ege <christian.ege@ifm.com>
Date: Thu, 11 Jan 2018 08:51:02 +0100
Subject: [PATCH] Added scripts and files for reproducing #18

---
 ifm3d/ros/kinetic/xenial/Dockerfile                |   6 ++
 .../ros/kinetic/xenial/hanging_publishers_test.py  | 120 +++++++++++++++++++++
 ifm3d/ros/kinetic/xenial/ifm_config.json           |  74 +++++++++++++
 3 files changed, 200 insertions(+)
 create mode 100644 ifm3d/ros/kinetic/xenial/hanging_publishers_test.py
 create mode 100644 ifm3d/ros/kinetic/xenial/ifm_config.json

diff --git a/ifm3d/ros/kinetic/xenial/Dockerfile b/ifm3d/ros/kinetic/xenial/Dockerfile
index 78dacb0..0fc4708 100644
--- a/ifm3d/ros/kinetic/xenial/Dockerfile
+++ b/ifm3d/ros/kinetic/xenial/Dockerfile
@@ -101,5 +101,11 @@ RUN /bin/bash -c "source /opt/ros/$ROS_DISTRO/setup.bash && mkdir -p ~/catkin/if
 
 RUN sudo chmod a+rwx /ros_entrypoint.sh
 
+# Copy the config for reproducing the crash
+COPY ./ifm_config.json /home/ifm3d/ifm_config.json
+COPY ./hanging_publishers_test.py /home/ifm3d/hanging_publishers_test.py
+RUN sudo chmod a+rwx /home/ifm3d/ifm_config.json
+RUN sudo chmod a+rwx /home/ifm3d/hanging_publishers_test.py
+
 ENTRYPOINT ["/ros_entrypoint.sh"]
 CMD ["bash"]
diff --git a/ifm3d/ros/kinetic/xenial/hanging_publishers_test.py b/ifm3d/ros/kinetic/xenial/hanging_publishers_test.py
new file mode 100644
index 0000000..8f87b68
--- /dev/null
+++ b/ifm3d/ros/kinetic/xenial/hanging_publishers_test.py
@@ -0,0 +1,120 @@
+#!/usr/bin/env python
+import roslaunch
+import rospy
+import os
+import signal
+import time
+from sensor_msgs.msg import Image
+from multiprocessing import Lock
+
+
+hostname = "192.168.0.83"
+port = 80
+base_topic = "ifm_left"
+
+
+class Atom(object):
+    # Don't trust assignment thread safety.
+    def __init__(self, v):
+        self._lock = Lock()
+        self._value = v
+
+    def reset(self, v):
+        with self._lock:
+            self._value = v
+
+    def deref(self):
+        with self._lock:
+            return self._value
+
+
+def kill_nodes(launcher, nodes):
+    for node in nodes:
[... 151 lines cut ...]
+        "MACAddress": "00:02:01:40:6C:24",
+        "Mainboard": "#!02_M162_B30_12345678_008026357",
+        "MiraSerial": "Not implemented"
+      },
+      "SWVersion": {
+        "Algorithm_Version": "0.1.7",
+        "Calibration_Device": "00:02:01:40:6c:24",
+        "Calibration_Version": "1.0.0",
+        "ELDK": "GOLDENEYE_YOCTO_HARDFP-312-06d9c894636352a6c93711c7284d02b0c794a527",
+        "IFM_Software": "1.0.56",
+        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00014-gb439e15-dirty (jenkins@dettlx152) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Thu Jul 13 10:49:08 CEST 2017",
+        "Main_Application": "0.4.1355"
+      },
+      "ifm3d_version": 200
+    }
+  }
+}
-- 
2.7.4
```
After launching the docker image by 
```
docker run -ti ifm3d-ros:latest
```
I do start a tmux session with a split and execute in the one pane the ``roscore`` and in the other the script ``hanging_publishers_test.py`` Now I am able to dig deeper in this issue and see what's going on on the camera side.

graugans, 2018-01-12:
I would like to give an update on this. It looks like the issue with the empty unit vectors is located. I will do more testing on my side and check if there are more side effects to take consideration on.

graugans, 2018-05-02:
Due to the fact FW version 1.0.22 is available for download can we close this issue? For more details see [Issue #56](https://github.com/lovepark/ifm3d/issues/56#issuecomment-385892061)

graugans, 2018-11-16:
I'll close this due to the fact this was fixed on firmware level with the release ``1.0.22`` In case this is still present we can reopen this issue.

closed 2018-11-16 by graugans