ifm/ifm3d #20: Cmake error, xmlrpc

pfrydlewicz, 2018-01-17:
While configuring for a visual studio project, I get the following cmake error. However, the macros seem not to be defined anywhere in the cmake file. Is the cmake file complete?
Or, shall I specify thos macros by hand? Is it a list of folders for each macro? (...LIBRARIES, ..._DIRS).

Thank you
Paul

CMake Error at C:/Program Files/CMake/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  XMLRPC was not found.  Make sure the entries XMLRPC_* are set.  (missing:
  XMLRPC_C_FOUND XMLRPC_LIBRARIES XMLRPC_INCLUDE_DIRS)

Arag24, 2018-01-17:
Hi Paul,
Would you like to compile ifm3d on Windows ? 
I personnally used : https://github.com/graugans/ifm3d/tree/windows/doc
and https://github.com/graugans/ifm3d/blob/windows/doc/doc/windows.md

Just take care to get the windows branch and put xml-rpc library in 3rdParty folder.
Also renaming `xmlrpc-c-config.main` by `xmlrpc-c-config` files in xmlrpc-c folder, and compiling, linking and running well.

Thanks to @graugans for this. Hope it helps.

graugans, 2018-01-17:
Thx @Arag24 for pointing out the issue with ``xmlrpc-c-config`` I'll fix this in the xmlrpc repo.

pfrydlewicz, 2018-01-18:
Thanks, that helped!

closed 2018-01-18 by pfrydlewicz