ifm/ifm3d #21: Windows, cmake issue with PCL or boost

pfrydlewicz, 2018-01-18:
Hello,

when configuring cmake under windows, I get the problem below.
I downloaded PCL version 1.7.2 as source code and pointed to ...\pcl-1.7.2\cmake as PCL_DIR.
Looking into the PCL folder, there is only a pcl_pclconfig.cmake file.

Where can I fix this?

- For now I only need the camera module, thus no need for PCL. However when switching off all modules this error still persists.

- Also, there is the BoostConfig.cmake file that is mentioned in the cmake error log with a static PCL folder as entry:
include("C:/Projets/PCL/SuperBuild_boost_1_50/CMakeExternal/Install/Boost/share/boost-${Boost_VERSION}/cmake/BoostConfig.cmake")

The boost (1.50.0) and PCL (1.7.2) directories are set appropriatly.

Thanks for help.

Here the cmake error message:
--------------------------------------------------------------------------------------------------

CMake Error at D:/Tools/Boost/share/cmake/boost/BoostConfig.cmake:15 (include):
  include could not find load file:

    C:/Projets/PCL/SuperBuild_boost_1_50/CMakeExternal/Install/Boost/share/boost-/cmake/BoostConfig.cmake
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.9/Modules/FindBoost.cmake:236 (find_package)
  modules/camera/CMakeLists.txt:19 (find_package)

Boost  found.
globbed: 
CMake Error at D:/Tools/Boost/share/cmake/boost/BoostConfig.cmake:15 (include):
  include could not find load file:

    C:/Projets/PCL/SuperBuild_boost_1_50/CMakeExternal/Install/Boost/share/boost-/cmake/BoostConfig.cmake
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.9/Modules/FindBoost.cmake:236 (find_package)
  modules/framegrabber/CMakeLists.txt:12 (find_package)

Boost  found.
Found Boost components:
   system
globbed: 
CMake Error at D:/Tools/Boost/share/cmake/boost/BoostConfig.cmake:15 (include):
  include could not find load file:

    C:/Projets/PCL/SuperBuild_boost_1_50/CMakeExternal/Install/Boost/share/boost-/cmake/BoostConfig.cmake
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.9/Modules/FindBoost.cmake:236 (find_package)
  modules/framegrabber/src/libifm3d_framegrabber/CMakeLists.txt:1 (find_package)

Boost  found.
Found Boost components:
   system;date_time;regex

graugans, 2018-01-18:
@pfrydlewicz thanks for testing this feature. I am sorry but the image module is still missing in the windows port. This is due to the fact that the PCL and OpenCV are big beasts and there are no pre compiled versions for all Visual Studio versions available. For me it was important to have step one before step two and the first step was the integration of the basic feature set. In step two what we are currently working on we try to provide as much information as possible to handle OpenCV and PCL. Meanwhile a solution could be to use the provided image module and strip down the image module to not use OpenCV and PCL. I guess this will do the trick for most of the windows users.

Of course you can add this reduced image module in your own project. You only need to derive from [ifm3d::ByteBuffer](https://github.com/lovepark/ifm3d/blob/master/modules/image/include/ifm3d/image/image_buffer.h#L40) I guess an simple_image module just putting out the image data as ``std::vector`` or ``std::array`` should be no big deal. If you have the time to strip down and write some test cases I guess @tpanzarella will merge this.

tpanzarella, 2018-01-19:
@pfrydlewicz We do have plans to create additional modules that will project the pixel data into alternate data structures. Some of which are header-only libraries like [Eigen](http://eigen.tuxfamily.org) and [xtensor](https://github.com/QuantStack/xtensor). This should make the build issues much simpler. I do not have a specific time frame for you except to say that it is coming *soon*.

Arag24, 2018-01-19:
@tpanzarella Would be really happy to get this module splitting. Great added value for Windows user; to my prospective.

graugans, 2018-01-19:
Here is a first draft version of the image module build instructions on Windows. I've not yet tested it myself.

```patch
From 0a1d46c453e60cf5f000a6fecbb81f4d49679c26 Mon Sep 17 00:00:00 2001
From: Marco Kirchner <Marco.Kirchner@ifm.com>
Date: Thu, 18 Jan 2018 16:42:47 +0100
Subject: [PATCH] add initial support for image module on windows

---
 doc/windows.md                                     | 101 ++++++++++++++++++---
 .../framegrabber/include/ifm3d/fg/byte_buffer.h    |  15 ++-
 .../src/libifm3d_framegrabber/byte_buffer.cpp      |   3 +-
 modules/image/src/libifm3d_image/CMakeLists.txt    |  13 ++-
 modules/tools/src/libifm3d_tools/CMakeLists.txt    |   6 +-
 5 files changed, 115 insertions(+), 23 deletions(-)

diff --git a/doc/windows.md b/doc/windows.md
index 45325fa..c37813a 100644
--- a/doc/windows.md
+++ b/doc/windows.md
@@ -1,22 +1,28 @@
 # How to build ifm3d library from source on Windows.
 
-As a first step we build ifm3d without the image module.
-
 ## Prepare dependent tools
 
-Make sure that you have installed CMake and VS2017 64bit before you move to the next step.
+Make sure that you have installed CMake and Visual Studio 64bit before you move to the next step.
 
 * [CMake v3.9.1](http://www.cmake.org)
 * [Visual Studio Community 2017](https://www.visualstudio.com)
 * [Git for Windows](https://git-for-windows.github.io/) or any git client
 
+Note: Visual Studio 2015 or later is required as older versions don't support all required features of C++11
+
 ## Download dependencies
 
-Start with downloading Glog, curl and XMLRPC libraries from repository listed below.
+Start with downloading the libraries from repositories listed below.
 
 * [Glog](https://github.com/google/glog)
 * [xmlrpc-c](https://github.com/graugans/xmlrpc-c)
 * [curl](https://github.com/curl/curl)
+* [Eigen](https://github.com/eigenteam/eigen-git-mirror)
+* [FLANN](https://github.com/mariusmuja/flann)
+* [VTK](https://gitlab.kitware.com/vtk/vtk)
+* [PCL](https://github.com/PointCloudLibrary/pcl)
+
+
 
 For all the download operations, the command line is a msys git bash. All operations are relative to each project top folder. 
 
@@ -44,6 +50,36 @@ $ git clone https://github.com/curl/curl.git
 $ git checkout -b curl-7_47_1 curl-7_47_1
 ```
 
+### Download FLANN
+```
+$ git clone https://github.com/mariusmuja/flann.git
+$ git checkout -b 1.9.1 1.9.1
+```
+
+### Download Eigen
+```
+$ git clone https://github.com/eigenteam/eigen-git-mirror.git
+$ git checkout -b 3.3.4 3.3.4
+```
+
+### Download VTK
[... 189 lines cut ...]
+set_property(TARGET ifm3d_image_shared PROPERTY POSITION_INDEPENDENT_CODE 1)
 
 target_link_libraries(
   ifm3d_image_shared
diff --git a/modules/tools/src/libifm3d_tools/CMakeLists.txt b/modules/tools/src/libifm3d_tools/CMakeLists.txt
index 629fe4e..a5a0d92 100644
--- a/modules/tools/src/libifm3d_tools/CMakeLists.txt
+++ b/modules/tools/src/libifm3d_tools/CMakeLists.txt
@@ -55,8 +55,12 @@ if (BUILD_MODULE_IMAGE)
     )
   set_target_properties(ifm3d_tools_shared PROPERTIES
     AUTOMOC TRUE
-    COMPILE_FLAGS "-Wno-deprecated -Wno-deprecated-declarations"
     )
+  if(NOT WIN32)
+    set_target_properties(ifm3d_tools_shared PROPERTIES
+      COMPILE_FLAGS "-Wno-deprecated -Wno-deprecated-declarations"
+      )
+  endif()
 else()
   set_target_properties(ifm3d_tools_shared PROPERTIES
     VERSION ${IFM3D_VERSION_STRING}
-- 
2.7.4
```

pfrydlewicz, 2018-01-26:
Thanks for the commitment to do a windows version.

closed 2018-01-26 by pfrydlewicz

GMSLLB, 2019-05-30:
Hello,
when configuring cmake of pcl under windows, I get the problem below.
-- Configuring incomplete, errors occurred!
See also "D:/%IFM3D_BUILD_DIR%/pcl/build/CMakeFiles/CMakeOutput.log".
See also "D:/%IFM3D_BUILD_DIR%/pcl/build/CMakeFiles/CMakeError.log".

I looked into the file CMakeError.log, it explained this:
D:\%IFM3D_BUILD_DIR%\pcl\build\CMakeFiles\CMakeTmp\src.cxx(2):fatal error
C1083: Unable to open the include file: "mm_malloc.h": No such file or directory.

How can I fix this?

Thanks for help!

Arag24, 2019-05-31:
@GMSLLB, could you please open a new Thread on this just for clarity please ? 
It looks like more a problem with  PCL building. What is version of PCL? We can propose also propose you Win x64 binaries with PCL build if preferred.
Just open new thread, please.

GMSLLB, 2019-06-04:
> @GMSLLB, could you please open a new Thread on this just for clarity please ?
> It looks like more a problem with PCL building. What is version of PCL? We can propose also propose you Win x64 binaries with PCL build if preferred.
> Just open new thread, please.

@Arag24 , the version of PCL is 1.8.1 from pcl-1.8.1 https://github.com/PointCloudLibrary/pcl.git; and I built PCL with Win x64. 
Now, it always exited that:
CMake Error at C:/Program Files/CMake/share/cmake-3.13/Modules/FindBoost.cmake:2064 <message>:
  Unable to find the requested Boost libraries.
  Unable to find the Boost header files. Please set BOOST_ROOT to the root directory containing Boost or     
  BOOST_INCLUDEDIR to the directory containing Boost's headers.
Call Stack <most recent call first>:
   cmake/pcl_find_boost.cmake:43 <find_package>
   CMakeLists.txt:418 <include>

graugans, 2019-06-04:
@GMSLLB please open a new issue with the exact Visual Studio Version you are currently using. The Windows version and if you are running a 64Bit or 32Bit version of Windows. Another thing which might make things struggle is that on Windows 10 some set-ups default to the powershell and not the old cmdline.