ifm/ifm3d #25: Windows build project error

BernaPT, 2018-01-31:
Hi, i dont understand what is the problem, but someone can help me, the CMakeError.txt have this

Determining if the include file sys/filio.h exists failed with the following output:
Change Dir: C:/ifm3d-deps/build/CMakeFiles/CMakeTmp

Run Build Command:"C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/MSBuild/15.0/Bin/MSBuild.exe" "cmTC_384c9.vcxproj" "/p:Configuration=Debug" "/p:VisualStudioVersion=15.0"
Microsoft (R) Build Engine version 15.5.180.51428 for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

Build started 31-01-2018 16:28:53.

Project "C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\cmTC_384c9.vcxproj" on node 1 (default targets).

PrepareForBuild:

  Creating directory "cmTC_384c9.dir\Debug\".

  Creating directory "C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\Debug\".

  Creating directory "cmTC_384c9.dir\Debug\cmTC_384c9.tlog\".

InitializeBuildStatus:

  Creating "cmTC_384c9.dir\Debug\cmTC_384c9.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.

ClCompile:

  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.12.25827\bin\HostX86\x64\CL.exe /c /Zi /W3 /WX- /diagnostics:classic /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_384c9.dir\Debug\\" /Fd"cmTC_384c9.dir\Debug\vc141.pdb" /Gd /TC /errorReport:queue "C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c"

  Microsoft (R) C/C++ Optimizing Compiler Version 19.12.25835 for x64

  Copyright (C) Microsoft Corporation.  All rights reserved.

  

  cl /c /Zi /W3 /WX- /diagnostics:classic /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_384c9.dir\Debug\\" /Fd"cmTC_384c9.dir\Debug\vc141.pdb" /Gd /TC /errorReport:queue "C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c"

  CheckIncludeFile.c

  

C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1): fatal error C1083: Cannot open include file: 'sys/filio.h': No such file or directory [C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\cmTC_384c9.vcxproj]

Done Building Project "C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\cmTC_384c9.vcxproj" (default targets) -- FAILED.

Build FAILED.

"C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\cmTC_384c9.vcxproj" (default target) (1) ->

(ClCompile target) -> 

  C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1): fatal error C1083: Cannot open include file: 'sys/filio.h': No such file or directory [C:\ifm3d-deps\build\CMakeFiles\CMakeTmp\cmTC_384c9.vcxproj]

    0 Warning(s)
    1 Error(s)

graugans, 2018-02-01:
@BernaPT could you please be so kind and provide us with the command line you have executed this. What project are you trying to build ifm3d or any of it's dependencies?

BernaPT, 2018-02-01:
@graugans i try the build for windows, i build dependencies.

use the command line you put in the document.
cmake -Ax64  -DCMAKE_WINDOWS_EXPORT_ALL_SYMBOLS=ON -DBUILD_SDK_PKG=OFF -DBUILD_TESTS=OFF -DBUILD_MODULE_EXAMPLES=OFF -DCMAKE_PREFIX_PATH="C:/ifm3d-deps/curl" -DBoost_INCLUDE_DIR="C:/ifm3d-deps/boost_1_64_0" -DBOOST_LIBRARYDIR="C:/ifm3d-deps/boost_1_64_0/lib64-msvc-14.1" -DBoost_COMPILER=-vc141 -DBoost_USE_STATIC_LIBS=ON -DEIGEN_INCLUDE_DIR="C:/ifm3d-deps/eigen/include/eigen3" -DCMAKE_BUILD_TYPE=Release ..
> cmake --build . --clean-first --config Release --target INSTALL

and the CMake ask for the OpenCV_DIR and i put, after i receives this errors.
send you the files
[CMakeError.log](https://github.com/lovepark/ifm3d/files/1687439/CMakeError.log)
[CMakeOutput.log](https://github.com/lovepark/ifm3d/files/1687440/CMakeOutput.log)

graugans, 2018-02-08:
@BernaPT thanks to @BigBoot we have updated the build instructions for windows in the Pull-Request #34. I guess you have mixed up your directories during the checkout/build.

BernaPT, 2018-02-08:
Thanks very much i go see the new update.

BernaPT, 2018-02-17:
with the new update its possible compile all, but when start the application dont find the opencv_core340.dll

[image]

BernaPT, 2018-02-17:
I find the problem, the direction its wrong in my pc because the opencv its build in VC15, normally this its not done in 14.1?

graugans, 2018-02-17:
It could also been an issue with OpenCV not beeing in the path.

closed 2018-02-17 by BernaPT