ifm/ifm3d #35: CMake Package Configuration File for the ifm3d

graugans, 2018-02-08:
Based on all the issues around the XMLRPC lib like described in #25 I was wondering how @tpanzarella thinks about the [CMake Package Configuration File](https://cmake.org/cmake/help/git-master/manual/cmake-packages.7.html#creating-packages) for the ifm3d lib.

tpanzarella, 2018-02-08:
I'm not entirely up to date on all the issues here, especially since I do not personally feel the *pain* inflicted by using Windows. However, I would suggest you create a branch, implement this feature and detail why we should mainline it. So long as it keeps the Linux core unaffected and stable, I would have no problems with it.

graugans, 2018-02-08:
It's not related to Windows. With this you get rid of the need to provide a CMake find script to locate the ifm3d and you can tune for example the ``CXX_FLAGS`` which are used to compile the application ifm3d is linked to.

graugans, 2018-02-08:
This post on the [CMake ML](https://cmake.org/pipermail/cmake/2010-April/036496.html) explains this pretty good. 

```
FindXXX.cmake modules are for non-native CMake software (i.e. not compiled and installed by CMake) and XXX-config.cmake modules are for CMake-based projects. So, if you are developing a software using CMake, you should create a XXX-config.cmake module for it which you install along with the binaries and other development files. See: https://cmake.org/pipermail/cmake/2010-April/036487.html
```

tpanzarella, 2018-02-08:
This sounds good. Do you want to take a pass at it?

graugans, 2018-02-08:
I would pass this to @BigBoot :)

graugans, 2018-02-09:
@tpanzarella during the internal discussion to which module the xxx-config.cmake should belong one idea came to my mind. How do you think about moving the examples folder into it's own project?

My suggestion is ``ifm3d-examples`` which CMakeLists.txt contains content like this:

```cmake
cmake_minimum_required (VERSION 3.0)
project (ifm3d-examples)

find_package (ifm3d 0.8 REQUIRED)

add_executable (ex-file_io main.cpp)
target_link_libraries (ex-file_io ifm3d::camera ifm3d::framegrabber ifm3d::image)
```
All the other dependant libraries and headers are filled in magically.

In my opinion the benefit of having a separate project makes it easier for an app developer to get started.

tpanzarella, 2018-02-09:
I think this is a fantastic idea. I can bootstrap a new repo with a stub Readme so it can stay on the day Github with the core packages.

graugans, 2018-03-19:
@tpanzarella One question was raised during implementation of this work. Which also was discovered by @maxses during creation of the Openembedded layer. Is this ``_shared`` suffix really needed for this libs? It seems there is no ``_static`` version of the lib available as build target. Can we get rid of this? Otherwise including linking against the libs would look like this:

```
target_link_libraries (ex-file_io ifm3d::camera_shared ifm3d::framegrabber_shared ifm3d::image_shared)
```

tpanzarella, 2018-03-19:
I prefer to leave the target named as it is to "future proof" us for *when* someone asks for a static library. However, on my `dev-0.9.0` branch, I have achieved what you want. It is managed through a target property. See [this](https://github.com/lovepark/ifm3d/blob/dev-0.9.0/modules/camera/src/libifm3d_camera/CMakeLists.txt#L77) as an example. Then on the *consuming side* linking look like [this](https://github.com/lovepark/ifm3d-examples/blob/master/CMakeLists.txt#L28) -- which is what you are asking for.

tpanzarella, 2018-05-04:
CMake architecture has been significantly reworked in 0.9.0 and will consider this request closed as a result.

closed 2018-05-04 by tpanzarella