ifm/ifm3d #38: [Question] Coding stle

graugans, 2018-02-12:
What is the intended coding sytle for this project to contribute to. From the source formating I do expect [GNU indentation style](https://en.wikipedia.org/wiki/Indentation_style#GNU_style). 

Is it worth of thinking about a [CONTRIBUTING.md](https://github.com/uncrustify/uncrustify/blob/master/CONTRIBUTING.md)?

tpanzarella, 2018-02-12:
For now the coding style should follow [the golden rule](https://llvm.org/docs/CodingStandards.html#golden-rule). I know that I have recently merged at least one PR that was *not* consistent, but we will have to go back and fix it. The style of the core of the code that I initially wrote has a very distinct style that should be very easy to parse -- we need to codify that (more on that later). Sadly, I do not like any of the "published standards" that are out there. For each one, there is *at least* one *major* thing that I cannot handle when looking at source code. So, the ifm3d code is currently, a mash-up of a few styles, taking the parts I personally like -- I guess this is first-mover advantage :-)

To answer your first question regarding brace indentation, yes. The link you provide captures the proper brace style -- it leaves no ambiguity as to where the block starts/stops. Additionally, even for a single line following an `if`, I prefer including braces, eg.:

```
if (YouDonIncludeBraces())
{
  EventuallyABugWillCreepIn();
}
```
There are other big peeves of mine as well. E.g., whitespace, not tabs (`whitespace-mode` in Emacs should make this obvious), no more than 80 characters per line (again, Emacs can enfore this, I am sure vim and others can too), etc.

I do think it is worth pulling together a `Contributing.md`. Probably better than that is a `.clang-format`. However, the latter should not be used by those not used to handling power tools. It should be written, tested, and signed off by our core team then integrated into the editors of those who which to contribute. A while back I started one but never finished it -- IIRC, the style most closely resembles [LLVM](https://llvm.org/docs/CodingStandards.html) but with modifications -- like GNU style curly braces, etc.

This is a good topic and we should figure out the best way to formalize the style. I would expect it to carry through to more than just ifm3d.

tpanzarella, 2018-02-12:
> Personally I can live with most of the coding style when it is consistent across a file. So the first-mover approach is fine for me. But then the first-mover needs to do the documentation part.

I will work on a document. I will put it in my queue. I cannot promise a date to turn it around, but, it is now on my radar and I will handle this.

theseankelly, 2019-10-09:
Cleaning out old issues; feel free to reactivate if still relevant.

closed 2019-10-09 by theseankelly