ifm/ifm3d #41: pkg-config files for ifm3d

pfrydlewicz, 2018-02-23:
Hi,
as far as I see there are no .pc files generated for pkg-config, am I right?
I think it would be of a real benefit to have those generated on make install.

What is the roadmap in this regard?
Thanks
Paul

tpanzarella, 2018-02-23:
You are correct. 

AFAIK, it is not on any of the development team’s roadmap. However, we do plan to update our cmake architecture to a so called more “modern” style which should help (if you are using cmake in your project). 

If this is really important to you, please flag this issue as an enhancement request and we will consider it on our next sprint.

pfrydlewicz, 2018-02-23:
Ok, that sounds good.
Please instruct me on how to post an enhancement request in a proper way.
Thanks

tpanzarella, 2018-02-23:
Make the case for what you want (politely) and use the github interface to add a label to the issue. There is already one called “enhancement”.  Even better is a PR with a rationale for why it should be merged ... but adding the enhancement label to this issue would be the bare minimum.

pfrydlewicz, 2018-02-23:
Ok, I'll do that

pfrydlewicz, 2018-02-23:
Sorry @tpanzarella, I don't see a way to assign labels.
Is it maybe disabled for the repository?

tpanzarella, 2018-02-23:
I’ll look into it once I’m at the office. Not sure, maybe a permissions thing. I’ll check.

graugans, 2018-02-23:
There is already an issue covering this [cmake package config stuff](https://github.com/lovepark/ifm3d/issues/35)

pfrydlewicz, 2018-02-24:
I agree with @graugans .
Both seem to be helpful, the FindX.make files and the .pc files.
The FindX.make files for installing libifm3d, some of the dependencies do not provide any .pc files.

.pc files are interesting for developing custom programs, even with different build environments than cmake. The user does not need to care about compiler flags (like the -std=c++11), directories and so on. Also dealing with different versions  of the libifm3d becomes way easier and less error prone, which is interesting for support staff.

theseankelly, 2019-10-09:
Cleaning out old issues; My understanding (based on comments issue #35 linked above) is that this isn't relevant anymore. Feel free to reactivate if I'm wrong.

closed 2019-10-09 by theseankelly