ifm/ifm3d #43: Port of libo3d3xx OEM module

tpanzarella, 2018-03-06:
One of the last major features not available in `ifm3d` yet available in [libo3d3xx](https://github.com/lovepark/libo3d3xx) is the zero-copy framegrabber for embedded/OEM builds. It would be good to port that software to ifm3d and allow us to begin deprecating `libo3d3xx`.

Arag24, 2018-04-29:
@tpanzarella ; @graugans  : One more remark for you guys, as described in your main Readme.md 

> Once 18.04 is released, we will drop support for 14.04 and only be supporting 18.04 and 16.04. If you are using this library embedded in a product, you should plan accordingly.

It looks like from 0.9.0 revision of ifm3d Ubuntu 14.04 is no more supported...
This information is missing from libo3d3xx. To my prospective it would be great to have alignement with this piece of info especillay if we want to focus everyone on ifm3d at the end. This is clearly a point for OEM module. (Typical questions from customer 😄 )

What do you think of ?

tpanzarella, 2018-04-30:
In my view, `libo3d3xx` is legacy software -- albeit, supported. I do not see moving it feature-forward as we are doing with `ifm3d`. To that end, I think `libo3d3xx` will continue to support 14.04 as it does today. I would urge all new developments to use `ifm3d`. Should `ifm3d` users want to continue with 14.04, a back port would need to be maintained by the stakeholders as we move the mainline forward to a more modernized stack.

closed 2022-12-15 by github-actions[bot]