ifm/ifm3d #44: Improve documentation and integrate documentation with readthedocs.org

tpanzarella, 2018-03-06:
Documentation for ifm3d could be significantly improved -- both the content and quality as well as the presentation. On the latter topic, this issue proposes integrating our documentation with the [Read The Docs](https://readthedocs.org) website. Putting the infrastructure in place to automatically build updated documentation upon release or commit to a particular branch would be ideal.

Also, let's start a list of what documentation needs to be made available:

- [ ] Supported platforms
- [ ] Building the software
- [ ] Building and installing on Windows (should be treated separately from above)
- [ ] How to implement a new image container
- [ ] How to configure the imager using the JSON interface
- [ ] Using the `ifm3d` command-line tool

Please help me add to this list!

(NOTE: Is there a good way to make this check list shared so others can add to it? This is beyond my github skills).

We note that, per @graugans [comment](https://github.com/lovepark/ifm3d/issues/35#issuecomment-364398823) in #35 we will continue to provide code examples on using the library. The side benefit of making the examples a separate project also provides a nice way to illustrate how to integrate `ifm3d` into your project to include the build chain.

tpanzarella, 2018-05-08:
@graugans Would appreciate if you can lead the charge on getting the last checkbox on my list above checked off.

graugans, 2018-05-09:
@tpanzarella frankly I am struggling a bit with this task. I do understand the need of documentation. I would  like to check what is provided in the official documentation to not have a duplicated information. On the other hand I understand that one is interested in this library may  want to have direct access to the documentation and not go through the whole ifm documentation first. Another point we should care about is the fact that we have different versions of firmware and each version may have a different set of parameters.

Meanwhile since quite some time we expose the underlying schema from which we generate the complete parameter set. This typically contains the type of parameter and its range plus a default parameter.

```
http://192.168.0.69/cgi-bin/schema.sh
```
The link above generates a tar ball  containing all schema information. With the help of a python script and a camera connected we can generate a list of parameters and their limits. This can be generated for each version. supporting the schema download.

graugans, 2018-05-09:
I give [doscon](https://github.com/lbovet/docson) a shot.

graugans, 2018-05-09:
This is my proposal:

```
$ wget https://github.com/lbovet/docson/archive/v1.1.0.zip
$ unzip  v1.1.0.zip
$ cd docson-1.1.0/
$ mkdir -p o3d3xx
$ curl -s http://192.168.0.69/cgi-bin/schema.sh | tar xvf - -C o3d3xx
$ python3 -m http.server
```
When you now start a web browser you can open the following URIs

## Device configuration
[http://127.0.0.1:8000/index.html#/o3d3xx/gecco/schema/device.json](http://127.0.0.1:8000/index.html#/o3d3xx/gecco/schema/device.json)

## Application
[http://127.0.0.1:8000/index.html#/o3d3xx/gecco/schema/device.json](http://127.0.0.1:8000/index.html#/o3d3xx/gecco/schema/application.json)

tpanzarella, 2018-05-09:
The information provided by the schema is good in terms of understanding what the *names* , *data types*, and *valid ranges* are but this does not answer the primary question of *semantically* what do the tunable parameters mean. For example, we get asked often things like "What is the 'SymmetryThreshold'"? What happens when I turn it up? Down? Also data types are *not* units of measure. Here is a simple example. The extrinsic calibration parameters exposed on the O3D are:

```
"ExtrinsicCalibRotX": "0",
"ExtrinsicCalibRotY": "0",
"ExtrinsicCalibRotZ": "0",
"ExtrinsicCalibTransX": "0",
"ExtrinsicCalibTransY": "0",
"ExtrinsicCalibTransZ": "0",
```

OK, so what is the coord frame? We know that these relate to the optical frame and not the camera frame because we establish a camera frame consistent with the Right Hand Rule (consistent with what most roboticists would expect). What are the units? First time I saw it, I just assumed meters and radians. However, I was *very* wrong. These are mm and degrees.

My point is, much more important than saying the parameter `ExtrinsicCalibRotX` takes a `float` is to define what it means and its unit of measure in the *real world*.

Unfortunately, the example parameters I use above for the extrinsics are actually the *obvious* ones. Others are much more obtuse. For example, what does `ThreeFreqMax3FLineDistPercentage` mean? What is its unit of measure? I simply see it is "80".

graugans, 2018-05-09:
I guess a few point we can address with a description of the parameters for others we need more of an application note. Others can me marked as internal use. The ``ThreeFreqMax3FLineDistPercentage`` is a good example. I guess less than a handful people know how to control this parameter.

GarrettPlace, 2018-05-09:
We should discuss how to communicate an internal vs. external parameter.  

Application notes are always a good idea.  

Could SV help in this effort?

graugans, 2018-05-09:
@GarrettPlace You are right. I started an internal discussion about some of the parameters, maybe there is a specific reason why we expose them. Related to SV, I guess he can support us when it comes to the application notes.

tpanzarella, 2018-05-09:
I'd just like to add that there is nothing wrong with exposing *deeply technical* parameters via the XMLRPC interface (and by association the JSON serialization). However, we just need to explain what they mean. In some cases, I assume, having access to these via the `ifm3d` Linux command line will be desirable. Said another way, hiding the parameter is not documenting it.

graugans, 2018-05-09:
@tpanzarella I agree hiding to avoid documentation. The idea behind this was, if nobody needs this than we can skip it. Today I learned we need those parameters so I'll document them.

closed 2022-12-14 by github-actions[bot]