ifm/ifm3d #45: memory leak on test

jChaudet, 2018-03-07:
I develop a wrapper of ifm3d based on the ROS one for an buildroot application.
When I test it with valgrind, I detect some memory leak (these tests are realized in buildroot (openCv 3.2) and Desktop environment). On my own test (wrapper), memory leaks increase with time.

**Desktop environment:** 
Ubuntu 16.04
kernel 4.10.0-9-generic
gcc version 6.3.0
Valgrind-3.11.0
vtk 7.1
openCv 3.4
pcl 1.8
standard installation of glog (apt-get install libgoogle-glog-dev)
ifm3d 0.8.2

(Desktop) In order to trace the leaks, I launched valgrind on test of ifm3d library (with the O3X and without).
$ valgrind --leak-check=full --show-leak-kinds=all --allow-mismatched-debuginfo=yes -v ./ifm3d-image-tests
[image_test_valgrind.txt](https://github.com/lovepark/ifm3d/files/1788502/image_test_valgrind.txt)
I had the same behavior with pcicclient_test, I didn't test the others.

Arag24, 2018-03-13:
@jChaudet : Question is OpenCV 3.2 or 3.4 used, please ?

jChaudet, 2018-03-13:
I have two separate configuration:

**Desktop environment:**
Ubuntu 16.04
kernel 4.10.0-9-generic
gcc version 6.3.0
Valgrind-3.11.0
vtk 7.1
openCv 3.4
pcl 1.8
standard installation of glog (apt-get install libgoogle-glog-dev)
ifm3d 0.8.2

**Buildroot**
gcc version 5.4.0
Valgrind-3.12.0
openCv 3.2
pcl 1.8
ifm3d 0.8.2

I test valgrind on the both configuration and the result is the same.
The attached file (image_test_valgrind.txt) comes from the desktop version.

jChaudet, 2018-03-13:
I complete my first message:

In order to industrialize the driver that I develop, I verify if my driver has some leaks and others weird behaviors. This driver looks like the ros wrapper.
To realize that, I create a simple main program:
- instantiate the sensor
- sleep 10 seconds
- Display a point cloud
- Close the main properly (delete pointer)
I launched Valgrind on that program and found some leaks.
-------------------------------------------------------------------------------

Today, I realize two new tests:
```
int main_test_sensor ()
{
    int retval = 0;

    ifm3d::Camera::Ptr base = std::make_shared<ifm3d::Camera>
    (ifm3d::DEFAULT_IP , 80 , ifm3d::DEFAULT_PASSWORD);

    std::cout << "Launch the sensor O3X." << std::endl;
    sleep(6);
    std::cout << "Verification of the sensor type." << std::endl;

    try
    {
        if (base->IsO3X())
        {
            json dump = base->ToJSON();
            std::string sFrameRate = dump["ifm3d"]["Apps"][0]["Imager"]["FrameRate"];
            std::cout << "Framerate is : " << sFrameRate << std::endl;
        }
        else
        {
            std::cout << "Sensor is not an O3X." << std::endl;
        }
    }
    catch (const ifm3d::error_t& ex)
    {
        if (ex.code() == IFM3D_XMLRPC_TIMEOUT)
        {
            std::cout << "Timeout." << std::endl;
            retval = -1;
        }
        else
        {
            std::cout << "Sensor is not connected." << std::endl;
            retval = -2;
        }
    }

    return retval;
}
```
With this single test, I launch valgrind and obtain:
[main_test_valgring_no_delete.txt](https://github.com/lovepark/ifm3d/files/1807387/main_test_valgring_no_delete.txt)

```
int main_test_sensor ()
{
    int retval = 0;

    ifm3d::Camera::Ptr cam;
    ifm3d::FrameGrabber::Ptr fg;
    ifm3d::ImageBuffer::Ptr im;

    try
    {
        std::cout << "Launch the driver O3X." << std::endl;

        cam = ifm3d::Camera::MakeShared(ifm3d::DEFAULT_IP, 80, ifm3d::DEFAULT_PASSWORD);
        sleep(6);

        fg = std::make_shared<ifm3d::FrameGrabber>(cam, ifm3d::DEFAULT_SCHEMA_MASK);

        im = std::make_shared<ifm3d::ImageBuffer>();

        std::cout << "Verification of the sensor type." << std::endl;
    }
    catch (const ifm3d::error_t& ex)
    {
        std::cout << "Sensor is not connected." << std::endl;
    }

    sleep(10);

    try
    {
        if (cam->IsO3X())
        {
            json dump = cam->ToJSON();
            std::string sFrameRate = dump["ifm3d"]["Apps"][0]["Imager"]["FrameRate"];
            std::cout << "Framerate is : " << sFrameRate << std::endl;
        }
        else
        {
            std::cout << "Sensor is not an O3X." << std::endl;
        }
    }
    catch (const ifm3d::error_t& ex)
    {
        if (ex.code() == IFM3D_XMLRPC_TIMEOUT)
        {
            std::cout << "Timeout." << std::endl;
            retval = -1;
        }
        else
        {
            std::cout << "Sensor is not connected." << std::endl;
            retval = -2;
        }
    }

    return retval;
}
```
And this one give me:
[main_valgrind_fg_im_cam.txt](https://github.com/lovepark/ifm3d/files/1807406/main_valgrind_fg_im_cam.txt)

If I increase the sleep time, the leaks stay the same.

tpanzarella, 2018-03-13:
I took a look into this. In short, I do not think there is anything to worry about. Let me explain...

I used your functions from above as a test bed and wrapped them into an executable that I ran 4 different ways. This code is available on [this issue 45 branch](https://github.com/lovepark/ifm3d/tree/issue-45). The executable in question is [this one](https://github.com/lovepark/ifm3d/blob/issue-45/modules/examples/ex-leak-check.cpp). Please take a look at that. It is important to follow along below.

In my first run, the `main()` function looked like this (unnecessary comments removed for clarity):
```
int main(int argc, const char **argv)
{
  return 0;
}
```
While this code links to the `ifm3d` libraries and it associated dependencies, it does, "nothing".  I ran it as follows:

```
$ valgrind --leak-check=full --show-leak-kinds=all ./ex-leak-check 2>/tmp/leak0.txt
```

As expected, there was no output to the terminal, however, we get the following output from valgrind.
[leak0.txt](https://github.com/lovepark/ifm3d/files/1808570/leak0.txt)

A few things we should note (snippets taken from `leak0.txt` file):
```
==18364== LEAK SUMMARY:
==18364==    definitely lost: 0 bytes in 0 blocks
==18364==    indirectly lost: 0 bytes in 0 blocks
==18364==    possibly lost: 0 bytes in 0 blocks
==18364==    still reachable: 79,240 bytes in 176 blocks
==18364==    suppressed: 0 bytes in 0 blocks
```
The reported leaks are "still reachable". In other words, valid pointers to these memory blocks still existed at the time `main()` exited, however, they were not deallocated. This is not really a problem. Particularly if we look at some of the detail in the stack traces. For example:

```
==18364== 160 bytes in 1 blocks are still reachable in loss record 175 of 176
==18364==    at 0x4C2E0EF: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==18364==    by 0x7A0DAC9: ??? (in /usr/lib/x86_64-linux-gnu/libgflags.so.2.1.2)
==18364==    by 0x7A0E2E2: google::FlagRegisterer::FlagRegisterer(char const*, char const*, char const*, char const*, void*, void*) (in /usr/lib/x86_64-linux-gnu/libgflags.so.2.1.2)
==18364==    by 0x7A0BE60: ??? (in /usr/lib/x86_64-linux-gnu/libgflags.so.2.1.2)
==18364==    by 0x40106B9: call_init.part.0 (dl-init.c:72)
==18364==    by 0x40107CA: call_init (dl-init.c:30)
==18364==    by 0x40107CA: _dl_init (dl-init.c:120)
==18364==    by 0x4000C69: ??? (in /lib/x86_64-linux-gnu/ld-2.23.so)
```
We see at the bottom of this stack is `_dl_init` and ultimately at the top of the stack is a call to `new` made by the Google GFlags (`libgflags.so`) library. This type of thing is common to perform one-time upfront memory pool allocators and other initialization routines in libraries. For example, it appears, OpenCV is doing something similar in this stack trace taken from the same valgrind output:

```
==18364== 8 bytes in 1 blocks are still reachable in loss record 1 of 176
==18364==    at 0x4C2E0EF: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==18364==    by 0x624C7FF: ??? (in /opt/ros/kinetic/lib/libopencv_core3.so.3.3.1)
==18364==    by 0x612EF7D: ??? (in /opt/ros/kinetic/lib/libopencv_core3.so.3.3.1)
==18364==    by 0x40106B9: call_init.part.0 (dl-init.c:72)
==18364==    by 0x40107CA: call_init (dl-init.c:30)
==18364==    by 0x40107CA: _dl_init (dl-init.c:120)
==18364==    by 0x4000C69: ??? (in /lib/x86_64-linux-gnu/ld-2.23.so)
```
In all of these cases where `_dl_init` is at the bottom of the stack, these are one-time allocations done when the shared object file is linked into your executable by the runtime linker. These are valid memory regions with valid pointers to them. They are allocated prior to `main()` (recall, our `main()` in this case does nothing), but never get deallocated before `main()` exits. However, the OS reclaims that memory after the executable finishes. So, for this example, I don't think we need to worry.

In my second run of the test bed, my main looked like this:

```
int main(int argc, const char **argv)
{
  return main_test_sensor1();
}
```

I ran it like (with an O3X plugged in):

```
$ valgrind --leak-check=full --show-leak-kinds=all ./ex-leak-check 2>/tmp/leak1.txt
Launch the sensor O3X.
[... 134 lines cut ...]
Framerate is : 5
```
Here is the valgrind output
[leak2-10.txt](https://github.com/lovepark/ifm3d/files/1808681/leak2-10.txt)

Here is the leak summary:
```
==18932== LEAK SUMMARY:
==18932==    definitely lost: 0 bytes in 0 blocks
==18932==    indirectly lost: 0 bytes in 0 blocks
==18932==    possibly lost: 0 bytes in 0 blocks
==18932==    still reachable: 80,945 bytes in 184 blocks
==18932==    suppressed: 0 bytes in 0 blocks
```

We note that, the amount of memory "leaked" is the same whether the function ran once or ten times. So, this, to a first order approximation, supports my hypothesis from above.

With all of the above being said, I feel like these leaks are not something to worry about. Summarizing my reasoning from the examples above include:

1. In a `main()` that does nothing, we incur the one-time allocations at shared object load time.
2. In your first example function, the memory leaked is identical to the `main()` that does nothing. So, again, shared object one-time allocations at load time.
3. In your second example, we incur a bit more "leaked memory" however, we believe this is JIT memory allocation and caching based on our test of whether it runs one time or ten times, the amount of leaked memory is the same.
4. Finally, in all cases, the memory "leaked" is "still reachable" which means valid pointers to the heap allocations exist they are just not explicitly deleted before main exists which is not what we would expect anyway in the case of these long-term memory pool allocations.

Is this consistent with your analysis and something you are comfortable with?

jChaudet, 2018-03-14:
I'm ok with the first part, _dl_init is a library constructor allocation. It will be interesting to know if there is a destructor to call (as with the library XML) or if we have to deal with this leak.

For the second part, I'm not sure that the leaks are constant. For example, in your file "leak2-10.txt", line 1604:

```
==18932== 64 bytes in 1 blocks are still reachable in loss record 176 of 181
==18932==    at 0x4C2E0EF: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==18932==    by 0x5E2DD47: std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::_M_insert_aux(__gnu_cxx::__normal_iterator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x5E2BDCF: google::GetLoggingDirectories[abi:cxx11]() (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x5E2CDDB: ??? (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x5E284DB: google::LogMessage::SendToLog() (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x5E2615A: google::LogMessage::Flush() (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x5E26378: google::LogMessage::~LogMessage() (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x528DEB4: ??? (in /home/tpanzarella/dev/ifm3d/build/modules/camera/src/libifm3d_camera/libifm3d_camera.so.0.8.2)
==18932==    by 0x52AB2F2: ifm3d::Camera::Impl::WrapInEditSession(std::function<void ()>) (in /home/tpanzarella/dev/ifm3d/build/modules/camera/src/libifm3d_camera/libifm3d_camera.so.0.8.2)
==18932==    by 0x5297F03: ifm3d::Camera::FromJSON_(nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, std::function<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>, std::function<void ()>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int) (in /home/tpanzarella/dev/ifm3d/build/modules/camera/src/libifm3d_camera/libifm3d_camera.so.0.8.2)
==18932==    by 0x529985F: ifm3d::Camera::FromJSON(nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&) (in /home/tpanzarella/dev/ifm3d/build/modules/camera/src/libifm3d_camera/libifm3d_camera.so.0.8.2)
==18932==    by 0x528AF8B: ifm3d::Camera::FromJSONStr(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/tpanzarella/dev/ifm3d/build/modules/camera/src/libifm3d_camera/libifm3d_camera.so.0.8.2)
==18932== 
==18932== 81 bytes in 2 blocks are still reachable in loss record 177 of 181
==18932==    at 0x4C2E0EF: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==18932==    by 0x55D83DE: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21)
==18932==    by 0x5E2CEB3: ??? (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x5E284DB: google::LogMessage::SendToLog() (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x5E2615A: google::LogMessage::Flush() (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x5E26378: google::LogMessage::~LogMessage() (in /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0)
==18932==    by 0x528DEB4: ??? (in /home/tpanzarella/dev/ifm3d/build/modules/camera/src/libifm3d_camera/libifm3d_camera.so.0.8.2)
==18932==    by 0x52AB2F2: ifm3d::Camera::Impl::WrapInEditSession(std::function<void ()>) (in /home/tpanzarella/dev/ifm3d/build/modules/camera/src/libifm3d_camera/libifm3d_camera.so.0.8.2)
==18932==    by 0x5297F03: ifm3d::Camera::FromJSON_(nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, std::function<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>, std::function<void ()>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int) (in /home/tpanzarella/dev/if [... cut ...]
==18932==    by 0x528AF8B: ifm3d::Camera::FromJSONStr(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/tpanzarella/dev/ifm3d/build/modules/camera/src/libifm3d_camera/libifm3d_camera.so.0.8.2)
```

The same functions are called several times and new leaks appear with different sizes. I agree that your test show us a constant leak but I don't think this test is enough to validate that the leaks are constant.

In order to verify this, I will test my driver on an embedded target during several hours/days and let you know how the memory evolves.

closed 2019-12-23 by jChaudet