ifm/ifm3d #47: Relax build dependency on PCL (Enhancement request)

drewm1980, 2018-03-08:
ifm3d's build scripts arent finding Ubuntu's system-wide PCL, even when I point it at the standard location where the libraries are installed. I'm also having issues with PCL's binary release, and building it from source on Ubuntu 16.04; I tried several tags of older releases from PCL's git repo; nothing just builds cleanly.

I have had several negative experiences with PCL over the years, I'd rather disable it in the build than burn time fixing it again.

Is PCL a dependency that can be relaxed by disabling some targets in the CMake build scripts when it's not available, or are you using it even in the critical code paths for reading data from sensors?

graugans, 2018-03-08:
@drewm1980 what version of Ubuntu are you using? There is a solution planned to have PCL not a must have. For details have a look at #42

tpanzarella, 2018-03-08:
As noted by @graugans above, plans are in place to make the image container of choice *pluggable*. To that end, *soon*, PCL will not be required at all. However, that is not the case today. But you do have some options that may potentially help.

The heaviest dependency on PCL right now is in the `tools` module. This is because our basic viewer application is a wrapper around PCL's. So, if you are just looking to build the library so that you can write code or use the [ROS interface](https://github.com/lovepark/ifm3d-ros) and visualize via rviz, the dependency on PCL drops significantly -- i.e., it really turns into a "header-only" dependency (in our `image` module) as we simply need to construct the point cloud. Since PCL is heavily templated, we do not link to anything PCL-related in the `image` module.

If you would like to try the above, you can try to build `ifm3d` like:

```
$ mkdir build
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_MODULE_EXAMPLES=OFF -DBUILD_MODULE_TOOLS=OFF -DBUILD_MODULE_PCICCLIENT=OFF ..
$ make
$ make check
$ make package
$ make repackage
$ sudo dpkg -i ifm3d_1.0.0_amd64-camera.deb
$ sudo dpkg -i ifm3d_1.0.0_amd64-framegrabber.deb
$ sudo dpkg -i ifm3d_1.0.0_amd64-image.deb
```
Realize, with this build, among a few other specialized things (e.g., pcic interface), you will not have access to the `ifm3d` command line tool. Again, this may not be a big deal if you just want to use the library.

Finally, in terms of your Ubuntu system, to build as I have suggested above (assuming 16.04), if you issue the following command, you should be OK:

```
$ sudo apt-get install libpcl-dev
```

tpanzarella, 2018-03-15:
@graugans For the 0.9.0 release, I was thinking about removing the `ifm3d viewer` from the core code base. This is the primary *pain* associated with PCL. I would then create a new repository, to be released in sync with 0.9.0, that is called `ifm3d-pcl-viewer`. This would have a new command line application called `ifm3d-pcl-viewer`. So, it would not be as uniform as the other core `ifm3d` tools, but, it would significantly reduce the PCL dependencies. This leads to the question of, do we need to come up with a more lightweight viewer that we ultimately add back into the core to be invoked as `ifm3d viewer`? Which then leads to the question, would this be a blocker to releasing 0.9.0? I would say "no", b/c the functionality would still exist, just in a different repo. However, we may contemplate not having a basic viewer as a blocker to 1.0.0, but that gives us a little time to think of the best solution. Thoughts?

graugans, 2018-03-16:
@tpanzarella That's an awesome idea to split the viewer from the core functionality. This makes things much easier for building an embedded version.

tpanzarella, 2018-03-16:
OK. This will be the plan for 0.9.0

drewm1980, 2018-04-10:
Hello @graugans and @tpanzarella, sorry for posting a bug report and then disappearing for a while! We were under pressure to quickly evaluate a small pile of 3D cameras for a new project; we got stuck with this camera and just had to move on in the short term. 

At our company, python is the glue that binds everything together; if the python bindings for a camera driver pretty much just work and produce numpy arrays, we're happy. Well, maybe we're not happy, but at least our ire is directed at some other library for viewing data (i.e. recompiling OpenCV yet again to make imshow work, or fighting with vispy's Qt/OpenGL dependency). I'm not sure there exists a good solution for writing a lightweight, cross platform viewer. Intel's realsense team has lots of development resources, and making a viewer "just work" seems to be tough even for them. It seems like they're at the point where the 2D part of their viewer usually works even when the 3D part is broken, which is commendable.

Anyway, thanks again for the response! I can't work on trying it immediately, but I very likely will at the start of our next 3D project. Without your open source driver, we would not have even bought this camera; your efforts are certainly appreciated.

tpanzarella, 2018-04-10:
Thanks for the feedback @drewm1980  Just a few notes for next time...

Our current "python binding" is through ROS. Getting our software setup to run within a ROS environment is quite simple and using rviz as the viewer works well. We have step-by-step instructions, assuming a fresh install of ubuntu, for both [14.04 + Indigo](https://github.com/lovepark/ifm3d-ros/blob/master/doc/indigo.md) or [16.04 + Kinetic](https://github.com/lovepark/ifm3d-ros/blob/master/doc/kinetic.md). We also have some non-ROS users who have effectively created `ifm3d` python bindings using [PyBind11](https://github.com/pybind/pybind11). We do have plans to create our own, non-ROS dependent Python binding to `ifm3d`, however, the timeline is not yet defined. FWIW, we will also be releasing Matlab bindings as well -- again, release data TBD. So, definitely check back for your next project.

Finally, I am working on the topic directly related to this issue (lessening the dependence on PCL). This update is planned for April. So, if you want to check back at the end of the month, compilation should be easier for you.

Again, thanks for the feedback.

tpanzarella, 2018-05-04:
As of 0.9.0, it will be possible to have a running ifm3d with out any dependency on PCL. Documentation will be included with the official release of 0.9.0 (*very* soon).

closed 2018-05-04 by tpanzarella