ifm/ifm3d #52: [Question] Importing dumps from another camera

graugans, 2018-03-29:
I am running ifm3d version ``0.8.3``
```
$ ifm3d --version
```

When I try to import the dump from another sensor I get the following error  message:
```
ifm3d error: 101000
Sensor: The parameter name is invalid
```
To get rid of this error message the failing field can be removed:
```
cat  /app/ifm.json | jq 'del(.ifm3d.Apps[0].Id)'  | ifm3d --ip=192.168.0.82 config
```
For convenience it might be useful to skip the network part while importing: 
```
cat  /app/ifm\(1\).json | jq 'del(.ifm3d.Net,.ifm3d.Apps[0].Id)'  | ifm3d --ip=192.168.0.82 config
```
Or even white listing the stuff you are interested to import
```
cat  /app/ifm.json | jq '.ifm3d.Apps[0].Imager'  | ifm3d --ip=192.168.0.82 config
```
So what is the intended use case for importing dumps?

tpanzarella, 2018-03-29:
Is the `Apps.Id` parameter something that will always be unique on a camera-to-camera basis and likely not important for anything but on-camera bookkeeping? If so, perhaps we skip that when doing an import. (There is already infrastructure in there to make that easy via the “read-only” LUT)

I would not suggest skipping the Network part on import, as the JSON interface *is* the way to set camera network parameters — I.e., I am not looking at the code at the moment, however, I do not think we expose the explicit networking mutators via the public api.

Finally, the intended use-case for importing dumps is that it *is* the interface for mutating camera/imager parameters. The actual API calls to set the individual parameters are quite thin ... intentionally. The idea is to take care in what you import — typically you would not want to simply dump from one camera and import into another — however, that is a supported/intended use case. Supporting pipelines like you present above is also the intended way for this interface to be used.

graugans, 2018-04-05:
I am perfectly fine how we deal with Ethernet, this was more a note to have an example how to exclude the Ethernet configuration when importing configuration. I'll internally recheck the use of the ``Apps.Id`` parameter an add a PR if we can set it as read-only.

graugans, 2018-05-03:
Due to the fact we have merged #61 this can also be closed

commit 146ff8ba (2022-02-02): Merge branch 'o3r/fix-version-parsing' into 'o3r/main-next'

fix version comparison & add unittests

Closes #52

See merge request syntron/support/csr/ifm3d/ifm3d!100
closed 2018-05-03 by graugans