ifm/ifm3d #53: Errors when compiling in debug mode from VS2015

tnikolla, 2018-04-04:
Windows 10 Home
Visual Studio 2015

I created a solution in vs with just the ex-file_io.cpp. It works on Release (output 3 files) but when I compile in Debug I get this error:

>Error	C4996	'std::uninitialized_copy::_Unchecked_iterators::_Deprecate': Call to 'std::uninitialized_copy' with parameters that may be unsafe - this call relies on the caller to check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNINGS. See documentation on how to use Visual C++ 'Checked Iterators'

I googled for a while with no results. Ideas and suggestions are welcome.

graugans, 2018-04-05:
I guess you can fix this by adding this to the top level ``CMakeLists.txt``

```cmake
if ("${CMAKE_CXX_COMPILER_ID}" MATCHES "MSVC")
    set(DEFAULT_COMPILE_DEFINITIONS ${DEFAULT_COMPILE_DEFINITIONS}
        _SCL_SECURE_NO_WARNINGS  # Calling any one of the potentially unsafe methods in the Standard C++ Library
        _CRT_SECURE_NO_WARNINGS  # Calling any one of the potentially unsafe methods in the CRT Library
    )
endif ()
```
The code above is originated [here](https://github.com/cginternals/cmake-init/blob/master/cmake/CompileOptions.cmake#L53)

tnikolla, 2018-04-11:
Still the same errors.

graugans, 2018-04-11:
@BigBoot any ideas suggestions on this?

BigBoot, 2018-04-11:
For a quick fix, adding

`#pragma warning(disable:4996)`

to the top of the affected file.
should disable the warning/error message.

tnikolla, 2018-04-11:
I tried that too. Didn't work.

graugans, 2018-04-11:
Here is another hint from one of our dev team:

```
I just done a google check. This is actually a warning but as SDL checks = Yes (/sdl-)" Its showing as error. The solution is Turning off "SDL checks =No (/sdl-)". 

Following link address the similar issue 
https://github.com/xricht17/twpp/issues/5 
```

tnikolla, 2018-04-13:
Still no success.

Can you recreate the problem? If not I'll close this issue. It's more of a VS issue and does not belong here.

inbangsa, 2018-06-07:
@graugans @tnikolla

I have tried to reproduce the issue but I got similar warning Not an Error :

warning C4996: 'std::equal::_Unchecked
_iterators::_Deprecate': Call to 'std::equal' with parameters that may be unsafe - this call relies on the caller to check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNINGS. See documentation on how to use Visual C++ 'Checked Iterators' [C:\ifm3d\ifm3d-examples-master\build\ex-file_io.vcxproj]

One can get away with this Warning by adding -DD_SCL_SECURE_NO_WARNINGS=ON in the  cmake command

inbangsa, 2018-06-07:
@tpanzarella @graugans 
For Debug/Release build  We should add a  variable 
e.g CONFIG = --Debug  or  --Release  
and change the cmake build  flag as  
cmake --build . --clean-first --config %CONFIG% --target INSTALL
let me know if this is OK. I will update it to document

tnikolla, 2018-07-25:
Since some time now I am no longer involved with ifm3d. Should I close the issue?

commit 813320bf (2022-01-17): Merge branch '53-replace-usage-of-etc-sudoers-with-etc-sudoers-d-ifm' into 'o3r/main-next'

replace usage of /etc/sudoers

Closes #53

See merge request syntron/support/csr/ifm3d/ifm3d!96
closed 2018-07-25 by tnikolla