ifm/ifm3d #55: How to get frames from multiple cameras?

tnikolla, 2018-04-13:
I want to get multiple point clouds and then merge them.

How do I grab frames from multiple cameras? Can someone think of an easy example and add it in the repository?

tpanzarella, 2018-04-13:
You would instantiate a camera (and associated framegrabber and image buffer) for each camera, discriminating each by IP address. I don't have time to test this right now, but the code would look something like this (substitute your IP addresses as appropriate):

```
auto cam1 = ifm3d::Camera::MakeShared("192.168.0.69");
auto fg1 = std::make_shared<ifm3d::FrameGrabber>(cam1);
auto im1 = std::make_shared<ifm3d::ImageBuffer>();

auto cam2 = ifm3d::Camera::MakeShared("192.168.0.70");
auto fg2 = std::make_shared<ifm3d::FrameGrabber>(cam2);
auto im2 = std::make_shared<ifm3d::ImageBuffer>();

fg1->WaitForFrame(im1.get(), 1000)
fg2->WaitForFrame(im2.get(), 1000)

... etc ...
```
Use appropriate system programming techniques (and hardware setup of cameras) as necessary for your application to do things like acquire frames concurrently, etc. The above is the basic idea though.

tnikolla, 2018-04-17:
It works only for the 192.168.0.69 address. I used also the ifmVisionAssistant to check the cameras and their IP and it shows both cameras with the same 192.168.0.69 IP address. Can I discriminate using MAC addresses? 

In my settings two cameras (4 in the future) are connected via a switch (D-Link DGS 1008D) to the laptop (till the workstation arrives). I don't use multithreading.

tpanzarella, 2018-04-17:
You should give each camera a unique IP address, then use a technique similar to what I show above.

tnikolla, 2018-04-17:
How can I give each camera a unique IP? I am new to IP addresses and IFM3D repository.

graugans, 2018-04-17:
@tnikolla it is simple to assign a separate IP addresses for each camera. I'll suggest to use the following setup:

- camera 1: 192.168.0.70
- camera 2: 192.168.0.71
- (camera 3: 192.168.0.72)
- (camera 4: 192.168.0.70)

Starting with 192.168.0.70 will not conflict when adding a standard O3X/O3D to you set-up, because those are set to 192.168.0.69

To change the IP from command line you can insert the first camera and only the first camera and execute the following:

```
ifm3d --ip=192.168.0.69 dump | jq '.ifm3d.Net.StaticIPv4Address="192.168.0.70"' | ifm3d --ip=192.168.0.69 config
```
After this add the second camera to the network
```
ifm3d --ip=192.168.0.69 dump | jq '.ifm3d.Net.StaticIPv4Address="192.168.0.70"' | ifm3d --ip=192.168.0.69 config
```

tnikolla, 2018-04-17:
@graugans Can you please suggest something without jq. I'm having problems with it. Thx.

graugans, 2018-04-18:
Sure
```
$ ifm3d --ip=192.168.0.69 dump > o3x_1_config.json
$ cp  o3x_1_config.json  o3x_2_config.json
# edit both files with an editor of your choice
# save the files

# Power up the first camera
$ ifm3d --ip=192.168.0.69 config < o3x_1_config.json
# Power up the second one
$ ifm3d --ip=192.168.0.69 config < o3x_2_config.json
```
I hope I made no mistake this is out of my head, currently no O3X nearby

graugans, 2018-07-23:
@tnikolla can we close this issue? Could you please check the documentation provided by the [Pull Request #5](https://github.com/lovepark/ifm3d-examples/pull/5/files) of the ifm3d-examples for any missing information?

tnikolla, 2018-07-25:
Yes, please. I forgot to close it.

closed 2018-07-25 by tnikolla