ifm/ifm3d #56: No temperature and exposure time meta data

pfrydlewicz, 2018-04-19:
Hi,
running "ifm3d viewer" (version 0.8.2, windows binaries from Emanuel) I get following messages:

W0419 10:21:12.487709  8104 image_buffer_impl.hpp:794] **Checking for illu temp and exposure
 times skipped (cant trust extidx)**

According to the source code there is no meta information available in the point cloud data.
Does this have a negative effect when I am not interested in this data?

How can this be resolved?

O3X FW 1.0.56

graugans, 2018-04-26:
The binaries from @Arag24  are a older than the master branch you've might used. The Illu Temperature stuff was added in [0.8.3](https://github.com/lovepark/ifm3d/commit/0d1ab221346b7852ff54bc82248fbd965a181104) BTW your O3X FW is quite old.

Arag24, 2018-04-29:
@pfrydlewicz  : No problem to make binaries from 0.8.3 if needed (you know harder was PCL not ifm3d). Just let me know if you want. Even, if from what I could see, it would change when 0.9.0 become official.
And Paul (I'm Damien not Emanuel 😃 )
@graugans : More open question related to FW. How can we get altest one? and what about compatibility with M version ? Because if I could see swupdate command but I have no idea about firmware packages available.

graugans, 2018-05-02:
@Arag24 and @pfrydlewicz the latest O3X1xx firmware is available for download on the official [ifm website](https://www.ifm.com/ifmus/web/dualis-download.htm) 

# Compatibility of 1.0.122
- M03457
- M03551
- O3X100
- O3X101
- O3X110
- O3X171
- O3X173

#Compatibility of FW v1.0.62
- M03457
- M03551
- O3X100
- O3X101
- O3X110

pfrydlewicz, 2018-05-03:
Thanks @graugans, I checked recently but couldn't find the firmware. Now it's online.
@Arag24, yes probably you're right, the biggest thing were the dependencies.

Thanks to all

closed 2018-05-03 by pfrydlewicz