ifm/ifm3d #57: Change camera password

nayan27, 2018-04-23:
Hi,

Is it possible to modify the device's password with this project ?

I see that you can set "PasswordActivated" to true with the json configuration. But there is no field for the actual password.

Thanks,

graugans, 2018-04-25:
Hello @nayan27 this is currently not supported. But from my point of view it is not a big thing to add this feature. The camera already does provide an XML-RPC command to activate the password. This command is below the device class and called ``activatePassword`` this method expects a single string parameter the password. To disable the password the method ``disablePassword`` is available. The ``disablePassword`` is parameter less. Both commands require a call to the ``save`` method of the device class.  
I'll suggest adding a method ``void SetPassword(std::string password)`` to the [Camera](https://github.com/lovepark/ifm3d/blob/master/modules/camera/include/ifm3d/camera/camera.h#L62) class. If the password string is empty the password is disabled.
Anny comments @tpanzarella?

nayan27, 2018-04-25:
@graugans Great this solution looks good. Do you believe you would be able to implement it or would you want me to do it ? If you want me to do it, should it be WrapInEditSession ? Also do I have the right to write to this repo ?

thanks for quick answers !

graugans, 2018-04-26:
@nayan27 It would be highly appreciated to have code contributed. We can do this but it is not yet scheduled. For changing stuff in the device config you'll need an edit session. But I am not sure about the ``WrapInEditSession``

The best way for contribution is creating a personal fork, change the code there and create a pull request.

tpanzarella, 2018-04-26:
As @graugans mentions, if you want to do this @nayan27  please provide a PR and we will review. However, since this is pretty straightforward we can also implement this and roll it in to 0.9.0 which I hope to release w/in 1 - 2 weeks. I think the interface that @graugans suggests is good. It would be exposed as part of the public interface but under the hood it would use the `WrapInEditSession` function to have all the messiness of the session stuff handled for it seamlessly.

nayan27, 2018-04-26:
@tpanzarella Thanks for the quick reply guys! I can wait for 1-2 weeks for this feature. 

Also, Is there a way to update the firmware of the device with this new library ?

Thanks,

graugans, 2018-04-27:
@nayan27 the update feature is already part of the library. If you have build the tools then you can use the ``ifm3d``  tool for updates.

## Save current config

```
$ifm3d --ip=192.168.0.70 dump > ifm3d_config.json
```
## Firmware update
```
$ ifm3d --ip=192.168.0.70 reboot -r
$ sleep 20
$ ifm3d --ip=192.168.0.70 swupdate < O3X_Firmware_1.0.64.swu
```

## Apply configuration

```
$ cat  ifm3d_config.json | jq 'del(.ifm3d.Apps[0].Id)'  | ifm3d --ip=192.168.0.71 config
```
Or after merging #61 you can skip the jq part

```
$ ifm3d --ip=192.168.0.71 config < ifm3d_config.json
```

graugans, 2018-05-07:
@tpanzarella this morning Sachin and I had a small discussion about the password feature. We both agreed having the password stored in plain-text within the JSON is not a good solution. The idea was adding a new command to alter the password. We both committed  to this proposal:

```
# Activate the password
$ ifm3d --ip=192.168.0.69 password --new=TopSecret
# Change the password
$ ifm3d --ip=192.168.0.69 --password=TopSecret password --new=HackMe 
# Disable the password
ifm3d --ip=192.168.0.69 --password=HackMe password --disable
```

In general providing the passwords as command line arguments can lead to leaking passwords through the history or through process explorers like ps or top. A more secure way is reading the password from a file. But the original design is already providing the command line argument so we moved forward with this approach. 

Any thoughts on this @tpanzarella, @theseankelly ?

tpanzarella, 2018-05-07:
I do not have *strong* feelings about this either way. However, my initial reaction is that the password mechanism is inherently insecure. The connection to the camera is http, so a simple `tcpdump` will sniff the password in a few seconds should someone really want to gain access. To that end, an alternate to what is presented above, could be:

1. We do not serialize the password in the JSON. Ever.

2. "Best practices" usage of `ifm3d` could look like this:
```
$ ifm3d --ip=192.168.0.70 --password=`cat ~/.ifm3d_passwd | mydecryption-algo.sh` foo
```

3. Store an encrypted version of your password in `~/.ifm3d_passwd` and be sure to `chmod 600` the file.

Just another way of looking at this problem which requires less tooling on our part. If we really wanted to make a secure system, it would have to start camera-side.

graugans, 2018-05-07:
I agree on the general safety aspect. The idea with insertion of the password is a nice one. Maybe we should add this to the documentation as a hint. Plus the fact the password can be sniffed by tcpdump so better don't use this feature over an insecure connection.

graugans, 2018-05-07:
@tpanzarella  in General you do agree with the proposal?

tpanzarella, 2018-05-07:
It is not clear to me. I don't see the difference between having a `password` sub-command vs. just using a top-level `--password` argument. To me they are equivalent. So, the former would be more work and the latter would be less work. I'd go with the latter. Please help me understand better if I am missing the point.

theseankelly, 2018-05-07:
Is the main function of the password **actually** to protect/restrict access from a security perspective, or is it more of a hard switch to prevent accidental changes to camera configuration?

inbangsa, 2018-05-07:
@tpanzarella --password flag is used to pass the password to the device for opening edit-session. Whereas password sub-command can be used to set the password as well as disable the password from camera.

tpanzarella, 2018-05-07:
@inbangsa Got it. Thanks for clarifying. Then, yes, the approach seems fine.

tpanzarella, 2018-05-07:
@inbangsa @graugans One (minor) point, I'd name the subcommand `passwd` instead of `password`. Mainly because we already apply some consistency with the traditional Unix command names for other operations  like `ls`, `cp`, `rm`. Again, this is not a big deal, but, certainly would be my preference.

graugans, 2018-05-07:
@tpanzarella that’s easy to fix. It also separate the manipulation from providing the password.

graugans, 2018-05-07:
@theseankelly the basic idea was protecting against altering settings from unauthorized persons. We had ongoing discussions about providing a master password to unlock an protected device. What is a backdor which can be exploited. So the current approach is installing a firmware nukes the config and removes the password. On the camera the password is hashed like in your Unix box and only hashes are compared. So no extraction of passwords possible except there is an error in the hash algo

closed 2018-05-08 by tpanzarella