ifm/ifm3d #58: XMLRPC not found Windows employment

haydenhager, 2018-04-23:
When I try to build the ifm3d project I get this error. 
input:  c:\IFM3D\ifm3d\build>cmake --build . --clean-first --config Release --target INSTALL

ouput:
Build FAILED.

"C:\IFM3D\ifm3d\build\INSTALL.vcxproj" (default target) (1) ->
"C:\IFM3D\ifm3d\build\ALL_BUILD.vcxproj" (default target) (3) ->
"C:\IFM3D\ifm3d\build\modules\examples\ex-exposure_times.vcxproj" (default target) (4) ->
"C:\IFM3D\ifm3d\build\modules\camera\src\libifm3d_camera\ifm3d_camera_shared.vcxproj" (default target) (5) ->
(ClCompile target) ->
  C:\IFM3D\ifm3d\modules\camera\src\libifm3d_camera\camera_impl.hpp(33): fatal error C1083: Cannot open include file: 'xmlrpc-c/clien
t.hpp': No such file or directory [C:\IFM3D\ifm3d\build\modules\camera\src\libifm3d_camera\ifm3d_camera_shared.vcxproj]

Also, when I open this file in Visual Studio it shows a missing link to xmlrpc-c. Any suggestions on fixing this issue.

graugans, 2018-04-24:
Looks like you ran into the same problem which is already described in #54 May I also recommend to use [Markdown code blocks](https://help.github.com/articles/creating-and-highlighting-code-blocks/) for the error output.

haydenhager, 2018-04-25:
I deleted everything and started the build from scratch and the problem didn't reoccur.

commit 901b4140 (2022-01-24): Merge branch '58-deadsnakes-ppa-dropped-support-for-ubuntu-16-04' into 'o3r/main-next'

Remove 16.04 based docker images

Closes #58

See merge request syntron/support/csr/ifm3d/ifm3d!97
closed 2018-04-25 by haydenhager