ifm/ifm3d #63: Remove PCL dependency on windows build instructions

theseankelly, 2018-05-04:
As of 0.9.0, PCL should not be necessary to build ifm3d core. However, I was unable to complete the build instructions (as written) without having PCL fully built. This should be corrected in the build instructions.

theseankelly, 2018-06-25:
http://polarionsy.tt.de.ifm/polarion/#/project/152561/workitem?id=152561-5

graugans, 2018-09-19:
@theseankelly the Windows was updated within one of the later dev branches. https://github.com/lovepark/ifm3d/blob/master/doc/windows.md#prepare-dependent-tools Which describes the dependencies of each of the image modules. If this satisfies your request please feel free to close this issue. Otherwise please make a proposal how to fix this.

closed 2018-09-19 by theseankelly