ifm/ifm3d #68: O3X image rectification

madsherlock, 2018-05-09:
Hello,

On the O3D3, the boolean JSON parameters "EnableRectificationAmplitudeImage" and  "EnableRectificationDistanceImage" determine whether image rectification is applied on the sensor, using the internal intrinsic calibration.

I believe these parameters also determine whether the point cloud is rectified.

**Is there a similar feature on the O3X?**
Using the same JSON parameters as with the O3D leads to the following error upon ifm3d config:

`ifm3d error: [json.exception.type_error.302] type must be string, but is number`

I believe the point cloud (and the XYZ image) are rectified according to the intrinsic calibration on the O3X,
and as such there should be a direct correspondence between distance, amplitude and XYZ image indexes.
However, I am looking to do a marker based robot to camera calibration, and am in need of rectified amplitude images.

Best regards,
Mikael

graugans, 2018-05-09:
Unfortunately, there is no rectification of the distance or amplitude matrix available in the O3X. There seems to be way way to calculate it based on the unit vectors.

madsherlock, 2018-05-11:
Thanks.
About the unit vectors:
They are the link between 3D position and depth image. However, I don't know if this relation is to the distorted, or to the undistorted depth image. In any case, I don't see how I can use this for rectification of the amplitude image. Any suggestions?

I guess I will have to apply my own intrinsic calibration for now, and hope that some day either the rectified images or the calibration values will be available as they are on the O3D3xx #62 .

graugans, 2018-05-17:
@madsherlock  the guys from the system engineering have provided me some [annotated Matlab code](https://gist.github.com/graugans/2c045f04292fa79dd22659b75e566131). I hope this will help, additionally I'll create a PR with the generated PDF so you'll see the images if you do not have Matlab available.

pfrydlewicz, 2018-05-21:
Hello @madsherlock ,
as far as I know, the unit-vectors are referring to the distorted (original) image arrays.

madsherlock, 2018-05-23:
Wow, thanks! That's really helpful!
From that document, it appears the unit vectors contain all the intrinsic parameters of the camera, including the distortion parameters, and can be used to remove distortion from both the amplitude and the depth images.
It seems the method can be extended from the nearest neighbour method to using an interpolation scheme for improved results, similar to what is done in OpenCV's undistortion function.

After closer inspection of my data, I no longer believe the point cloud from the O3X is rectified; Please correct me if I am wrong in this.

It would seem the unit vectors are the only data from the O3X containing calibration information.
This should be enough though!

closed 2018-05-30 by madsherlock