ifm/ifm3d #71: Building on windows, problem with Eigen lib

pfrydlewicz, 2018-05-21:
The windows step-by-step guide is pretty nice.

However, when following through, I encounter a problem when trying to build the Eigen lib (v3.3.4).
Any ideas?
The project file is not created. Any idea of additional dependencies (BLAS or others)?

Thanks.
Paul

Here part of the output:
-----------------------------------------------------------------------------------
-- A library with BLAS API not found. Please specify library location.
-- LAPACK requires BLAS.
-- Could NOT find CHOLMOD (missing: CHOLMOD_INCLUDES CHOLMOD_LIBRARIES)
-- Could NOT find UMFPACK (missing: UMFPACK_INCLUDES UMFPACK_LIBRARIES)
-- Could NOT find SUPERLU (missing: SUPERLU_INCLUDES SUPERLU_LIBRARIES SUPERLU_VERSION_OK)

-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find BLAS (missing: BLAS_SEQ_LIBRARIES BLAS_LIBRARY_DIRS)
-- Could NOT find MPI_C (missing: MPI_C_LIBRARIES MPI_C_INCLUDE_PATH)
-- Could NOT find MPI_CXX (missing: MPI_CXX_LIBRARIES MPI_CXX_INCLUDE_PATH)
-- A version of Pastix has been found but pastix_nompi.h does not exist in the include dir
ectory. Because Eigen tests require a version without MPI, we disable the Pastix backend.
-- Could NOT find SPQR (missing: SPQR_INCLUDES SPQR_LIBRARIES)
-- Found unsuitable Qt version "" from NOTFOUND
globbed:
CMake Error at D:/Tools/Boost/share/cmake/boost/BoostConfig.cmake:15 (include):
  include could not find load file:

    C:/Projets/PCL/SuperBuild_boost_1_50/CMakeExternal/Install/Boost/share/boost-/cmake/Bo
ostConfig.cmake
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.9/Modules/FindBoost.cmake:236 (find_package)
  test/CMakeLists.txt:342 (find_package)

Boost  found.
-- Could NOT find GOOGLEHASH (missing: GOOGLEHASH_INCLUDES GOOGLEHASH_COMPILE)
-- Could NOT find ADOLC (missing: ADOLC_INCLUDES ADOLC_LIBRARIES)
-- Could NOT find MPFR (missing: MPFR_INCLUDES MPFR_LIBRARIES MPFR_VERSION_OK) (Required i
s at least version "2.3.0")

graugans, 2018-07-23:
We address this while working on the ``dev-0.9.3`` branch.

graugans, 2018-08-17:
Moved to ``dev-0.10.0``

inbangsa, 2018-08-20:
@pfrydlewicz  I am trying to reproduce this issue,  I got the log very similar to the your, and  the configuration is successful.. Basically Eigen is header only Library and Cmake is used for Installing only. 

Can you provide with your complete log 

Here is my build log: 
``` 
H:\eigen_Test\eigen\build>cmake -Ax64 -DCMAKE_INSTALL_PREFIX="H:\eigen_Test\inst
all" ..
-- Building for: Visual Studio 14 2015
-- The C compiler identification is MSVC 19.0.24215.1
-- The CXX compiler identification is MSVC 19.0.24215.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio
14.0/VC/bin/x86_amd64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio
14.0/VC/bin/x86_amd64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studi
o 14.0/VC/bin/x86_amd64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studi
o 14.0/VC/bin/x86_amd64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test standard_math_library_linked_to_automatically
-- Performing Test standard_math_library_linked_to_automatically - Success
-- Standard libraries to link to explicitly: none
-- Performing Test COMPILER_SUPPORT_OPENMP
-- Performing Test COMPILER_SUPPORT_OPENMP - Success
-- Performing Test EIGEN_COMPILER_SUPPORT_CPP11
-- Performing Test EIGEN_COMPILER_SUPPORT_CPP11 - Success
-- Found unsuitable Qt version "" from NOTFOUND
-- Performing Test EIGEN_COMPILER_SUPPORT_CXX11
-- Performing Test EIGEN_COMPILER_SUPPORT_CXX11 - Success
-- The Fortran compiler identification is unknown
-- A cache variable, namely BLAS_DIR, has been set to specify the install direct
ory of BLAS
-- Looking for BLAS -- mkl.h not found
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Looking for MKL BLAS: not found
-- Looking for Goto BLAS: not found
-- Looking for Open BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Atlas BLAS: not found
-- Looking for PhiPACK BLAS: not found
-- Looking for CXML BLAS: not found
-- Looking for DXML BLAS: not found
-- Looking for SunPerf BLAS: not found
-- Looking for SCSL BLAS: not found
-- Looking for SGIMATH BLAS: not found
-- Looking for IBM ESSL BLAS: not found
-- Looking for IBM ESSL MT BLAS: not found
-- Looking for ACML BLAS: not found
-- Looking for ACML BLAS: not found
-- Looking for ACML BLAS: not found
-- Looking for Apple BLAS: not found
-- Looking for NAS BLAS: not found
-- Looking for Generic BLAS: not found
-- Looking for Generic BLAS: not found
-- Looking for Generic BLAS: not found
-- Looking for Generic BLAS: not found
-- Looking for Generic BLAS: not found
CMake Warning at cmake/FindBLAS.cmake:1371 (message):
  BLA_VENDOR has been set to All but blas libraries could not be found or
  check of symbols failed.

[... 105 lines cut ...]
-- VSX:               Using architecture defaults
-- ARM NEON:          Using architecture defaults
-- ARMv8 NEON:        Using architecture defaults
-- S390X ZVECTOR:     Using architecture defaults
-- C++11:             OFF
-- SYCL:              OFF
-- CUDA:              OFF
--
CXX:               C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x8
6_amd64/cl.exe
 CXX_FLAGS:         /DWIN32 /D_WINDOWS /W4 /GR /EHsc /EHsc /wd4127 /wd4505 /wd47
14 /D_CRT_SECURE_NO_WARNINGS /D_SCL_SECURE_NO_WARNINGS
 Sparse lib flags:

-- ************************************************************
--
-- Configured Eigen 3.3.4
--
-- To build/run the unit tests, read this page:
--   http://eigen.tuxfamily.org/index.php?title=Tests
--
-- Configuring done
-- Generating done
-- Build files have been written to: H:/eigen_Test/eigen/build
```

inbangsa, 2019-02-05:
@pfrydlewicz  can we close this issue.

theseankelly, 2020-01-12:
Closing due to lack of activity; please reactivate as needed.

closed 2020-01-12 by theseankelly

echowang82, 2020-03-09:
> The windows step-by-step guide is pretty nice.
>
> However, when following through, I encounter a problem when trying to build the Eigen lib (v3.3.4).
> Any ideas?
> The project file is not created. Any [...]

I got the similar error when i tried to build Eigen lib (v3.3.7) . Have you solved this problem?  If yes, could you share your solution?