ifm/ifm3d #72: Max FPS

nayan27, 2018-06-04:
Hi there,

Using a O3D303 in low exposure type without temporal or spatial filter, with the continuous acquisition mode. In ifmAssistant, i get a framerate of 8-9.

But using a basic program based on the ifm3d library, i only get exactly 1 fps. Am i missing something ?

[image]

graugans, 2018-06-05:
Could you please be so kind and specify the software versions (ifm3d, firmware, Vision Assitant) you are using?

You can verify your results by using the ``ifm3d hz`` tool.

nayan27, 2018-06-05:
After looking at the ifm3d hz tool as you mentionned, i noticed that using the ByteBuffer instead of the ImageBuffer seems to solve the problem and FPS are now coming in way faster. Any idea on why ImageBuffer is slower ?

tpanzarella, 2018-06-06:
Please attach the full output of `ifm3d dump` so that we can reproduce your configuration and provide some insights into your issue.

nayan27, 2018-07-26:
Hi simply to let you know that using ByteBuffer did get rid of the delay problem and that it suits our application for now ! I believe you can close this issue if you wish ! Thanks

closed 2018-09-06 by tpanzarella