ifm/ifm3d #73: Session already active

nayan27, 2018-06-05:
Hi there,

I am using the ifm3d library and encounter some problem when the application crashes or when i kill it.

When attempting to retrieve the camera's configuration, i get a SEHException. When i open the camera with ifmAssistant, i cannot access the Application or Device tabs and get a Edit session already active error message. 

After a certain delay of about 10 minutes. The issue is gone (as if the edit session has timeout). Any way to work around this problem or maybe change the edit session's timeout ?

Thank you,

tpanzarella, 2018-06-06:
```
I am using the ifm3d library and encounter some problem when the application crashes or when i kill it.
```

I do not understand. What is “the application”? What exactly crashes? Can you post some code that is reproducible?

```
When attempting to retrieve the camera’s configuration, i get a SEHException.
```

What is an `SEHException`? This does not sound like an `ifm3d` construct to me. Can you post your `/tmp/pds.INFO` file?

From what I can deduce from your issue, I think your application is “crashing” with an “Edit session” open with the camera. Since only a single edit session is allowable on the camera, it would make sense that you need to wait until it times out to move forward. To fast track things, you could reboot your camera which will wipe out any open edit sessions. However, it sounds like you have a larger issue here if things are crashing. If you are experiencing instability in `ifm3d` please share a minimal code example which reproduces the issue so we can investigate.

nayan27, 2018-06-06:
Sorry for the lack of clarity in my issue. The application i am talking about is my own in which i use the library to communicate with the camera.

My issue is the following, Let's assume my application communicating with camera is closed at a moment when it has an edit session open with the camera. A few seconds later, the application is restarted and attempt to assign the camera's configuration (requiring an edit session), i will get an error.

Is there a way to theoritically work around this ? I am not experiencing instability with the library. It is working well and i think that behavior should be handled in my own code. I am simply asking if you already encountered that kind of situation and would have any idea on how to work around it beside simply waiting for the camera to timeout the edit session. Reboot could be worth giving a try although it takes some time as well.

Thank you,

tpanzarella, 2018-06-06:
> Sorry for the lack of clarity in my issue. The application i am talking about is my own in which i use the library to communicate with the camera.

No need to apologize; thanks for the clarity.

In the normal case of your application closing gracefully, this should be handled for you. If you are explicitly creating a session with `cam->RequestSession()` you *should* call `cam->CancelSession()` at some point to close the session. However, even if you do not, and you are using the camera smart pointer, when the reference count goes to zero, it will trigger the camera dtor whereby we explicitly cancel the session (should it be open) on your behalf.

If you have a minimal code example that produces the behavior you are looking to avoid, that would be helpful.

graugans, 2018-06-07:
@tpanzarella I guess  @nayan27 is working on a ``C#`` or ``.Net`` wrapper. I am not aware how this managed and unmanaged stuff works and what happens with the C++ code when the above application is *crashing* I can imagine the DTORs are not called.

During the design of the session handling we added the ability to inject an session id. The method ``requestSession`` provides two parameters one defining a session ID and the other one defining the password.

A possible solution could be to specify the Session ID to be used for creating the Camera object as an optional parameter. In this case the camera object can **reuse** the session and this will prevent you from running into the issue with the session already in use issue. A disadvantage I do see from this approach is the fact that in this **hijacked** scenario you might not know what is the logical state of the camera. In this case a ``cam->CancelSession()`` may would help to unwind any unwanted changes.

nayan27, 2018-07-23:
@graugans  i like your idea with the requestSession using an ID but the ifm3D library does not provide an interface to use this.

Also i will create another issue if you want me to but i find the fromJSON method to configure the device takes a lot of time of execute. I don't mind for the configuration but in the case where i simply want to change the TriggerMode to activate or deactivate the acquisition, the time is too long to change that simple parameter. I looked at the code for  SetTemporaryApplicationParameters but it seems to support only exposure and channel parameters.
Any idea ?

Thanks !

graugans, 2018-07-24:
@nayan27 we already know that the configuration takes longer than needed. This is due to the session handling within ifm3d. We are working on this within the next sprint.

For your issue one solution could be to add two applications and do an application switch to disable the camera.

nayan27, 2018-07-26:
@graugans Thank you for your answer about the configuration timing issue. That being said you haven't address the problem that the ifm3d library does not provide an interface to requestSession with an ID. Do you believe that could also be added soon ?

Thanks again!

nayan27, 2018-08-02:
Hi @graugans,

Don't mean to bother you but i would like to know if i could expect a fix for the configuration timing issue as well as a requestSession with an ID interface.

Thank you!

graugans, 2018-08-03:
@nayan27 currently we are working on the [dev-0.9.3](https://github.com/lovepark/ifm3d/tree/dev-0.9.3) release branch of the ifm3d which is mostly bugfixing and documentation improvements. The next sprint will include feature updates like the session ID and the speed improvements. So if you can give us the priorities we may can switch internally and provide you with some preview code. In any case it may worth to contact @GarrettPlace from the Sales team so we do have a better understanding of your application.

GarrettPlace, 2018-08-07:
Hi Nayan - Thank you for all of your efforts.  Christian and I spoke about the applications and needs.  A question from my side:  what is your most critical blocker at this time concerning the move to FW 1.20.790?

nayan27, 2018-08-07:
Hi Garrett,

What we need would be the possibility to call a requestSession using a specific ID so that we can prevent problems of session already openned on the device.

We would also need the configuration from JSON to take much less time than it actually does (about 3-4 seconds from my experience right now).

We still need to do some more testing but 1.20.790 looks faster in addition to the new features.

Thank you,

graugans, 2018-08-10:
Hi @nayan27 

we have prepare the reusing of the session. Please take a look at the following patch this should apply on the latest dev-0.9.3 branch. If we apply this to ``dev-0.9.3`` or ``dev-0.10.0`` or what ever release is the decision of @tpanzarella.

```patch
From 9b46d3f7450cb278f00c7887c7e0d76d56da6063 Mon Sep 17 00:00:00 2001
From: Bangadkar <Sachin.Bangadkar@ifm.com>
Date: Wed, 8 Aug 2018 17:29:51 +0530
Subject: [PATCH] Add the possibility for reusing the SessionID

The session ID can be passed by using the env variable
IFM3D_SESSION_ID. This will help during debugging sessions when
for example the parent process crashes. But keep in mind that
with great power comes great responsibility.

Reviewed-by: Christian Ege <christian.ege@ifm.com>
---
 doc/environment.md                                 | 42 +++++++++++++++
 modules/camera/src/libifm3d_camera/camera_impl.hpp | 63 +++++++++++++++++++---
 2 files changed, 98 insertions(+), 7 deletions(-)
 create mode 100644 doc/environment.md

diff --git a/doc/environment.md b/doc/environment.md
new file mode 100644
index 0000000..1f6e78d
--- /dev/null
+++ b/doc/environment.md
@@ -0,0 +1,42 @@
+ifm3d - Environment Variables
+=============================
+
+This document will help you to understand the different environment variables used with ifm3d library
+
+Following is the list of environment variables used with ifm3d.
+
+- [IFM3D_IP](#IFM3D_IP)
+- [IFM3D_SESSION_ID](#IFM3D_SESSION_ID)
+
+#### IFM3D_IP
+
+IFM3D_IP environment variable is used to set the device IP. If you fail to pass device IP while creating ifm3d::Camera class object
+then ifm3d will look for the environment variable IFM3D\_IP for the device IP value. If IFM3D\_IP is not set in the environment variable
+then ifm3d will try to connect to default IP(192.168.0.69)
+
+Setting the variable on Linux
+```
+$export IFM3D_IP=192.168.0.68
+```
+Setting the value on Windows
+```
+>set IFM3D_IP=192.168.0.68
+```
+
+#### IFM3D\_SESSION\_ID
+
+This is used to pass the user session ID to ifm3d. Typically the sensor generates a unique Session ID
+which is used to prevent multiple control instance from interfering each other. But during development sometimes
+it is intended to reuse a session. This can be the case during debugging sessions when it gets annoying to wait until
+the session automatically is closed. But listen to Uncle Ben, with great power comes great responsibility.
+Use this feature wisely. The session value must be a 32 char long hex value.
+
+Setting the variable on Linux
+```
+$export IFM3D_SESSION_ID=3c701f01bc3346c0940dac33f85ef8e6
+```
+Setting the value on Windows
+```
+>set IFM3D_SESSION_ID=3c701f01bc3346c0940dac33f85ef8e6
+```
+
diff --git a/modules/camera/src/libifm3d_camera/camera_impl.hpp b/modules/camera/src/libifm3d_camera/camera_impl.hpp
index f833284..810dc3d 100644
--- a/modules/camera/src/libifm3d_camera/camera_impl.hpp
+++ b/modules/camera/src/libifm3d_camera/camera_impl.hpp
@@ -600,13 +600,62 @@ ifm3d::Camera::Impl::ApplicationList()
 std::string
 ifm3d::Camera::Impl::RequestSession()
 {
-  xmlrpc_c::value_string val_str(
-    this->_XCallMain("requestSession",
-                     this->Password().c_str(),
-                     std::string("")));
-
-  this->SetSessionID(static_cast<std::string>(val_str));
-  this->Heartbeat(ifm3d::MAX_HEARTBEAT);
+  // Check if there is a session provided by the environment variable
+  // and this is a valid session ID
+  const std::string session = [](char* env){
+    if (env == nullptr)
+      {
+        return std::string("");
+      }
+      else
+      {
+        const std::string session = { env };
+        std::regex s_regex("[a-f0-9]{32}");
+        if ( std::regex_match(session,s_regex) == false )
+          {
+            LOG(ERROR) << "Invalid session provided by environment variable "
+                       << "IFM3D_SESSION_ID";
+            throw ifm3d::error_t( IFM3D_INVALID_SESSIONID );
+          }
+        return session;
+      }
+  }(std::getenv("IFM3D_SESSION_ID"));
+  try
+    {
+      xmlrpc_c::value_string val_str(
+        this->_XCallMain("requestSession",
+                         this->Password().c_str(),
+                         session));
+      this->SetSessionID(static_cast<std::string>(val_str));
+      this->Heartbeat(ifm3d::MAX_HEARTBEAT);
+    }
+  catch(const ifm3d::error_t  &ex)
+    {
+      const auto current_session = this->SessionID();
+      if ( (ex.code() == IFM3D_SESSION_ALREADY_ACTIVE) &&
+           (!session.empty()) )
+        {
+          // The device already has an active session
+          // check if we can reuse the session ID provided
+          // by the environment variable
+          this->SetSessionID(session);
+          try
+            {
+              LOG(INFO) << "We are using a recycled session ID: " << session;
+              this->Heartbeat(ifm3d::MAX_HEARTBEAT);
+            }
+          catch(...)
+            {
+              this->SetSessionID(current_session);
+              throw ex; // rethrow the IFM3D_SESSION_ALREADY_ACTIVE exception
+            }
+        }
+    else
+      {
+         this->SetSessionID(current_session);
+         throw ex;
+      }
+    }
   return this->SessionID();
 }

--
2.7.4

```

graugans, 2018-08-10:
@nayan27 During an internal discussion @tpanzarella was asking about the way you use the JSON interface. The smaller the scope the faster the changes are. As a reference this is how Tom solved the mute operation in the [ifm3d-ros nodlet](https://github.com/lovepark/ifm3d-ros/blob/d15e44d5f07fdb0158b1a8ecfd191b75bb509758/src/camera_nodelet.cpp#L298).

nayan27, 2018-08-10:
Hi @graugans, Thank you for having a look at this and providing a solution rapidly.
First of all, i must mention that some of our applications uses multiple IFM camera's and so using this method, if we set the session id in the environment variable, ALL ifm camera's used would be using that session ID... That is no problem to me but simply wanted to mention this here.

Also, you mention that the feature is mainly intended for debug purposes because we don't want to wait the whole timeout of the session before opening another one. Does that mean that the timeout is smaller or negligeable is release ? What would be the problem of always setting this new IFM3D_SESSION_ID environment variable other than allowing concurrent applications to take control of the same camera ?

Thank you,

Y

graugans, 2018-08-13:
@nayan27 having multiple cameras using the same session is not an issue. 

The idea of the session ID is to have a unique connection between the camera and the client. If we use the same session id for all connections it is easy to destroy an already established session between another client. This client will bail out with a XML-RPC error message. In the field this can lead to a lot of false positives and those will be hard to debug for the ifm support team. Because of this I personally would use this as a debug tool only and not within the field. This is a power-tool and used wrong can lead to unpredictable behavior. Maybe I am a bit too sensible on this.

BTW Tom already provided a more flexible implementation in the  [dev-0.10.0_session_management](https://github.com/lovepark/ifm3d/tree/dev-0.10.0_session_management) branch.

nayan27, 2018-08-13:
> Does that mean that the timeout is smaller or negligeable is release?

@graugans Well you mention the timeout for session is 300 seconds. Is it the same in Release mode than in debug ?
It makes sense to have this kind of long timeout in debug when a developper can be using breakpoints and all. But in release configuration, the timeout could be way less i believe. This is an humble opinion and i have no issues related to that.

@tpanzarella Do you know in which revision you intend to release this feature ? When would it be officially available ?

Thank you!

tpanzarella, 2018-08-13:
@nayan27 The changes have been merged on to `dev-0.10.0` so you can access it now. The relevant [commit](https://github.com/lovepark/ifm3d/commit/36e3802bf0581041c6b5acdc57ac54f70ba9a23f) is available for your review. We do not have a target date set for official release as @graugans and his team are pushing some new features as well. We can always push this up to suit a customer need. What is your required timing?

graugans, 2018-08-14:
@nayan27 At the moment there is no dynamic selection of the timeout depending on the debug or release mode. As far a I know we have not planned to change this. You are welcome to provide a patch implementing this.

In my humble opinion we can release ``dev-0.10.0`` when ever you want. We can push the open features to ``0.11.0`` they need a rebasing anyway. At the moment we have those features in the pipeline:

- Speed improvements for ``fromJSON``
-Support for intrinsic calibration Parameters

graugans, 2018-08-20:
@nayan27 there is a Pull-Request available #87 which need in some cases the half of the time for an import.

nayan27, 2018-08-21:
@graugans Have you set a date for the official release of dev-0.10.0 ? Is it soon ? Has the Pull-Request been completed ? Should i wait a little to have both features ?

Thank you,

tpanzarella, 2018-08-21:
@nayan27 I think @graugans says his PR is done. I will try to take a look at it sometime today and release it if all testing goes well.