ifm/ifm3d #79: Cmake error on Debian 9: set_target_properties with incorrect number of parameters

R1MCAI, 2018-06-29:
I try to compile the projects under Debian 9.0 on a virtual machine but get an error

==
$ cmake -DCMAKE_INSTALL_PREFIX=/usr ..
-- The CXX compiler idetification is GNU 6.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compiler features
-- Detecting CXX compiler features - done
CMake Error at src/libifm3d_camera/CMakeLists.txt:75 (set_target_properties):
   set_target_properties called with incorrrect number of arguments.

CMake Error at src/libifm3d_camera/CMakeLists.txt:86 (install):
   install TARGETS given no LIBRARY DESTINATION for shared library target "iflm3D_camera_shared".

(...)
===

How can I solve this?

graugans, 2018-07-05:
@R1MCAI sorry for the late reply, I am currently on the road. The issue you have reported is most probably related to some missing variable assignment. To debug this we do need the file `` src/libifm3d_camera/CMakeLists.txt`` which is located in your ``build`` folder.

R1MCAI, 2018-07-05:
[CMakeLists.txt](https://github.com/lovepark/ifm3d/files/2166434/CMakeLists.txt)
Attached the needed file.

graugans, 2018-07-05:
@R1MCAI Sorry, it was my fault. I assume the issue is the missing ``xmlrpc`` library. Can you please check if the following packages are installed?

```
apt-get update && apt-get install -y libboost-all-dev \
    jq \
    libcurl4-openssl-dev \
    libgtest-dev \
    libgoogle-glog-dev \
    libxmlrpc-c++8-dev \
    libopencv-dev \
    libpcl-dev \
    libproj-dev \
    build-essential \
    coreutils \
    cmake
```

R1MCAI, 2018-07-05:
missing packages are
libjq1 libonig4 jq libcurl4-openssl-dev libgflags-dev libgoogle-glog-dev 
several suggested packages (not listed here but included in the update

after installation: still the same error on cmake :-(

R1MCAI, 2018-07-06:
reading in details the messages I have more information
[image]
Hope it helps

graugans, 2018-07-07:
In this case I need to find a box to fire up a debian 9.0 installation. This might take a few days.

AlfTetzlaff, 2018-07-10:
I have installed the opencv-only build yesterday on Debian 9 and it worked fine. Tell me, if you want files or log output posted.

graugans, 2018-07-10:
Thanks, for the update. I'll reach out to you when I need more information.

R1MCAI, 2018-07-10:
I do have libopencv-dev installed, with several other packages.
Aptitude doesn't propose a opencv-only package.
@AlfTetzlaff : is the build you done a package or a specific sub version of the modules?

I've added python-opencv package (with aptitude) but no change, still error.

AlfTetzlaff, 2018-07-11:
You don't need any python packages. Check your $PATH environment variable. Mine looks like this:
``bash: /usr/local/bin:/usr/bin:/bin:``
Try to delete your build folder and then cmake & build again. I followed the instructions for the build that doesn't require PCL.
``cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_MODULE_OPENCV=ON -DBUILD_MODULE_IMAGE=OFF ..``
In addition to that, check where your googletest cmake files are located at. Mine were in /usr/src/googletest, but ifm3d expects them in /usr/src/gtest. So it could be, that you have to give the right path by setting
``-DGTEST_CMAKE_DIR=/usr/src/googletest``

R1MCAI, 2018-07-11:
my path is
/usr/local/bin;/usrt/bin

same status with proposed command
[image]

AlfTetzlaff, 2018-07-11:
Ok, it looks like you want to compile only the camera module, is that correct? I assumed you want to build the "whole thing" and gave instructions for that. If you want to build everything, you have to run the commands from the ifm3d base directory (which is three up from what you showed). There you have to create the build folder and so on...

I'm now guessing: If you want to compile certain models only, you still have to work from the ifm3d base directory. There, you have to run cmake, but with some -D... command. Type cmake -D and hit tab twice. It will show up ~100 options. Among those you find -DBUILD_MODULE_xyz, where you can choose the module to build. You then have to give for example -DBUILD_MODULE_CAMERA=ON.

R1MCAI, 2018-07-11:
I'd just like to "compile something related to this ifm3d system". I tried a first step in camera directory and got the error, that's why I stopped and asked for support.
From which folder should I launch the command to get a result?
[image]
I tried from ~/Workspaces/LovePark/ifm3d/build and I get a result
[image]
Now, I have to go on tests.
thanks for your efficient support.

AlfTetzlaff, 2018-07-11:
If your problem is solved, feel free to close this issue :)

R1MCAI, 2018-07-11:
it looks OK.
Now I'll look for time to test the functions and what we can do in our application.
(for information, the application is getting only a line or a column of images (X/Y/A/Confidence) instead of all the pixels, to reduce work load on the treatment machine. If ncessary I'll open another topic).

closed 2018-07-11 by R1MCAI

graugans, 2018-07-11:
@R1MCAI The warning above is due to typo. You have used ``MAKE_INSTALL_PREFIX`` the right one is ``CMAKE_INSTALL_PREFIX``. Sometime with CMake it helps to erase the build folder and restart from the beginning. I am not sure if I got your question about the line or column right. It seems you only get a line of data. and expect a whole image. You can check out the examples project how to deal with the data. https://github.com/lovepark/ifm3d-examples there for example is one cpp file which stores the amplitude data and the distance data so you can see it is a complete image: https://github.com/lovepark/ifm3d-examples/blob/master/ex-file_io.cpp

R1MCAI, 2018-07-11:
oh yes, little typo, great effet. Congratulations.
Anyway the command sudo cmake -DCMAKE_INSTALL_PREFIX=/usr .. in folder modules/camera/build gives always 
===========
CMake Error at src/libifm3d_camera/CMakeLists.txt:75 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at src/libifm3d_camera/CMakeLists.txt:86 (install):
  install TARGETS given no LIBRARY DESTINATION for shared library target
  "ifm3d_camera_shared".

CMake Warning at CMakeLists.txt:20 (message):
  Skipping libifm3d-camera unit tests!

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.7)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/utilisateur/Workspaces/LovePark/ifm3d/modules/camera/build/CMakeFiles/CMakeOutput.log".
See also "/home/utilisateur/Workspaces/LovePark/ifm3d/modules/camera/build/CMakeFiles/CMakeError.log".
===========

For my need, it is opposite: I always get the whole images from O3D302 when I'd need olny 1 column or 1 line. Objectif is to reduce load on TCP and on reception machine. Working with your "source" driver and firmware is a direction suggested by ifm.

AlfTetzlaff, 2018-07-11:
I think the error is due to the fact, that the CMakeLists.txt files in the module folders are automatically created. If I'm not totally wrong, they are just not intended to be used by you. (?)

Regarding you line problem: I think you can somehow set a region of interest (ROI) for the camera. If you set it to be a line, your problem should be solved. But I don't know how to set it, or if the driver supports it.

R1MCAI, 2018-07-12:
ROI will not give the information on each point of the line, almost an average value. I need the profil on the line: "simply" extracting a line from X/Y/Z images, if possible without getting all the data from the sensor, just what is needed.