ifm/ifm3d #82: CMake build failed with VS2015

Fibird, 2018-07-28:
There are mainly two errors: 
- error C2784: could not deduce template argument for ...
- error C2893: Failed to specialize function template ...

Fibird, 2018-07-29:
I solved this problem just now. It is the lower version of my MSVC 2015 compiler that caused this error. After I update my compiler from 14.0.23107.0 to 14.0.25420.10, ifm3d was built successfully. The "json" must need to be compiled by Microsoft Visual C++ 2015 / Build Tools 14.0.25123.0 (and possibly later) according to [this](https://github.com/nlohmann/json#supported-compilers) and ifm3d relys on it, so ifm3d also has to need Microsoft Visual C++ 2015 / Build Tools 14.0.25123.0. And I strongly suggest adding this requirement into ["windows.md"](https://github.com/lovepark/ifm3d/blob/dev-0.8.3/doc/windows.md).

inbangsa, 2019-02-05:
We already Added this dependencies to our  [Windows installation file](https://github.com/ifm/ifm3d/blob/master/doc/windows.md) 

Please close the issue.