ifm/ifm3d #84: Wait after camera IP Change

nayan27, 2018-08-09:
Hi,

I am encountering another little problem while attempting to change the camera's IP address.

The way i do it is simple: I modify the json elemt named "StaticIPv4Address" and load the new json configuration. 

Now here is the tricky part:
I then wait for the camera to come back by doing something very similar to the ifm3d_tools::swupdate_app::waitForUpdateFinish...
But instead of waiting for the status to be equal to FWU_STATUS_SUCCESS (3), i simply wait for the curl operation to return CURL_OK.

Now this works fine when the new ip address is in the same subnet as my ethernet adapter's. But if different, the curl operation seem to get stuck forever. (i guess the curl can't work if subnets are different).

So my real question is: When changing an ip address in ifmAssistant, how do you wait for the camera to be back online (even when new ip is not in same subnet) ?

Thank you again for the support guys!

nayan27, 2018-08-10:
In the same idea, i also seem to have a memory leak related to using curl to know whether the device is in recovery mode or not. I use the following code:
`bool CIFMSensor::CheckRecovery(char* ipaddress)
    {
    CURL* curl = curl_easy_init();
    CURLcode curlResult = CURLE_OK;

    std::string ip = ipaddress;
    std::string url = "http://" + ip + ":" + std::to_string(FWU_RECOVERY_PORT) + FWU_CHECK_RECOVERY_URL;
    curl_easy_setopt(curl, CURLOPT_URL, url.c_str());
    curl_easy_setopt(curl, CURLOPT_NOBODY, true);

    curlResult = curl_easy_perform(curl);
    long statusCode;
    curl_easy_getinfo(curl, CURLINFO_RESPONSE_CODE, &statusCode);

    if (curlResult != CURLE_OK || statusCode != 200)
        {
        curl_easy_cleanup(curl);
        return false;
        }

    curl_easy_cleanup(curl);

    return true;
    }
`
I am most probably not using this correctly. Would you have an idea on what is wrong or how i should do it? 

Thank you once again!

graugans, 2018-08-13:
@nayan27 To become a better understanding what You try to achieve, can you ping the camera from your host (PC) after changing the IP address? 

I do expect the following set-up on your side. A camera connected to your Ethernet port by a direct cable without switch or router in place. Now you do a change of the IP and want to check if the  camera has changed its IP but due to the fact that the cameras IP is now in a different sub-net there is no possible connection with curl or any other TCP/IP or ICMP based communication like ping isn't it? I am sorry this feature is considered confidential so I can not share this information on GitHub.

graugans, 2018-08-14:
@nayan27 As long as your switch is not a routing switch it doesn't matter. You are right, this feature is covered by the NDA. For the recovery mode, you can use the curl approach. You can take a look at [this code snippet](https://github.com/lovepark/ifm3d/blob/master/modules/tools/src/libifm3d_tools/swupdate_app.cpp#L298) for details.

nayan27, 2018-08-14:
@graugans Don't I need to do some curl cleanup ? I was already using those code snippet as you can seem in the code i snipped in a previous comment.

Also i don't feel you answered my question above:
```
I thought that changing an ip address on the camera would trigger some kind of internal re-initialization and so i thought that using a similar method as the waitForFinishUpdate would tell me when the camera is done re-configuring it's network settings.

But now that you mention it, effectively CURL can not be used if the subnet is diffrent. So what do you suggest ? What does ifmAssistant does to inquire the status of the camera ?
```

graugans, 2018-08-15:
@nayan27 you are right, the camera reinitializes when a new ip configuration is set. As described above you can not use any TCP/IP based mechanism like Curl which relies on routing of the host machine to communicate with the camera. You need as you mentioned the NDA code. My tip is use the MAC-Adress of the sensor and the NDA code. Sorry for being that vague on this topic, but as mentioned above this is rated confidential and I will not discuss this on a public GitHub issue tracker.

From the CURL[ online help](https://curl.haxx.se/libcurl/c/curl_easy_init.html):

This function must be the first function to call, and it returns a CURL easy handle that you must use as input to other functions in the easy interface. _This call_ **MUST** _have a corresponding call_ ``to_ curl_easy_cleanup`` _when the operation is complete_.

If you did not already call ``curl_global_init``, ``curl_easy_init`` does it automatically. **This may be lethal in multi-threaded cases,** since ``curl_global_init`` is not thread-safe, and it may result in resource problems because there is no corresponding cleanup.

**You are strongly advised to not allow this automatic behaviour**, by calling ``curl_global_init`` yourself properly. See the description in ``libcurl(3)`` of global environment requirements for details of how to use this function.

From the code snippet above and the online help I would assume that you need a call to [curl_global_cleanup](https://curl.haxx.se/libcurl/c/curl_global_cleanup.html) because the call to [curl_easy_init](https://curl.haxx.se/libcurl/c/curl_easy_init.html) does create a global initialization. May I recommend to put all the Curl stuff in a own class so you can use the destructor for the clean-up tasks.

Maybe this information from the Curl manual is also worth to mention:

``If you are initializing libcurl from a Windows DLL you should not initialize it from DllMain or a static initializer because Windows holds the loader lock during that time and it could cause a deadlock.  
``

Due to the fact that the xmlrpc library we do use also relies on CURLas backend you have to check when is the right time to do the global Curl initialization and cleanup. I think calling a ``curl_global_cleanup`` after a check of the recovery may cause an unstable xmlrpc communication.

nayan27, 2018-08-21:
Thank you for the advice, i will have a look at all this!

theseankelly, 2019-10-09:
Closing out old issues, feel free to reactivate if you're still having trouble with this!

commit f2f0ffeb (2023-06-12): Merge branch '84-create-a-pluggable-logging-interface-and-switch-the-default-logging-to-something-more' into 'main'

Remove glog and create custom logging facility

Closes #84

See merge request syntron/support/csr/ifm3d/ifm3d!254
closed 2019-10-09 by theseankelly