ifm/ifm3d #88: XMLRPC Timeout

Semih67, 2018-08-20:
When I run catkin_make_isolated I get these errors:

e/nvidia/catkin_ws/build_isolated/ifm3d'
-- ifm3d found component: camera
-- ifm3d found component: framegrabber
CMake Warning at /usr/lib/cmake/ifm3d-0.9.0/ifm3d-config.cmake:44 (message):
  ifm3d could not find component: image
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

CMake Error at CMakeLists.txt:4 (find_package):
  Found package configuration file:

    /usr/lib/cmake/ifm3d-0.9.0/ifm3d-config.cmake

  but it set ifm3d_FOUND to FALSE so package "ifm3d" is considered to be NOT
  FOUND.

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/catkin_ws/build_isolated/ifm3d/CMakeFiles/CMakeOutput.log".
<== Failed to process package 'ifm3d': 
  Command '['/home/nvidia/catkin_ws/devel_isolated/gpio_control/env.sh', 'cmake', '/home/nvidia/catkin_ws/src/ifm3d-ros', '-DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel_isolated/ifm3d', '-DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/nvidia/catkin_ws/build_isolated/ifm3d && /home/nvidia/catkin_ws/devel_isolated/gpio_control/env.sh cmake /home/nvidia/catkin_ws/src/ifm3d-ros -DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel_isolated/ifm3d -DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install_isolated -G 'Unix Makefiles'

Command failed, exiting.
catkin_make_isolated  44,82s user 24,52s system 140% cpu 49,351 total

And in ~/dev/ifm3d/build I'm running make check and get these errors

[ RUN      ] Image.MoveCtor
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.MoveCtor (1035 ms)
[ RUN      ] Image.MoveAssignmentOperator
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.MoveAssignmentOperator (2032 ms)
[ RUN      ] Image.CopyCtor
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.CopyCtor (2022 ms)
[ RUN      ] Image.CopyAssignmentOperator
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.CopyAssignmentOperator (2037 ms)
[ RUN      ] Image.References
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.References (2031 ms)
[ RUN      ] Image.CloudMechanics
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.CloudMechanics (2025 ms)
[ RUN      ] Image.XYZImage
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.XYZImage (2026 ms)
[ RUN      ] Image.ComputeCartesian
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.ComputeCartesian (2034 ms)
[ RUN      ] Image.TimeStamp
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.TimeStamp (1019 ms)
[ RUN      ] Image.IlluTemp
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.IlluTemp (1017 ms)
[----------] 10 tests from Image (17281 ms total)

[----------] Global test environment tear-down
[==========] 10 tests from 1 test case ran. (17281 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 10 tests, listed below:
[  FAILED  ] Image.MoveCtor
[  FAILED  ] Image.MoveAssignmentOperator
[  FAILED  ] Image.CopyCtor
[  FAILED  ] Image.CopyAssignmentOperator
[  FAILED  ] Image.References
[  FAILED  ] Image.CloudMechanics
[  FAILED  ] Image.XYZImage
[  FAILED  ] Image.ComputeCartesian
[  FAILED  ] Image.TimeStamp
[  FAILED  ] Image.IlluTemp

10 FAILED TESTS
modules/image/test/CMakeFiles/check_image.dir/build.make:57: recipe for target 'modules/image/test/CMakeFiles/check_image' failed
make[3]: *** [modules/image/test/CMakeFiles/check_image] Error 1
CMakeFiles/Makefile2:751: recipe for target 'modules/image/test/CMakeFiles/check_image.dir/all' failed
make[2]: *** [modules/image/test/CMakeFiles/check_image.dir/all] Error 2
CMakeFiles/Makefile2:76: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:184: recipe for target 'check' failed
make: *** [check] Error 2

Do you guys understand my problem?

graugans, 2018-08-20:
@Semih67 it looks like you are using an version ``0.9.0`` of the ifm3d lib. May I recommend an update? It also looks like a mixture between two issues. 

The issue related to ifm3d is the XML-RPC timeout. For this I recommend to follow the [troubleshooting guide](https://github.com/lovepark/ifm3d/blob/master/doc/troubleshoot.md) we've introduced in the last release.

The second issue with the catkin build seems to be related to the configuration used during the ifm3d build. It looks like you have disabled the ifm3d image module.

Could you or @tpanzarella please edit the issue and format the output with 
    ```
    // ...
   // code
   //....
   ```

This would make my live easier in reading the output and finding the issue.

Semih67, 2018-08-20:
I also get an error if i try to update.

And how can I enable the ifm3d image module?

graugans, 2018-08-20:
@Semih67 what kind of error do you get?

Semih67, 2018-08-20:
I can't look it up now which kind of error it was. I can write it tomorrow. But if I typed this:` sudo apt-get update && apt-get install -y libboost-all-dev git jq libcurl4-openssl-dev \
                                            libgtest-dev libgoogle-glog-dev  \
                                            libxmlrpc-c++8-dev libopencv-dev \
                                            libpcl-dev libproj-dev \
                                            build-essential coreutils cmake`

I got the error. And if I removed the update everything went fine

And again the question how can I enable the ifm3d image?

Semih67, 2018-08-20:
And I also pinged the IP and it was an unreachable host

graugans, 2018-08-21:
@Semih67 I would prefer to have solved one problem after the other. For me it looks like you have issues with your network configuration. This can be caused for example on issues with cooperate proxy like described in the [troubleshooting](https://github.com/lovepark/ifm3d/blob/master/doc/troubleshoot.md) guide. To check your current network configuration you can use the ``ip`` tool or ``ifconfig`` depending on your set-up the ip address of your camera can be checked with the ifm Vision Assistant under Windows or with the avahi tool as described in the  [troubleshooting](https://github.com/lovepark/ifm3d/blob/master/doc/troubleshoot.md) guide.

```
deegech @ ifmld0002 ~
└─ $ ▶ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp8s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:13:3b:0f:d7:df brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.1/24 brd 192.168.0.255 scope global enp8s0
       valid_lft forever preferred_lft forever
    inet6 fe80::ec82:6388:e60c:ef4b/64 scope link 
       valid_lft forever preferred_lft forever
3: eno1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 6c:0b:84:65:5d:c7 brd ff:ff:ff:ff:ff:ff
    inet 172.25.121.103/24 brd 172.25.121.255 scope global dynamic eno1
       valid_lft 688540sec preferred_lft 688540sec
    inet6 fe80::a685:f398:9711:a735/64 scope link 
       valid_lft forever preferred_lft forever
4: docker0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default 
    link/ether 02:42:8b:ff:83:74 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
    inet6 fe80::42:8bff:feff:8374/64 scope link 
       valid_lft forever preferred_lft forever
6: vethc39edc5@if5: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master docker0 state UP group default 
    link/ether 06:b4:19:d0:07:15 brd ff:ff:ff:ff:ff:ff link-netnsid 0
    inet6 fe80::4b4:19ff:fed0:715/64 scope link 
       valid_lft forever preferred_lft forever
```
or by ifconfig
```
deegech @ ifmld0002 ~
└─ $ ▶ ifconfig 
docker0   Link encap:Ethernet  HWaddr 02:42:8b:ff:83:74  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:8bff:feff:8374/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:16 errors:0 dropped:0 overruns:0 frame:0
          TX packets:51 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1589 (1.5 KB)  TX bytes:6237 (6.2 KB)

eno1      Link encap:Ethernet  HWaddr 6c:0b:84:65:5d:c7  
          inet addr:172.25.121.103  Bcast:172.25.121.255  Mask:255.255.255.0
          inet6 addr: fe80::a685:f398:9711:a735/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:163292 errors:0 dropped:0 overruns:0 frame:0
          TX packets:45946 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:204449393 (204.4 MB)  TX bytes:5002109 (5.0 MB)
          Interrupt:20 Memory:fb200000-fb220000 

enp8s0    Link encap:Ethernet  HWaddr 00:13:3b:0f:d7:df  
          inet addr:192.168.0.1  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::ec82:6388:e60c:ef4b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:110 errors:0 dropped:0 overruns:0 frame:0
          TX packets:82 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:12270 (12.2 KB)  TX bytes:9408 (9.4 KB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:47895 errors:0 dropped:0 overruns:0 frame:0
          TX packets:47895 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:150454460 (150.4 MB)  TX bytes:150454460 (150.4 MB)

vethc39edc5 Link encap:Ethernet  HWaddr 06:b4:19:d0:07:15  
          inet6 addr: fe80::4b4:19ff:fed0:715/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:16 errors:0 dropped:0 overruns:0 frame:0
          TX packets:82 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1813 (1.8 KB)  TX bytes:9767 (9.7 KB)
```

The image module is activated by default. Please follow the detailed installation guidelines for [ifm3d-ros](https://github.com/lovepark/ifm3d-ros/blob/master/doc/building.md) for details.

It seems you are located in Germany if it is easier for you we also can arrange a phone call in German.

Semih67, 2018-08-21:
It would be nice to have a phone call but first I want to try it by your help.
The first problem I get is the update error: `AppStream system cache was updated, but problems were found: Metadata files have errors: /var/cache/app-info/xmls/fwupd.xml
Reading package lists... Done
E: Problem executing scripts APT::Update::Post-Invoke-Success 'if /usr/bin/test -w /var/cache/app-info -a -e /usr/bin/appstreamcli; then appstreamcli refresh-cache > /dev/null; fi'
E: Sub-process returned an error code
sudo apt-get update  7,05s user 2,20s system 92% cpu 10,042 total`

Maybe if I solve this everything will work.

Semih67, 2018-08-21:
Ok after I upgraded the packages I could update. So the problem is solved. But when I type in ifconfig I get in eth0 a mac address

Semih67, 2018-08-21:
But nevertheless I should be able to compile the package. But I can't

graugans, 2018-08-21:
@Semih67 to speed up you can send me an e-mail to christian.ege@ifm.com with your phone number so I can call you.

Semih67, 2018-08-21:
I'm at work now could you not write here what I have to do?

Semih67, 2018-08-21:
Just write me what I have to check

graugans, 2018-08-21:
If ifconfig only shows a MAC address but no IP address I do assume you have only one Ethernet port which is configured to DHCP. If you plug-in the camera to this port you will not receive an ip address on this port. To connect to the camera you can try to check if this command can find the camera:

```
avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation
```

This may need the installation of ``avahi-utils`` on ubuntu.

graugans, 2018-08-21:
This command above gives in my case the following result:
```
deegech @ ifmld0002 ~
└─ $ ▶ avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation
=;enp8s0;IPv4;O3D303-40-1C-55\032\09100\05802\05801\05840\0581c\05855\093;Workstation;local;O3D303-40-1C-55.local;192.168.0.70;9;
=;enp8s0;IPv4;O3D3XX-40-8D-BD\032\09100\05802\05801\05840\0588d\058bd\093;Workstation;local;O3D3XX-40-8D-BD.local;192.168.0.81;9;
```
Which will allow me to ping the camera with it's local name:

```
ping O3D303-40-1C-55.local
```

Semih67, 2018-08-21:
And if I use your command then it stops at grep: Videos ... and it can't locate the camera

graugans, 2018-08-21:
May I recommend to get another Ethernet interface, for example an USB one, to have the Camera plus regular Internet connection or by adding a Ethernet Switch. There are plenty of how-to how to change the ip address of your host out on the internet. But be careful, this maybe cuts you off from the local office net and will prevent you from internet access.

Semih67, 2018-08-21:
I'm using a switch now and pinging 192.168.0.69 but it says the same that the host is unreachable

Semih67, 2018-08-21:
And do you know how to change the ip in the bashrc file?

graugans, 2018-08-21:
So the reachability of the camera heavily depends on your network set-up for example from my output above you can see multiple networks. Typically your office network is different from the ``19.168.0.x`` network to solve this you can talk to the administrators of your network to find a free IP address or they maybe recommend to add a new network interface. Your PC and the camera need to be in the same network to talk to each other. To debug this the output of the ``ifconfig`` command would be helpful.

Semih67, 2018-08-21:
Ok it's working now, I can ping the camera. But I still get the error if I try to compile it. It still says:

´but it set ifm3d_FOUND to FALSE so package "ifm3d" is considered to be NOT
FOUND.

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/catkin_ws/build_isolated/ifm3d/CMakeFiles/CMakeOutput.log".
<== Failed to process package 'ifm3d':
Command '['/home/nvidia/catkin_ws/devel_isolated/gpio_control/env.sh', 'cmake', '/home/nvidia/catkin_ws/src/ifm3d-ros', '-DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel_isolated/ifm3d', '-DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/nvidia/catkin_ws/build_isolated/ifm3d && /home/nvidia/catkin_ws/devel_isolated/gpio_control/env.sh cmake /home/nvidia/catkin_ws/src/ifm3d-ros -DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel_isolated/ifm3d -DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install_isolated -G 'Unix Makefiles'

Command failed, exiting.
catkin_make_isolated 44,82s user 24,52s system 140% cpu 49,351 total`

graugans, 2018-08-21:
To address this you have to recompile the ifm3d library. It seems that the ``image`` module was disabled when it was compiled the first time.  A full installation guide can be found [here](https://github.com/lovepark/ifm3d#installing-the-software) or [here](https://github.com/lovepark/ifm3d-ros/blob/master/doc/building.md) for using ifm3d with ros the later one is the recommended one.

graugans, 2018-08-21:
BTW the image module is [default activated](https://github.com/lovepark/ifm3d/blame/master/CMakeLists.txt#L20) since two years. Most probably it was disabled building the first time due to issues with PCL. May I ask for which Ubuntu version and platform you are building? I ask this because it looks like you build for an RaspberryPi because of ``gpio_control`` which in the opposite makes no sense because the user name used is ``nvidia``. Are you trying to cross compile the library or is it native compilation?

Semih67, 2018-08-21:
I try to recompile it but I get errors:

if I do make check:
``` 
[  FAILED  ] Image.MoveCtor (1891 ms)
 
[  FAILED  ] Image.ComputeCartesian (2095 ms)
 
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] Image.MoveCtor
[  FAILED  ] Image.ComputeCartesian
 
 2 FAILED TESTS
modules/image/test/CMakeFiles/check_image.dir/build.make:57: recipe for target 'modules/image/test/CMakeFiles/check_image' failed
make[3]: *** [modules/image/test/CMakeFiles/check_image] Error 1
CMakeFiles/Makefile2:753: recipe for target 'modules/image/test/CMakeFiles/check_image.dir/all' failed
make[2]: *** [modules/image/test/CMakeFiles/check_image.dir/all] Error 2
CMakeFiles/Makefile2:78: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:184: recipe for target 'check' failed
make: *** [check] Error 2
make check  3,42s user 2,10s system 5% cpu 1:36,47 total```
 
 
And in make package everything is fine but if I run make repackage:
 
  ```File "./ifm3d-dpkg-deps.py", line 226, in main
    retval = dpkg.run()
  File "./ifm3d-dpkg-deps.py", line 70, in run
    self.deb_deps_ = self.generate_deb_deps()
  File "./ifm3d-dpkg-deps.py", line 151, in generate_deb_deps
    out = subprocess.check_output(["dpkg", "-s", deb])
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['dpkg', '-s', 'diversion by notify-osd to']' returned non-zero exit status 2
CMakeFiles/repackage.dir/build.make:57: recipe for target 'CMakeFiles/repackage' failed
make[3]: *** [CMakeFiles/repackage] Error 1
CMakeFiles/Makefile2:103: recipe for target 'CMakeFiles/repackage.dir/all' failed
make[2]: *** [CMakeFiles/repackage.dir/all] Error 2
CMakeFiles/Makefile2:110: recipe for target 'CMakeFiles/repackage.dir/rule' failed
make[1]: *** [CMakeFiles/repackage.dir/rule] Error 2
Makefile:197: recipe for target 'repackage' failed
make: *** [repackage] Error 2
make repackage  16,42s user 5,24s system 99% cpu 21,874 total```

Semih67, 2018-08-21:
should I remove the ifm3d and clone it again? Would it solve the problem?

Semih67, 2018-08-21:
It's 16.04 xenial. I'm working on a jetson board and try to compile it normally

Semih67, 2018-08-21:
How is it possible that I'm building it for the raspberry pi. I just did what in the introduction was listed

Semih67, 2018-08-21:
Yes the image module is activate in the cmake file but should it not be activated?

Semih67, 2018-08-21:
I made one more make check and this time I got only one failed test: FrameGrabber.ByteBufferMoveAssignmentOperator
I don't know why the errors are changing

graugans, 2018-08-21:
@Semih67 the ``FrameGrabber.ByteBufferMoveAssignmentOperator`` failed test is strange. To have a better way of diagnostics detailed error output would definitely help to understand the root cause. There are two checks within the test. Maybe the first failed due to the network problems you had before.

Semih67, 2018-08-21:
It's working now. I could compile it with no errors. I don't know why but thank you for your help

graugans, 2018-08-21:
@Semih67 the ``make repackage`` call is only needed if you intend to install the debs on a different installation. For me this python script is a hack to mitigate issues caused by CPack. Maybe @tpanzarella can explain this in detail.

Semih67, 2018-08-21:
And now I get errors if I try to launch it. I cloned ifm3d-ros in my src directory and compiled everything but now if I type: roslaunch ifm3d camera.launch I get these errors:

```
[ERROR] [1534851498.022008330]: Failed to load nodelet [/ifm3d/camera] of type [ifm3d_ros/camera_nodelet] even after refreshing the cache: Could not find library corresponding to plugin ifm3d_ros/camera_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1534851498.022143626]: The error before refreshing the cache was: Could not find library corresponding to plugin ifm3d_ros/camera_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1534851498.022549737]: Failed to load nodelet '/ifm3d/camera` of type `ifm3d_ros/camera_nodelet` to manager `camera_standalone_nodelet'
[ifm3d/camera-2] process has died [pid 19378, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load ifm3d_ros/camera_nodelet camera_standalone_nodelet __name:=camera __log:=/home/nvidia/.ros/log/33e688b4-a519-11e8-981f-c93c23426fc0/ifm3d-camera-2.log].
log file: /home/nvidia/.ros/log/33e688b4-a519-11e8-981f-c93c23426fc0/ifm3d-camera-2*.log```

GarrettPlace, 2018-08-21:
Hi Semih67 - Thank you for all of your time.  Would it make sense to arrange a 1 hour block of time to discuss the current status and what may be required moving forward?  The may help accelerate the integration.  I can arrange the meeting with the appropriate people (in the appropriate Countries) is this will help.  Please contact me at garrett.place@ifm.com to discuss.

graugans, 2018-08-21:
@Semih67 I fully agree with @GarrettPlace we should spend that time to accelerate this. What ever _real time_ channel you prefer. Meanwhile I can imagine due to the rebuild of the ifm3d library you may have different versions of the ifm3d and the ifm3d-ros nodlet. But I am not an expert when it comes to the ROS code.

tpanzarella, 2018-08-22:
@graugans @Semih67 The `make repackage` target is indeed useful for creating packages to be deployed to other machines. However, it is also useful if you are just installing locally as it updates the debian control file to express proper dependencies on other packages, which, would prevent any problems should you attempt to uninstall a package that `ifm3d` depends upon while `ifm3d` is still installed -- `dpkg` will prevent that from happening.

CMake's CPack extension provides the [CPACK_DEBIAN_PACKAGE_SHLIBDEPS](https://cmake.org/cmake/help/v3.6/module/CPackDeb.html#variable:CPACK_DEBIAN_PACKAGE_SHLIBDEPS) variable that turns on a wrapper to [dpkg-shlibdeps](http://man7.org/linux/man-pages/man1/dpkg-shlibdeps.1.html) for generating proper dependencies in your debian control file. However, when I was building the packaging infrastructure, the cpack machinery fell short, mostly as a result of how we modularize `ifm3d`. To that end, I wrote [a python script](https://github.com/lovepark/ifm3d/blob/master/cmake/utils/ifm3d-dpkg-deps.py.in) that basically does the same thing (i.e., by reading the ifm3d-generated elf files ... shared objects only) but exploits knowledge of `ifm3d` *a priori*. So.... I'd agree with @graugans that this is a "hack" but I would not consider it "hacky". It gets the job done. If you are having a specific issue with this Python script, please give me the details and I will look into it when I get some time.

Bottom-line, I personally recommend installing via `dpkg` but only after running `make repackage`. TIFWIW.

tpanzarella, 2018-08-22:
@Semih67 You state:

> And now I get errors if I try to launch it. I cloned ifm3d-ros in my src directory and compiled everything but now if I type: roslaunch ifm3d camera.launch I get these errors

Did you compile *and* install? I recommend you follow [these instructions](https://github.com/lovepark/ifm3d-ros/blob/master/doc/building.md) as closely as you can. The error you show seems to indicate that ROS pluginlib cannot load the .so that implements the camera nodelet. This would be indicative of it not being *installed* properly which is orthogonal to *compiling* it.

Semih67, 2018-08-27:
Hey thank you for your help, I started to install ifm3d again and I want to do it step by step now. What I have done so now is that I cloned the link for ifm3d in catkin_ws/src . So this is the step where I'm unsure, because in your instructions you have created a build folder. And I don't know where you do this, in which folder. I cloned it to the src folder and tried to compile it and got the error that it could't find some packages: ```

CMake Error at CMakeLists.txt:4 (find_package):
  Could not find a package configuration file provided by "ifm3d" (requested
  version 0.9.0) with any of the following names:
    ifm3dConfig.cmake
    ifm3d-config.cmake
  Add the installation prefix of "ifm3d" to CMAKE_PREFIX_PATH or set
  "ifm3d_DIR" to a directory containing one of the above files.  If "ifm3d"
  provides a separate development package or SDK, be sure it has been
  installed.
```

So I think my last mistake was that I cloned it in the wrong folder am I right?

inbangsa, 2018-08-27:
@Semih67 ifm3d-ros is a wrapper over ifm3d as a ros-interface. Please build and install ifm3d as per [ifm3d build](https://github.com/lovepark/ifm3d/blob/master/README.md) instructions. You might have missed 

```
$ sudo make install
```

As per the information provided by you in above comment it seems ifm3d installation is missing from your system. 

If library is not installed in standard paths.. use -DCMAKE_PREFIX_PATH=<installation path> with catkin_make for building ifm3d-ros.

Semih67, 2018-08-27:
I'm following the instructions but it's unclear for me where I have to create the build folder. I did it in catkin_ws/src/ifm3d and followed the instructions, but if I try to compile my packages its not recognizing ifm3d even if I did sudo make install

Semih67, 2018-08-27:
Ohh I get now it's in catkin_ws/build . I was confused because it's already built in ros

inbangsa, 2018-08-27:
@Semih67 Are you trying to build [ifm3d](https://github.com/lovepark/ifm3d) or [ifm3d-ros](https://github.com/lovepark/ifm3d-ros) with catkin_make.. 

Ifm3d-ros depends on ifm3d and as per instruction you need clone  both. Then First install ifm3d with [this](https://github.com/lovepark/ifm3d/blob/master/README.md) and the ifm3d-ros [this](https://github.com/lovepark/ifm3d-ros/blob/master/doc/building.md). 

Ifm3d-ros instruction also describe how to install ifm3d lib..

Semih67, 2018-08-27:
Ok I have done this and could compile it after I installed ifm3d-ros. But now I can't launch it it says: [camera.launch] is neither a launch file in package [ifm3d] nor is [ifm3d] a launch file name

Semih67, 2018-08-27:
I didn't do this step: ```ln -s ~/dev/ifm3d-ros ifm3d```
When I did this earlier I got an error. So is it necessary to do this? And these too: ´´´$ catkin_make run_tests
$ catkin_make -DCMAKE_INSTALL_PREFIX=${LPR_ROS}/ifm3d install```
I have google cartographer in my src folder so I have to compile it with catkin_make_isolated but it's not running with the isolated

inbangsa, 2018-08-27:
Try to use source command

In a seprate terminal restart roscore 

And then in another terminal try this:
source ${LPR_ROS}/install/setup.bash 

As per the instructions this is already added to .bashrc file..

Semih67, 2018-08-27:
This is not working: source ${LPR_ROS}/install/setup.bash I have also tried it with source ${LPR_ROS}/opt/ros/kinetic/setup.bash but the same. I also used the ordinary source command but no success: source /opt/ros/kinetic/setup.bash

inbangsa, 2018-08-27:
$ catkin_make -DCMAKE_INSTALL_PREFIX=${LPR_ROS}/ifm3d install
This command will install ifm3d ros package and needed

And then source command must be

 $ Source {LPR_ROS}/ifm3d/install/setup.bash

Semih67, 2018-08-28:
When I run this command 
$ catkin_make -DCMAKE_INSTALL_PREFIX=${LPR_ROS}/ifm3d install 
I get this error: -- Install Configuration: ""
CMake Error at cmake_install.cmake:38 (file):
file problem creating directory: /ifm3d
Invoking "make install -j4 -l4" failed

My src folder looks like this: ifm3d ifm3d-ros CMakeLists.txt
And in the ifm3d folder I've the build folder.

Semih67, 2018-08-28:
This is my setup.bash: 
``` #!/usr/bin/env bash
# generated from catkin/cmake/templates/setup.bash.in
CATKIN_SHELL=bash
# source setup.sh from same directory as this file
_CATKIN_SETUP_DIR=$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" > /dev/null && pwd)
. "$_CATKIN_SETUP_DIR/setup.sh" ```

Semih67, 2018-08-28:
Ok I solved the problem. The problem was that my install folder was in catkin_ws. So this ``` catkin_make -DCMAKE_INSTALL_PREFIX=${LPR_ROS}/ifm3d install ``` could never run with success, because it's the wrong path. I just ran the command catkin_make install and it worked. This ``` $ Source {LPR_ROS}/ifm3d/install/setup.bash ``` is also the wrong path for my environment. My path is source install/setup.bash . I ran these commands while I was in the catkin_ws

Semih67, 2018-08-28:
But now I get a warning when I launch the camera: [warn] ... : Timeout waiting for camera. I also got this when I tested the camera.

inbangsa, 2018-08-28:
@Semih67 
whats the location of this setup.bash ? 

 if you have followed the Instruction then please check 
```
  echo $LPR_ROS
```
and after this please check 
```
ls $LPR_ROS/ifm3d
```
if the installation is fine then you will find setup.bash in this folder. 

Have you done ~/.bashrc: changes as per the [installation Instruction](https://github.com/lovepark/ifm3d-ros/blob/master/doc/building.md#step-by-step-build-instructions) 

'''
from your comment:
My src folder looks like this: ifm3d ifm3d-ros CMakeLists.txt
'''
how you got the ifm3d and ifm3d-ros both folder at this place..  as per the instruction there should only be soft-link to ifm3d-ros (which is in dev directory). if you have not done the softlink then you should have ifm3d-ros only

inbangsa, 2018-08-28:
@Semih67  oho great that you solve the issue 

Now for timeout error can you please checkout our [troubleshoot guide](https://github.com/lovepark/ifm3d/blob/master/doc/troubleshoot.md#connecting-the-camera)

graugans, 2018-09-19:
@Semih67 can we close this issue?

Semih67, 2018-09-24:
yes you can close it

graugans, 2018-09-24:
@tpanzarella can you please close this issue?

closed 2018-09-24 by tpanzarella