ifm/ifm3d #89: ExposureTime assignment when changing imager type

nayan27, 2018-08-21:
Hi,

I believe i encountered another little problem:
Let's say the camera is in "under5m_high" imager type. In this specific imager type (high), there is no "ExposureTime" parameter in the imager (only a ExposureTimeList). So, if my application attempts to modify the imager type (to under5m_low) AND assign an ExposureTime at the same time (same FromJSON call), the "ExposureTime" element won't be found in the current json configuration and so there will be problems. Am I mistaken ?

Also, why do we have 2 separate way of identifying exposure times ? ("ExposureTime", "ExposureTime"+ "ExposureTimeRatio", "ExposureTimeList"). Why don't we simply use "ExposureTimeList" all the time ?

Thank you

tpanzarella, 2018-08-21:
> Also, why do we have 2 separate way of identifying exposure times ? ("ExposureTime", "ExposureTime"+ "ExposureTimeRatio", "ExposureTimeList"). Why don't we simply use "ExposureTimeList" all the time ?

`ExposureTimeList` is read-only and used to report back the actual exposure times being used in usec. When using only a single exposure mode, you specify the exposure time explicitly via the `ExposureTime` parameter. When using a dual exposure mode you specify the long exposure explicitly and the short exposure as the ratio (divide the long by the ratio to get the short). When using three exposures ("high"), the integration times are chosen for you -- which is why you cannot set them.

> .... so there will be problems. Am I mistaken ?

Sure. You need to change the imager type and save the change. Then set the allowable parameters for that imager type .. e.g., trying to set the exposure time on an imager type that does not allow for that will result in an error generated camera-side.

nayan27, 2018-08-21:
@tpanzarella  Thank you for the quick answer! I understand. I did not know the ExposureTimeList was a read-only parameter. I must have missed that information at some point.

On other issue we talked about the fact that the FromJSON configuration was taking a little while do complete. With what you are saying, i am now oblige to use it twice to set the device in the correct state if there is a change in image type. Wouldn't it be simpler that the camera detects and assign the imager type first and then consider the exposure elements if necessary ? Would this role fall into the library's responsabilities or would it be a camera firmware thing ?

Thanks !

tpanzarella, 2018-08-21:
It is a camera firmware thing.

Also, as noted earlier by @graugans you do not need to import a full JSON dump every time, just a snippet that sets the exposure for the particular application is all you need.

graugans, 2018-08-22:
@nayan27 Yes I am still have to document all the parameters. #44 . @tpanzarella I do not understand why this is a camera thing. The camera does provide an interface to change one parameter at a time so you are doomed to change the ``imager.Type`` before changing the ``ExposureTime`` and ``ExposureTimeRatio``. From my understanding the issue comes from how the ifm3d interface works. With the JSON you are changing a set of parameters. Due to the fact we do use a JSON object for the imager paramters there is no ordering available _An object is an unordered set of name/value pairs._ I guess if we use an array, _An array is an ordered collection of values_, instead of an object the requirements from @nayan27 could be achieved. See [http://json.org](http://json.org) for the details.  But I am not sure if this is achievable in terms of backwards compatibility.

Another option could be to adding a check [here](https://github.com/lovepark/ifm3d/blob/master/modules/camera/src/libifm3d_camera/camera.cpp#L981) 
like this:
```cpp
              json t_type = j_im["Type"];
              if (! t_type.is_null())
                {
                  j_im.erase("Type");
                  this->pImpl->ChangeImagerType(v);
                }
```
This will change the Type before applying the parameters

graugans, 2018-08-22:
Okay, again it was more complex than I thought in the beginning. Here is a patch which should solve the issue

```patch
From 7a58e15175dd31f3819aab10f8cab62c43cd3092 Mon Sep 17 00:00:00 2001
From: Christian Ege <christian.ege@ifm.com>
Date: Wed, 22 Aug 2018 08:18:55 +0000
Subject: [PATCH] Fix issue #89 by setting imager type first

In the issue it was stated that when switching from one
imager type to the other the ifm3d bails out due to conflicting
parameters of the current image type and the new parameters this is
solved by setting the imager type before applying the corresponding
parameters. The downside of this is the fact we have to read the
new configuration after changing the imager type to compare to the
right values during import. Due to the fact we stay in the edit session
we save some time compared to two consecutive calls to ``FromJSON``

---
 modules/camera/src/libifm3d_camera/camera.cpp | 74 +++++++++++++++++----------
 1 file changed, 47 insertions(+), 27 deletions(-)

diff --git a/modules/camera/src/libifm3d_camera/camera.cpp b/modules/camera/src/libifm3d_camera/camera.cpp
index f4eba42..39ef44a 100644
--- a/modules/camera/src/libifm3d_camera/camera.cpp
+++ b/modules/camera/src/libifm3d_camera/camera.cpp
@@ -923,29 +923,32 @@ ifm3d::Camera::FromJSON(const json& j)

               VLOG(IFM3D_TRACE) << "Application of interest is at index="
                 << idx;
-
-              // now in `current` (which is a whole camera dump)
-              // we need to find the application at index `idx`.
-              json curr_app = json({});
-              json curr_apps = current["ifm3d"]["Apps"];
-              bool app_found = false;
-              for (auto& a : curr_apps)
+              auto get_current_app = [&current,&idx]()
                 {
-                  if (std::stoi(a["Index"].get<std::string>()) == idx)
+                  // now in `current` (which is a whole camera dump)
+                  // we need to find the application at index `idx`.
+                  json curr_app = json({});
+                  json curr_apps = current["ifm3d"]["Apps"];
+                  bool app_found = false;
+                  for (auto& a : curr_apps)
                     {
-                      curr_app = a;
-                      app_found = true;
-                      break;
+                      if (std::stoi(a["Index"].get<std::string>()) == idx)
+                        {
+                          curr_app = a;
+                          app_found = true;
+                          break;
+                        }
                     }
-                }
-
-              if (! app_found)
-                {
-                  LOG(ERROR) << "Could not find an application at index="
-                  << idx;
-                  throw ifm3d::error_t(IFM3D_JSON_ERROR);
-                }

+                  if (! app_found)
+                    {
+                      LOG(ERROR) << "Could not find an application at index="
+                      << idx;
+                      throw ifm3d::error_t(IFM3D_JSON_ERROR);
+                    }
+                  return curr_app;
+                };
+              json curr_app = get_current_app();
               // at this point both the new and current
               // should have the same index and type and
               // we want to make sure of that.
@@ -978,18 +981,35 @@ ifm3d::Camera::FromJSON(const json& j)
                 {
                   j_im.erase("TemporalFilter");
                 }
+              // If the imager settings do contain a  Type
+              // ensure the we first change the type this ensures
+              // all other consecutive parameters are accepted if they
+              // do fit the imager Type.
+              const std::string img_type = {"Type"};
+              json j_type = j_im[img_type];
+              if ((! j_type.is_null()) && (idx > 0) )
+                {
+                  json j_img = json({});
+                  j_img[img_type] = j_type;
+                  this->FromJSON_(curr_app["Imager"],j_img,
+                    [this](const std::string& k, const std::string& v)
+                      {
+                        (void)k;
+                        this->pImpl->ChangeImagerType(v);
+                      },
+                    [this](){ this->pImpl->SaveApp(); },
+                    "Imager", idx);
+                  j_im.erase(img_type);
+                  VLOG(IFM3D_TRACE)
+                    << "New imager type, updating our dump";
+                  current = this->ToJSON_(false);
+                  curr_app = get_current_app();
+                }

               this->FromJSON_(curr_app["Imager"], j_im,
                 [this](const std::string& k, const std::string& v)
                   {
-                    if (k == "Type")
-                      {
-                        this->pImpl->ChangeImagerType(v);
-                      }
-                    else
-                      {
-                        this->pImpl->SetImagerParameter(k,v);
-                      }
+                    this->pImpl->SetImagerParameter(k,v);
                   },
                 [this](){ this->pImpl->SaveApp(); },
                 "Imager", idx);
--
2.7.4
```

Comments and improvements are welcome.

nayan27, 2018-08-23:
Changes looks good.
I don't udnerstand what takes a lot of time in the FromJSON function but i am wondering if doing this change and adding a this->pImpl->SaveApp(); would make it even longer.

Any thought on that ?

graugans, 2018-08-24:
@nayan27 you are right ``this->pImpl->SaveApp(); `` maybe not needed. This is a draft only. so optimizing can be achieved. A big potion of the time is spend on the xmlrpc calls. As Tom suggested. the easiest way would be if the camera itself creates the JSON in one call. The second optimization is accepting the JSON on the cam.

nayan27, 2018-08-28:
@graugans  @tpanzarella  Hey simply doing a little follow up here! For now i can work around this by assigning the ExposureType first and then going on with the rest of the parameters but since the fromJSON already takes a lot of time it is a little unoptimized to work that way. Any idea on when this issue could be solved ?

Thank you!

graugans, 2018-09-10:
@nayan27 sorry for the delay, I was in vacation for the last two weeks. From my point of view the patch above is a band aid to prevent the user from two consecutive calls to ``FromJSON`` and should be faster due to the fact the session is not reopened. There was no feedback from @tpanzarella related to this so I am not sure if I shall create a Pull Request for the change.

theseankelly, 2020-01-12:
Closing due to lack of activity.

closed 2020-01-12 by theseankelly