ifm/ifm3d #90: static build

Postea-CS, 2018-08-28:
Is there a way to do static build?

tpanzarella, 2018-08-28:
We currently do not have the struts in place to perform a static build.

Postea-CS, 2018-08-29:
How about using another version of boost? One that I build myself. tried export BOOST_ROOT but still isn't finding -lBoost:system.

tpanzarella, 2018-08-29:
We use CMake as our meta-build system. To that end, we employ Boost's integration with CMake to locate our Boost related dependencies *and* to properly set our compiler and linker flags. So, for example, in our `framegrabber` module where we rely on ASIO (header-only ... so nothing to link to) and System, we specify that dependency like: `find_package(Boost REQUIRED COMPONENTS system)`. What this amounts to in terms of linking is that we specify `-lboost_system` (not as you spell it above). Similarly, in our `tools` module, we utilize Boost program options (for parsing command-line arguments), which we call out as `find_package(Boost REQUIRED COMPONENTS program_options)` and manifests as a linker specification like `-lboost_program_options`.

If your custom Boost build does not include the CMake integration (i.e., acquiring compiler/linker settings via `find_package(...)`, you could edit our CMakeList.txt files as necessary to explicitly link to the libraries you installed. I could try to provide some guidance on that if necessary.

closed 2018-09-06 by Postea-CS