ifm/ifm3d #91: Provide swupdate functionality as a library

theseankelly, 2018-09-06:
Create shared lib/dll for swupdate functionality with the following features:

- accept path to SWU file as input
- expose a status indication/heartbeat for propagating status information (progress bar)

This does not need to be a standalone library independent from ifm3d

inbangsa, 2018-09-08:
There is a document [firmwareupdate](https://github.com/lovepark/ifm3d/blob/master/doc/firmwareUpdate.md) which provides a python script which accepts SWU file and updates the device.  There is no progress bar in it but it prints progress logs on console. 

Let us know if this helps you ?

graugans, 2018-09-10:
A __real__ progress bar will not work. We have no progress information during firmware updates. Due to the different types of images it is not really predictable how long each step will take. Because of this we have a spinner in the ifm Vision Assistant and not a progress bar. What is the expected time frame for this?

theseankelly, 2018-09-10:
Hey Sachin -- it should be a linkable C++ library to build in to our customers' fleet management software.

Since our own internal tooling doesn't support a progress bar, we can/should go without it for this request. Thanks for the info.

tpanzarella, 2019-07-26:
This feature is available as of `0.14.0`.

closed 2019-07-26 by tpanzarella