ifm/ifm3d #92: sdk 'hang' on destructor

Postea-CS, 2018-09-06:
I think i had this issue with lib03d3 as well and since most of the code are similar. i have the same issue as well in ifm3d. 

On exiting the program, in some cases, it seems the code is frozen/hang, not exiting the executable for a long time(8 mins).  i have track it down to the ifm sdk i believe. I manage to 'avoid' it by modifying the framegrabber and pulling the Stop() function out to public and calling it. 

this is not happening all the time it seems to happen when i let the executable run for a while. I am not sure exactly what is the issue and i am not 100% sure i have fixed the issue in this sdk. i know for lib03d3, after pulling the Stop() out, i haven't seen the issue anymore.

tpanzarella, 2018-09-06:
Can you provide a minimal example that exhibits this behavior? I.e., something that shows your usage pattern with the code. This is the first I have heard of anything like this. Realize that most of the core structures within ifm3d are wrapped in `std::shared_ptr` and so a reference count is maintained for any code participating in its ownership. To that end, if you have a dangling reference hanging around, this behavior would be consistent and by design.

Again, a minimal code example which exhibits this behavior would be helpful.

tpanzarella, 2018-11-07:
Is this still an open item? Can it be closed?

Postea-CS, 2018-11-08:
Hi Tom,

We are still having the issue, I can’t seem to reproduce it with a standalone code, but its pretty much still happening on our exe. I haven’t had the time to try different way to reproduce this on a standalone, just the one time …

Regards,
Chin

theseankelly, 2019-10-09:
Closing out old issues, please reactivate if this is still an issue.

closed 2019-10-09 by theseankelly