ifm/ifm3d #93: Package creation and deployment of ifm3d and ifm3d-ros with ROS bloom

eggerk, 2018-09-11:
Hi,

I'm trying to create and deploy debian packages for ifm3d and ifm3d-ros.

I can create deb packages out of ifm3d by following the instructions:
```
cd ifm3d
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/usr ..
make
make package
make repackage
sudo dpkg -i *.deb
```

However, when I try to create packages of ifm3d-ros using [ROS bloom](http://wiki.ros.org/bloom), I get an error. I ran the following commands:
```
cd ifm3d-ros
bloom-generate rosdebian
fakeroot debian/rules binary
```

I get the following error output:
```
...

	chmod 0644 -- debian/.debhelper/ros-melodic-ifm3d/dbgsym-root/usr/lib/debug/.build-id/11/a7b340e8cf0cbeebc3b0e4a2ec04915689aadd.debug
	chown 0:0 -- debian/.debhelper/ros-melodic-ifm3d/dbgsym-root/usr/lib/debug/.build-id/11/a7b340e8cf0cbeebc3b0e4a2ec04915689aadd.debug
	strip --remove-section=.comment --remove-section=.note --strip-unneeded debian/ros-melodic-ifm3d/opt/ros/melodic/lib/libifm3d_ros.so
	objcopy --add-gnu-debuglink debian/.debhelper/ros-melodic-ifm3d/dbgsym-root/usr/lib/debug/.build-id/11/a7b340e8cf0cbeebc3b0e4a2ec04915689aadd.debug debian/ros-melodic-ifm3d/opt/ros/melodic/lib/libifm3d_ros.so
	install -d debian/.debhelper/ros-melodic-ifm3d/dbgsym-root/usr/share/doc
	ln -s ros-melodic-ifm3d debian/.debhelper/ros-melodic-ifm3d/dbgsym-root/usr/share/doc/ros-melodic-ifm3d-dbgsym
   dh_makeshlibs -O-v -O--buildsystem=cmake
	rm -f debian/ros-melodic-ifm3d/DEBIAN/shlibs
   debian/rules override_dh_shlibdeps
make[1]: Entering directory '/ifm3d-ros'
# In case we're installing to a non-standard location, look for a setup.sh
# in the install tree that was dropped by catkin, and source it.  It will
# set things like CMAKE_PREFIX_PATH, PKG_CONFIG_PATH, and PYTHONPATH.
if [ -f "/opt/ros/melodic/setup.sh" ]; then . "/opt/ros/melodic/setup.sh"; fi && \
dh_shlibdeps -l/ifm3d-ros/debian/ros-melodic-ifm3d//opt/ros/melodic/lib/
	dpkg-shlibdeps -Tdebian/ros-melodic-ifm3d.substvars -l/ifm3d-ros/debian/ros-melodic-ifm3d//opt/ros/melodic/lib/ debian/ros-melodic-ifm3d/opt/ros/melodic/lib/libifm3d_ros.so
dpkg-shlibdeps: warning: can't extract name and version from library name 'libimage_transport.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libimage_transport.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libcv_bridge.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libcv_bridge.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libnodeletlib.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libnodeletlib.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libclass_loader.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libclass_loader.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libroscpp.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libroscpp.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'librosconsole.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'librosconsole.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libroscpp_serialization.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libroscpp_serialization.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'librostime.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'librostime.so'
dpkg-shlibdeps: error: no dependency information found for /usr/lib/libifm3d_image.so.0.10.0 (used by debian/ros-melodic-ifm3d/opt/ros/melodic/lib/libifm3d_ros.so)
Hint: check if the library actually comes from a package.
dh_shlibdeps: dpkg-shlibdeps -Tdebian/ros-melodic-ifm3d.substvars -l/ifm3d-ros/debian/ros-melodic-ifm3d//opt/ros/melodic/lib/ debian/ros-melodic-ifm3d/opt/ros/melodic/lib/libifm3d_ros.so returned exit code 2
dh_shlibdeps: Aborting due to earlier error
debian/rules:50: recipe for target 'override_dh_shlibdeps' failed
make[1]: *** [override_dh_shlibdeps] Error 25
make[1]: Leaving directory '/ifm3d-ros'
debian/rules:22: recipe for target 'binary' failed
make: *** [binary] Error 2
```

Something seems to be wrong with the libraries from this repository:
```
dpkg-shlibdeps: error: no dependency information found for /usr/lib/libifm3d_image.so.0.10.0 (used by debian/ros-melodic-ifm3d/opt/ros/melodic/lib/libifm3d_ros.so)
Hint: check if the library actually comes from a package.
dh_shlibdeps: dpkg-shlibdeps -Tdebian/ros-melodic-ifm3d.substvars -l/ifm3d-ros/debian/ros-melodic-ifm3d//opt/ros/melodic/lib/ debian/ros-melodic-ifm3d/opt/ros/melodic/lib/libifm3d_ros.so returned exit code 2
```

(These two commands work fine for me for creating and deploying debian packages of other catkin packages.)

I'm on Ubuntu 18.04 and ROS Melodic.

Thank you for your help.

graugans, 2018-09-12:
Hello @eggerk I have created a ifm3d-core bloom release repository here: https://github.com/graugans/ifm3d-release this might help you with creating bloom packages for ifm3d-ros which is also on my TODO list. Unfortunately I am pretty new to all this bloom and ROS stuff an low in cycles to push this to the official repos. Please let me know if this is of any help for you.

eggerk, 2018-09-12:
Thank you for your help.

With your release repo, I managed to create debian packages for for ifm3d and ifm3d-ros for Ubuntu 16.04/ROS kinetic. I haven't been able to test these packages with a physical sensor yet, but the debs install well.

These are the commands I ran:
```sh
git clone https://github.com/graugans/ifm3d-release.git
cd ifm3d-release
git checkout debian/kinetic/xenial/ifm3d-core
# Need to install package.xml in a different place as it will conflict with the file from
# ifm3d-ros otherwise.
sed -i "s@share/ifm3d@share/ifm3d-core@" CMakeLists.txt
DEB_BUILD_OPTIONS=nocheck fakeroot debian/rules binary
sudo dpkg -i ../ros-kinetic-ifm3d-core_0.9.2-9xenial_amd64.deb
```

Then, I can also deploy ifm3d-ros in a similar way:
```sh
git clone https://github.com/lovepark/ifm3d-ros.git
cd ifm3d-ros
bloom-generate rosdebian
fakeroot debian/rules binary
sudo dpkg -i ../ros-kinetic-ifm3d_0.6.0-0xenial_amd64.deb
```

Here are the debians I generated this way: [ifm_debs.tar.gz](https://github.com/lovepark/ifm3d/files/2374319/ifm_debs.tar.gz)

I will try to test the debs on an actual system soon and will report back to you.

Finally, do you remember how you created the release repository? Because we also need debs for Ubuntu 18.04/ROS melodic, but this fails with your current release repository because it cannot properly find pthread. Then, we could at least build our own temporary deployment-pipeline until you find time to provide debs yourselves.

graugans, 2018-09-12:
@eggerk thanks for testing. I followed the guide on the ROS wiki [http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty) There is a specific topic for switching from one ROS release to another here: [http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty#Porting_patches_from_one_rosdistro_to_another](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty#Porting_patches_from_one_rosdistro_to_another) Unfortunately today I will not find a timeslot to do this. 

BTW at teh moment the bash completion is missing for the ifm3d tools because it is installed to the system path ``/etc/`` which is prevented by catkin isolated installation. From my point of view this is a minor issue. But I have no simple workaround at the moment.

eggerk, 2018-09-12:
Ok I managed to make it work under 18.04. I copied the package.xml from your release-repo and added a line in the CMakeLists.txt to install the package.xml and then went via `bloom-generate`.

For some reason, I got some very weird error messages in my docker container. But that seems to be a problem of my docker and your code. (It works fine if done directly on my system or within other dockers.)

commit a719bce9 (2022-09-05): Merge branch 'restructuring/tests' into 'o3r/main-next'

Test Restructuring to work with all the device

Closes #93

See merge request syntron/support/csr/ifm3d/ifm3d!84
closed 2018-09-12 by eggerk