ifm/ifm3d #94: [windows][ifm3d-tools]Compilation Issue

inbangsa, 2018-09-17:
Trying compiling dev-0.11.0 on windows with  VS 2015 shows following error 

```
 jitter_app.cpp
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\tuple(164): warning C4244: 'initializing': conversion from 'std::ios_base::iostate' to 'float', possible loss of data [H:\ifm3d\ifm3d\build\modules\tools\src\libifm3d_tools\ifm3d_tools_shared.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\tuple(520): note: see reference to function template instantiation 'std::_Tuple_val<_This>::_Tuple_val<_Ty>(_Other &&)' being compiled
          with
          [
              _This=float,
              _Ty=std::ios_base::iostate,
              _Other=std::ios_base::iostate
          ]
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\tuple(519): note: see reference to function template instantiation 'std::_Tuple_val<_This>::_Tuple_val<_Ty>(_Other &&)' being compiled
          with
          [
              _This=float,
              _Ty=std::ios_base::iostate,
              _Other=std::ios_base::iostate
          ]
  H:\ifm3d\ifm3d\modules\tools\src\libifm3d_tools\fg\jitter_app.cpp(132): note: see reference to function template instantiation 'std::tuple<float,float>::tuple<int,int,0,0>(std::tuple<int,int> &&)' being compiled
  H:\ifm3d\ifm3d\modules\tools\src\libifm3d_tools\fg\jitter_app.cpp(132): note: see reference to function template instantiation 'std::tuple<float,float>::tuple<int,int,0,0>(std::tuple<int,int> &&)' being compiled
  H:\ifm3d\ifm3d\modules\tools\src\libifm3d_tools\fg\jitter_app.cpp(215): note: see reference to function template instantiation 'std::tuple<float,float> mean_stdev<float>(const std::vector<float,std::allocator<_Ty>> &)' being compiled
          with
          [
              _Ty=float
          ]
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\numeric(20): warning C4244: '=': conversion from 'float' to 'int', possible loss of data [H:\ifm3d\ifm3d\build\modules\tools\src\libifm3d_tools\ifm3d_tools_shared.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\numeric(30): note: see reference to function template instantiation '_Ty std::_Accumulate_unchecked<const float*,_Ty,_Fn2>(_InIt,_InIt,_Ty,_Fn2 &)' being compiled
          with
          [
              _Ty=int,
              _Fn2=std::plus<void>,
              _InIt=const float *
          ]
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\numeric(38): note: see reference to function template instantiation '_Ty std::accumulate<_InIt,_Ty,std::plus<void>>(_InIt,_InIt,_Ty,_Fn2)' being compiled
          with
          [
              _Ty=int,
              _InIt=std::_Vector_const_iterator<std::_Vector_val<std::_Simple_types<float>>>,
              _Fn2=std::plus<void>
          ]
  H:\ifm3d\ifm3d\modules\tools\src\libifm3d_tools\fg\jitter_app.cpp(135): note: see reference to function template instantiation '_Ty std::accumulate<std::_Vector_const_iterator<std::_Vector_val<std::_Simple_types<float>>>,int>(_InIt,_InIt,_Ty)' being compiled
          with
          [
              _Ty=int,
              _InIt=std::_Vector_const_iterator<std::_Vector_val<std::_Simple_types<float>>>
          ]
  schema_app.cpp
  Generating Code...
h:\ifm3d\ifm3d\modules\tools\src\libifm3d_tools\fg\jitter_app.cpp(141): error C4716: '<lambda_19be6af884a382f0569ba7ba5466302a>::operator()': must return a value [H:\ifm3d\ifm3d\build\modules\tools\src\libifm3d_tools\ifm3d_tools_shared.vcxproj]
Done Building Project "H:\ifm3d\ifm3d\build\modules\tools\src\libifm3d_tools\ifm3d_tools_shared.vcxproj" (default targets) -- FAILED.
```

tpanzarella, 2018-09-17:
@inbangsa  I did not look too closely at this, but perhaps the problem is [here](https://github.com/lovepark/ifm3d/blob/dev-0.11.0/modules/tools/src/libifm3d_tools/fg/jitter_app.cpp#L140) This lambda should return `void` not `T` as it is mutating the `ssd` variable by reference as the lambda, in this case, is a closure.

inbangsa, 2018-09-17:
Thanks for the commit. tested on windows now dev-011.0 is compiling fine with VS 2015.  It seems there is a difference on compiler setting for windows and Linux. Closing the issue

commit 4075ec6d (2023-02-01): Merge branch 'versioned_docs' into 'main'

Prepare versioned docs

Closes #94

See merge request syntron/support/csr/ifm3d/ifm3d!184
closed 2018-09-17 by inbangsa