ifm/ifm3d #97: ifm3d-examples build error "ifm3d_FOUND to FALSE"

asaini12, 2018-10-09:
I am having issues building the "ifm3d-examples" package. I am not sure what's causing the issue. Command "ifm3d ls" runs successfully. Also I have successfully built and run the point cloud viewer "ifm3d-pcl-viewer" package.  

It would be very helpful if someone can help me troubleshoot the issue.

```

-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ifm3d found component: camera
-- ifm3d found component: framegrabber
-- ifm3d found component: image
CMake Warning at /usr/lib/cmake/ifm3d-0.10.0/ifm3d-config.cmake:44 (message):
  ifm3d could not find component: pcicclient
Call Stack (most recent call first):
  CMakeLists.txt:19 (find_package)

CMake Error at CMakeLists.txt:19 (find_package):
  Found package configuration file:

    /usr/lib/cmake/ifm3d-0.10.0/ifm3d-config.cmake

  but it set ifm3d_FOUND to FALSE so package "ifm3d" is considered to be NOT
  FOUND.

-- Configuring incomplete, errors occurred!

```

graugans, 2018-10-10:
Unfortunately this is caused by the fact @tpanzarella changed the pcicclient from default **on** to default **off**. I'll provide an pull request for  https://github.com/lovepark/ifm3d-examples which will only build the examples for which the required modules do exists. Meanwhile you can comment the examples out which depends on the pciccleint module or rebuild the ifm3d library with the pcicclient enabled. Sorry for this  inconvenience.

asaini12, 2018-10-10:
Thank  you @graugans. I rebuilt the ifm3d with pcicclient enabled and it worked.

graugans, 2018-10-18:
@asaini12 can you please close this issue when you agree with the solution. There is also a Pull-Request in [ifm3d-examples](https://github.com/lovepark/ifm3d-examples/pull/6) which addresses this

closed 2018-10-18 by tpanzarella

coa06yl, 2022-11-01:
@graugans Hi graugans, I have this error msg when I catkin_make. It will be great if you can help me solve the issue. Many thanks!

[image]

PR #98 (2018-10-12): Provide a possibility to define modules as  optional
This will fix #97 when the corresponding Pull Request is opened in [ifm3d-examples](https://github.com/lovepark/ifm3d-examples). With this fix we can define optional packages while using the  ``find_package`` like the following example.

```
find_package(ifm3d 0.9.0 CONFIG
    REQUIRED COMPONENTS camera framegrabber
    OPTIONAL_COMPONENTS image pcicclient
    )
```