ifm/ifm3d #100: undefined reference to `ifm3d::DEFAULT_XMLRPC_PORT'

pedghz, 2018-10-18:
### I am trying the code mentioned in the basic_usage page and I get the following errors:
Anye one know what is wrong here?

> CMakeFiles/ifm_o3d303.dir/main.cpp.o: In function 'main':
main.cpp:(.text+0x143): undefined reference to 'ifm3d::DEFAULT_XMLRPC_PORT'
main.cpp:(.text+0x152): undefined reference to 'ifm3d::DEFAULT_PASSWORD[abi:cxx11]'
main.cpp:(.text+0x157): undefined reference to 'ifm3d::DEFAULT_IP[abi:cxx11]'
main.cpp:(.text+0x15f): undefined reference to 'ifm3d::Camera::MakeShared(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
main.cpp:(.text+0x296): undefined reference to 'ifm3d::ImageBuffer::AmplitudeImage()'
main.cpp:(.text+0x2df): undefined reference to 'ifm3d::ImageBuffer::XYZImage()'
CMakeFiles/ifm_o3d303.dir/main.cpp.o: In function 'cv::Mat::~Mat()':
main.cpp:(.text._ZN2cv3MatD2Ev[_ZN2cv3MatD5Ev]+0x39): undefined reference to 'cv::fastFree(void*)'
CMakeFiles/ifm_o3d303.dir/main.cpp.o: In function 'cv::Mat::operator=(cv::Mat const&)':
main.cpp:(.text._ZN2cv3MataSERKS0_[_ZN2cv3MataSERKS0_]+0x111): undefined reference to 'cv::Mat::copySize(cv::Mat const&)'
CMakeFiles/ifm_o3d303.dir/main.cpp.o: In function 'cv::Mat::release()':
main.cpp:(.text._ZN2cv3Mat7releaseEv[_ZN2cv3Mat7releaseEv]+0x47): undefined reference to 'cv::Mat::deallocate()'
CMakeFiles/ifm_o3d303.dir/main.cpp.o: In function 'bool ifm3d::FrameGrabber::WaitForFrame<ifm3d::ImageBuffer>(ifm3d::ByteBuffer<ifm3d::ImageBuffer>*, long, bool, bool)':
main.cpp:(.text._ZN5ifm3d12FrameGrabber12WaitForFrameINS_11ImageBufferEEEbPNS_10ByteBufferIT_EElbb[_ZN5ifm3d12FrameGrabber12WaitForFrameINS_11ImageBufferEEEbPNS_10ByteBufferIT_EElbb]+0x65): undefined reference to 'ifm3d::FrameGrabber::WaitForFrame(long, std::function<void (std::vector<unsigned char, std::allocator<unsigned char> >&)>)'
CMakeFiles/ifm_o3d303.dir/main.cpp.o:(.data._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvE8vlocal__[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvE8vlocal__]+0x0): undefined reference to 'google::kLogSiteUninitialized'
CMakeFiles/ifm_o3d303.dir/main.cpp.o: In function 'ifm3d::error_t::~error_t()':
main.cpp:(.text._ZN5ifm3d7error_tD2Ev[_ZN5ifm3d7error_tD5Ev]+0xd): undefined reference to 'vtable for ifm3d::error_t'
CMakeFiles/ifm_o3d303.dir/main.cpp.o:(.data._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvE8vlocal___0[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvE8vlocal___0]+0x0): undefined reference to 'google::kLogSiteUninitialized'
CMakeFiles/ifm_o3d303.dir/main.cpp.o:(.data._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvE8vlocal___1[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvE8vlocal___1]+0x0): undefined reference to 'google::kLogSiteUninitialized'
CMakeFiles/ifm_o3d303.dir/main.cpp.o: In function 'ifm3d::ByteBuffer<ifm3d::ImageBuffer>::Organize()::{lambda(ifm3d::image_chunk, unsigned int, unsigned long)#1}::operator()(ifm3d::image_chunk, unsigned int, unsigned long) const':
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm]+0x2bb): undefined reference to 'google::LogMessage::LogMessage(char const*, int, int)'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm]+0x2c7): undefined reference to 'google::LogMessage::stream()'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm]+0x2f0): undefined reference to 'google::LogMessage::~LogMessage()'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm]+0x303): undefined reference to 'IFM3D_PIXEL_FORMAT_ERROR'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm]+0x30d): undefined reference to 'ifm3d::error_t::error_t(int)'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm]+0x317): undefined reference to 'typeinfo for ifm3d::error_t'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUlNS_11image_chunkEjmE_clES3_jm]+0x340): undefined reference to 'google::LogMessage::~LogMessage()'
CMakeFiles/ifm_o3d303.dir/main.cpp.o: In function 'ifm3d::ByteBuffer<ifm3d::ImageBuffer>::Organize()::{lambda(unsigned int, unsigned long, unsigned long, unsigned long)#2}::operator()(unsigned int, unsigned long, unsigned long, unsigned long) const':
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm]+0xfd): undefined reference to 'google::LogMessage::LogMessage(char const*, int, int)'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm]+0x109): undefined reference to 'google::LogMessage::stream()'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm]+0x132): undefined reference to 'google::LogMessage::~LogMessage()'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm]+0x145): undefined reference to 'IFM3D_PIXEL_FORMAT_ERROR'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm]+0x14f): undefined reference to 'ifm3d::error_t::error_t(int)'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm]+0x159): undefined reference to 'typeinfo for ifm3d::error_t'
main.cpp:(.text._ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm[_ZZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEvENKUljmmmE0_clEjmmm]+0x182): undefined reference to 'google::LogMessage::~LogMessage()'
CMakeFiles/ifm_o3d303.dir/main.cpp.o: In function 'ifm3d::ByteBuffer<ifm3d::ImageBuffer>::Organize()':
main.cpp:(.text._ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv[_ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv]+0xed): undefined reference to 'ifm3d::IMG_BUFF_START'
main.cpp:(.text._ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv[_ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv]+0x108): undefined reference to 'ifm3d::get_chunk_index(std::vector<unsigned char, std::allocator<unsigned char> > const&, ifm3d::image_chunk, unsigned long)'
main.cpp:(.text._ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv[_ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv]+0x126): undefined reference to 'ifm3d::IMG_BUFF_START'
main.cpp:(.text._ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv[_ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv]+0x141): undefined reference to 'ifm3d::get_chunk_index(std::vector<unsigned char, std::allocator<unsigned char> > const&, ifm3d::image_chunk, unsigned long)'
main.cpp:(.text._ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv[_ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv]+0x14f): undefined reference to 'ifm3d::IMG_BUFF_START'
main.cpp:(.text._ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv[_ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv]+0x16a): undefined reference to 'ifm3d::get_chunk_index(std::vector<unsigned char, std::allocator<unsigned char> > const&, ifm3d::image_chunk, unsigned long)'
main.cpp:(.text._ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv[_ZN5ifm3d10ByteBufferINS_11ImageBufferEE8OrganizeEv]+0x178): undefined reference to 'ifm3d::IMG_BUFF_START'
[... cut ...]
main.cpp:(.text._ZN9__gnu_cxx13new_allocatorIN5ifm3d11ImageBufferEE7destroyIS2_EEvPT_[_ZN9__gnu_cxx13new_allocatorIN5ifm3d11ImageBufferEE7destroyIS2_EEvPT_]+0x18): undefined reference to 'ifm3d::ImageBuffer::~ImageBuffer()'
collect2: error: ld returned 1 exit status
make[2]: *** [ifm_o3d303] Error 1
make[1]: *** [CMakeFiles/ifm_o3d303.dir/all] Error 2
make: *** [all] Error 2
13:50:27: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project ifm_o3d303 (kit: Desktop)
When executing step "Make"
13:50:27: Elapsed time: 00:03.

graugans, 2018-10-18:
Hello @pedghz this looks like you are trying to compile the code but do not link against the ifm3d lib. Please take a look at the https://github.com/lovepark/ifm3d-examples repo for details how to get started.

pedghz, 2018-10-18:
@graugans You are probably right. I tried to build that repo as well now and I get a related error there as well. Here is the result from `cmake..`:

> cmake ..
> -- The C compiler identification is GNU 5.4.0
> -- The CXX compiler identification is GNU 5.4.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- ifm3d found component: camera
> -- ifm3d found component: framegrabber
> -- ifm3d found component: image
> CMake Warning at /usr/lib/cmake/ifm3d-0.10.0/ifm3d-config.cmake:44 (message):
>   ifm3d could not find component: pcicclient
> Call Stack (most recent call first):
>   CMakeLists.txt:20 (find_package)
> 
> 
> CMake Error at CMakeLists.txt:20 (find_package):
>   Found package configuration file:
> 
>     /usr/lib/cmake/ifm3d-0.10.0/ifm3d-config.cmake
> 
>   but it set ifm3d_FOUND to FALSE so package "ifm3d" is considered to be NOT
>   FOUND.
>

tpanzarella, 2018-10-18:
Please see [this](https://github.com/lovepark/ifm3d-examples/issues/3) issue.

closed 2018-10-18 by pedghz

graugans, 2018-10-18:
@tpanzarella I guess we also can close #97 because it is a duplicate of [this](https://github.com/lovepark/ifm3d-examples/issues/3) and finally feedback on this [PR](https://github.com/lovepark/ifm3d-examples/pull/6) would be nice to avoid those issues.