ifm/ifm3d #101: Serializing ByteBuffer

pfrydlewicz, 2018-10-22:
Am I right in my understanding that in order to serialize a ByteBuffer I just need to copy the public Bytes vector?

Does deserialization work just the other way around?

Thanks for clarification
Paul

tpanzarella, 2018-10-22:
Hi Paul,

I am not sure I entirely understand your question, but you are correct that the "Bytes" vector is everything you need to copy/move/"serialize"/etc. an `ifm3d::ByteBuffer`. In terms of "deserializing", yes, we provide the `SetBytes` method as part of the public interface. The details of all those mechanics are found [here](https://github.com/lovepark/ifm3d/blob/master/modules/framegrabber/include/ifm3d/fg/detail/byte_buffer.hpp). 

The above being said, if you are interested in hooking into the `ifm3d` pixel parsing and providing your own image container (i.e., you don't like OpenCV for example), I highly encourage you to read [this](https://github.com/lovepark/ifm3d/blob/master/doc/img_container.md) document. And you can use both the [image module](https://github.com/lovepark/ifm3d/tree/master/modules/image) and/or [opencv module](https://github.com/lovepark/ifm3d/tree/master/modules/opencv) as a concrete working example of complying to the `ByteBuffer` interface to construct a more productive image container.

Sorry if this is not what you are asking. However, I hope it helps.

Best,
Tom

pfrydlewicz, 2018-10-22:
Thank you for explanation and the hint with SetBytes. That's exactly what I needed.

My use case is writing/reading data to/from file. 
I still wonder if it's better to save the raw data or the image matrices itself. But when going with the raw data I'm on the save side, the image matrices can still be retrieved afterwards ...

Thanks again,
Paul

graugans, 2018-11-13:
@pfrydlewicz can we close this issue? It looks like your questions have been answered.

closed 2018-11-19 by pfrydlewicz