ifm/ifm3d #102: MAKE libifm3d_camera.so.0.10.0: undefined reference to `xmlrpc_c::...

HuDongAlex, 2018-10-23:
## I use the command to install XML:  sudo apt-get install libxmlrpc-c++8-dev

### cmake

[image]

### make

[image]

somebody can help me ? thank you very much !!

graugans, 2018-10-23:
@HuDongAlex from the boost version I do expect that you are running a Ubuntu 18.10 version. Unfortunatly currently we do support Ubuntu 18.04 and 16.04 which are the latest LTR releases. So can you please be so kind and provide your full error log.  By the way GitHub provides a very neat feature [to embed code snippets into issues](https://help.github.com/articles/creating-and-highlighting-code-blocks/). This will make things much easier on my side. Thanks in advance.

HuDongAlex, 2018-10-23:
@graugans thank you for your reply in advance !

###this is my system infos. I just build the project——ifm3d, and I don't changge any code.

[image]

## How can I get the full error log which you want ? thank you !

HuDongAlex, 2018-10-23:
@graugans 

## The output of make.
"
Scanning dependencies of target ifm3d-camera-tests
[ 22%] Building CXX object modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/ifm3d-camera-camera-tests.cpp.o
[ 24%] Building CXX object modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/ifm3d-camera-err-tests.cpp.o
[ 26%] Building CXX object modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/ifm3d-camera-testrunner.cpp.o
[ 28%] Building CXX object modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/ifm3d-camera-version-tests.cpp.o
[ 30%] Linking CXX executable ifm3d-camera-tests
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::carriageParm_curl0::carriageParm_curl0(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_string::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >() const'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::rpcPtr::rpcPtr(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xmlrpc_c::paramList const&)'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_string::cvalue[abi:cxx11]() const'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_string::value_string(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_struct::value_struct(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xmlrpc_c::value, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, xmlrpc_c::value> > > const&)'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_struct::operator std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xmlrpc_c::value, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, xmlrpc_c::value> > >() const'
collect2: error: ld returned 1 exit status
modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/build.make:131: recipe for target 'modules/camera/test/ifm3d-camera-tests' failed
make[2]: *** [modules/camera/test/ifm3d-camera-tests] Error 1
CMakeFiles/Makefile2:473: recipe for target 'modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/all' failed
make[1]: *** [modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
hudong@ubuntu:/disk2/ifm3d-0.11.0/build$ 
"

graugans, 2018-10-23:
Please use `` ``` `` to mark code.

HuDongAlex, 2018-10-23:
😀，I am so sorry, this is my first to use issues.

```
hudong@ubuntu:/disk2/ifm3d-0.11.0/build$ sudo make
Scanning dependencies of target ifm3d_camera_shared
[  2%] Building CXX object modules/camera/src/libifm3d_camera/CMakeFiles/ifm3d_camera_shared.dir/camera.cpp.o
[  4%] Building CXX object modules/camera/src/libifm3d_camera/CMakeFiles/ifm3d_camera_shared.dir/err.cpp.o
[  6%] Building CXX object modules/camera/src/libifm3d_camera/CMakeFiles/ifm3d_camera_shared.dir/logging.cpp.o
[  8%] Building CXX object modules/camera/src/libifm3d_camera/CMakeFiles/ifm3d_camera_shared.dir/util.cpp.o
[ 10%] Building CXX object modules/camera/src/libifm3d_camera/CMakeFiles/ifm3d_camera_shared.dir/version.cpp.o
[ 12%] Linking CXX shared library libifm3d_camera.so
[ 12%] Built target ifm3d_camera_shared
Scanning dependencies of target gtest
[ 14%] Building CXX object gtest_bin/googlemock/gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[ 16%] Linking CXX static library libgtest.a
[ 16%] Built target gtest
Scanning dependencies of target gtest_main
[ 18%] Building CXX object gtest_bin/googlemock/gtest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
[ 20%] Linking CXX static library libgtest_main.a
[ 20%] Built target gtest_main
Scanning dependencies of target ifm3d-camera-tests
[ 22%] Building CXX object modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/ifm3d-camera-camera-tests.cpp.o
[ 24%] Building CXX object modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/ifm3d-camera-err-tests.cpp.o
[ 26%] Building CXX object modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/ifm3d-camera-testrunner.cpp.o
[ 28%] Building CXX object modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/ifm3d-camera-version-tests.cpp.o
[ 30%] Linking CXX executable ifm3d-camera-tests
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::carriageParm_curl0::carriageParm_curl0(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_string::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >() const'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::rpcPtr::rpcPtr(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xmlrpc_c::paramList const&)'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_string::cvalue[abi:cxx11]() const'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_string::value_string(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_struct::value_struct(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xmlrpc_c::value, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, xmlrpc_c::value> > > const&)'
../src/libifm3d_camera/libifm3d_camera.so.0.11.0: undefined reference to `xmlrpc_c::value_struct::operator std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xmlrpc_c::value, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, xmlrpc_c::value> > >() const'
collect2: error: ld returned 1 exit status
modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/build.make:131: recipe for target 'modules/camera/test/ifm3d-camera-tests' failed
make[2]: *** [modules/camera/test/ifm3d-camera-tests] Error 1
CMakeFiles/Makefile2:473: recipe for target 'modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/all' failed
make[1]: *** [modules/camera/test/CMakeFiles/ifm3d-camera-tests.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

## Is this all right ？

graugans, 2018-10-23:
I have just fired up a fresh Ubuntu 18.04 in my [docker](https://github.com/graugans/docker_images/blob/master/ifm3d/ubuntu/bionic/Dockerfile). But it builds fine.

HuDongAlex, 2018-10-23:
@graugans In my pure ubuntu18.04 follow the steps in the docker you describe step by step to the "RUN/home/ifm3d/vim/bundle/YouCompleteMe/the py - clang - completer", but I find that I don't have YouCompleteMe bundle directory folder. I can't do the rest. What should I do? Looking forward to your help!

HuDongAlex, 2018-10-23:
@graugans ok，How can I simply verify that my library and program are installed? I can get the depth image directly from ifm3d, the color one, okay? I really appreciate your help!

graugans, 2018-10-23:
@HuDongAlex after installing ifm3d there is a command line tool called ``ifm3d`` with ``ifm3d dump`` you can check if you can establish a connection to the camera. But maybe you also have to adjust the command line to fit the IP address of the camera. Check [this guide](https://github.com/lovepark/ifm3d/blob/master/doc/troubleshoot.md).

HuDongAlex, 2018-10-23:
@graugans  yes. it worked !! 😀，thank you very much!!

graugans, 2018-11-13:
@HuDongAlex can we close this issue? It seems you have solved the problem.

HuDongAlex, 2018-11-13:
@graugans yes, you can close it.