ifm/ifm3d #103: make check is telling sensor's header is not in the correct format.

pedghz, 2018-10-23:
I was using the same sensor a couple of days ago and I was able to build and use this code with the sensor(O3D303) then. Now I am starting from scratch again but this time I am getting these errors while I am checking the built code with 'make check':

> make check
[ 22%] Built target ifm3d_camera_shared
[ 37%] Built target ifm3d_framegrabber_shared
[ 44%] Built target ifm3d_image_shared
[ 51%] Built target gtest
[ 59%] Built target gtest_main
[ 70%] Built target ifm3d-image-tests
[==========] Running 10 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 10 tests from Image
[ RUN      ] Image.MoveCtor
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.MoveCtor (183 ms)
[ RUN      ] Image.MoveAssignmentOperator
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.MoveAssignmentOperator (196 ms)
[ RUN      ] Image.CopyCtor
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.CopyCtor (200 ms)
[ RUN      ] Image.CopyAssignmentOperator
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.CopyAssignmentOperator (201 ms)
[ RUN      ] Image.References
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.References (199 ms)
[ RUN      ] Image.CloudMechanics
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.CloudMechanics (200 ms)
[ RUN      ] Image.XYZImage
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.XYZImage (200 ms)
[ RUN      ] Image.ComputeCartesian
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.ComputeCartesian (209 ms)
[ RUN      ] Image.TimeStamp
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.TimeStamp (1129 ms)
[ RUN      ] Image.IlluTemp
unknown file: Failure
C++ exception with description "Sensor: Header version mismatch while parsing data" thrown in the test body.
[  FAILED  ] Image.IlluTemp (197 ms)
[----------] 10 tests from Image (2914 ms total)

>[----------] Global test environment tear-down
[==========] 10 tests from 1 test case ran. (2914 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 10 tests, listed below:
[  FAILED  ] Image.MoveCtor
[  FAILED  ] Image.MoveAssignmentOperator
[  FAILED  ] Image.CopyCtor
[  FAILED  ] Image.CopyAssignmentOperator
[  FAILED  ] Image.References
[  FAILED  ] Image.CloudMechanics
[  FAILED  ] Image.XYZImage
[  FAILED  ] Image.ComputeCartesian
[  FAILED  ] Image.TimeStamp
[  FAILED  ] Image.IlluTemp

>10 FAILED TESTS
modules/image/test/CMakeFiles/check_image.dir/build.make:57: recipe for target 'modules/image/test/CMakeFiles/check_image' failed
make[3]: *** [modules/image/test/CMakeFiles/check_image] Error 1
CMakeFiles/Makefile2:741: recipe for target 'modules/image/test/CMakeFiles/check_image.dir/all' failed
make[2]: *** [modules/image/test/CMakeFiles/check_image.dir/all] Error 2
CMakeFiles/Makefile2:76: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:184: recipe for target 'check' failed
make: *** [check] Error 2

graugans, 2018-10-23:
@pedghz what SW version do you have running on the camera?

pedghz, 2018-10-23:
@graugans Camera hardware info:
>Connector: #!02_A300_001_02272689_008001175
Diagnose: #!02_D322_C32_02855660_008023267
Frontend: #!02_F342_C34_17_00001_008023607
Illumination: #!02_I300_001_02839215_008001175
MACAddress: 00:02:01:40:62:54
Mainboard: #!02_M381_003_02839906_008022954
MiraSerial: 0ef0-5909-00b1-0714
Article: O3D303 AD

Camera software info:
>Algorithm_Version: 1.2.19
Calibration_Device: 00:02:01:40:62:54
Calibration_Version: 0.9.0
Diagnostic_Controller: 0.3.207
ELDK: GOLDENEYE_ELDK-2016-07-14_09-36-02-2e9219cce224332ecbe2f36934d2c3c7aa738c78
IFM_Software: 1.6.2114
Linux: Linux version 3.11.0-00002-gd1bbe0b (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Fri Jun 24 16:34:26 CEST 2016
Main_Application: 0.4.222

So the firmware version is 1.6.2114 and the device is O3D303 and te ifm3d version I am using is 0.11.0

pedghz, 2018-10-23:
I just updated the firmware to 1.23.1522 and that problem was solved

commit 01e69fd7 (2018-11-07): Merge branch 'dev-0.12.0_pr105' into dev-0.12.0. Closes #103.
closed 2018-10-23 by pedghz