ifm/ifm3d #108: Questions Regarding Camera Intrinsics

Jmeyer1292, 2018-11-06:
Hello, I apologize for what I hope is a dumb question. I was hoping to use the [intrinsics](https://github.com/lovepark/ifm3d/blob/3d86125b39f0cebc9107de6d42dc0794d9b97f68/modules/framegrabber/include/ifm3d/fg/byte_buffer.h#L309) reported by the IFM driver in combination with the raw depth image to reproduce the point cloud returned by the XYZ image.

I'm having a bit of trouble and I noticed that the intrinsics that I get back look like:
```
intrinsics: fx: 352.657
fy: 352.938
cx: 180.422 // mx in the driver
cy: 127.531 // my in the driver
alpha: -0.000419404
k1: 0.374941
k2: 0.515173
k5: -0.00199539
k3: -9.96921e-06
k4: 0.00105275
```
Given that my images are only `176 x 132`, it seems odd that cx and cy are basically in the corner of the image. Now I may be interpreting these parameters wrong: they are labeled mx and my after all. However, "main point" is difficult to google for and a cursory search revealed no immediate clues.

So I guess I have the following questions:
 * What are the `mx` and `my` parameters? Especially given that it appears to be off the image.
 * Does anyone have an example, either through code or documents, of "reprojecting" the depth image to produce a cloud?
 * (Side note) I have noticed a discrepancy between the IFM tool in Windows (which reports +Z as out of the camera in its visualization) and the driver which appears to report +X as out of the camera. Is this a function of default extrinsics? Where is this translation done?

Thank you,
Jonathan

Jmeyer1292, 2018-11-06:
More googling has shown that the reporting images are binned. Is what is being reported the unbinned camera values? Should all pixel values be divided by two when working with the range images as reported by default?

tpanzarella, 2018-11-06:
> Does anyone have an example, either through code or documents, of "reprojecting" the depth image to produce a cloud?

You can look at [this](https://github.com/lovepark/ifm3d/blob/master/modules/image/test/ifm3d-image-tests.cpp#L307) unit test. This does not contemplate the intrinsics. Just the unit vectors, extrinsics (from chip to camera glass), and the radial distance image.

> (Side note) I have noticed a discrepancy between the IFM tool in Windows (which reports +Z as out of the camera in its visualization) and the driver which appears to report +X as out of the camera. Is this a function of default extrinsics? Where is this translation done?

If using the standard "image" module, it is being done [here](https://github.com/lovepark/ifm3d/blob/master/modules/image/src/libifm3d_image/image_buffer_impl.hpp#L211) I originally wrote this code for usage in ROS, so, I wanted to covert to the standard right-hand coord frame in-line with the pixel parsing. If you want to use some other coord frame you can [write a custom image container](https://github.com/lovepark/ifm3d/blob/master/doc/img_container.md).

Related, the camera can unwarp the image based on the intrinsics for you by setting the various "Rectification" flags. You can do this programmatically or via `ifm3d dump` and `ifm3d config` on the command line.

Jmeyer1292, 2018-11-06:
Thanks @tpanzarella. Can you comment on the `mx()` and `my()` question? Is that the intrinsic parameter set of the "full sensor"?

tpanzarella, 2018-11-06:
I cannot, but I think @graugans could or maybe he could get you in touch with the proper person in Germany who can comment on it. For my usage, me and my team consume the on-camera computed point cloud. If we need to apply the intrinsics, we set it explicitly via the rectification flags. TBH, we rarely need to do that though.

graugans, 2018-11-07:
@Jmeyer1292 as @tpanzarella pointed out I am as well not the expert when it comes to the image data itself. I can give you any information about the communication but have limited knowledge on the data side. I try to translate the information I do have from German to English.

```
------------------8<------------------8<------------------8<------------------8<------------------8<------------------8<
```
our camera model is similar to the common [Bouguet model](http://www.vision.caltech.edu/bouguetj/calib_doc/htmls/parameters.html)

The only difference is that we shift the coordinate system by 0.5 pixels. Another difference is the fact that we specify the direction 3D to 2D as ``inverse intrinsic`` and the direction 2D to 3D as ``intrinisc``. At the moment we do only provide the ``intrinsic`` parameters within the ``ifm3d``

To rectify a image with the ``inverse intrinsic`` parameters you can use the following Matlab Code: 

```matlab
KKNoAlpha = [ par.fx   0     par.mx;
                0    par.fy  par.my;
                0      0        1   ];

% Generate Pixel coordinates

[mx,my] = meshgrid(0:size(matrix,2)-1, 0:size(matrix,1)-1);
coords = [mx(:)+0.5,my(:)+0.5, ones(numel(mx),1)]'; % --> homogeneous  coordinates

 % Transfer in a normalized coordinate system (f=1, origin  is center Dim: -1..1)

coords_norm = KKNoAlpha\coords;

% apply rectification
R2 = coords_norm(1,:).^2+coords_norm(2,:).^2;
R4 = R2.^2;
R6 = R2.^3;

radial_dist = 1+par.k(1)*R2+par.k(2)*R4+par.k(5)*R6;

a1 = 2.*coords_norm(1,:).*coords_norm(2,:);
a2 = R2 + 2*coords_norm(1,:).^2;
a3 = R2 + 2*coords_norm(2,:).^2;

tangential_dist = [par.k(3)*a1 + par.k(4)*a2 ; par.k(3)*a3 + par.k(4)*a1];
coord_dist = [(ones(2,1)*radial_dist).*coords_norm(1:2,:) + tangential_dist; ones(1,numel(mx))];

% transformation --> pixel
KK = KKNoAlpha;
KK(1,2) = par.fx*par.alpha;
coord_dist = KK*coord_dist;
 

% Bilinear Interpolation
cx_d = reshape(coord_dist(1,:)-0.5,size(mx));
cy_d = reshape(coord_dist(2,:)-0.5,size(mx));
rectified = interp2(mx,my,matrix,cx_d,cy_d,'linear');
```

```
------------------8<------------------8<------------------8<------------------8<------------------8<------------------8<
```

Could you please be so kind and give me some background what is your use-case and if you will need the ``inverse intrinsic`` parameters. A example for both ``intrinsic`` and ``inverse intrinsic`` parameters you can find in issue #62 .

theseankelly, 2019-10-09:
Closing old issues. Note that we now also have a python example of the above matlab code here: https://github.com/ifm/ifm3d-examples/blob/master/image_rectification/python/Image_Rectification.ipynb

commit bc58a396 (2023-01-27): Merge branch 'fix_source_install_docs' into 'main'

Fix missing file in install toctree + notes in source build

Closes #108

See merge request syntron/support/csr/ifm3d/ifm3d!207
closed 2019-10-09 by theseankelly