ifm/ifm3d #110: Filtering the pointcloud

Semih67, 2018-11-12:
I have placed the camera 20cm above the ground and when I start rviz the pointcloud is also detecting the floor. I have looked in the configuration and could not find anything to solve the problem. So I have started to write a filter for the pointcloud to delete the points of the z-axis. My question is, is there an other way to do this. Can I configure something?

graugans, 2018-11-12:
@Semih67 there is a parameter called ClippingCuboid within the Imager Settings:

```json
{
"$schema": "http://json-schema.org/draft-04/schema#",
"title": "O3D3xx clipping cuboid configuration",
"type": "object",

"properties": {
"XMin": {
"type": "number"
},
"XMax": {
"type": "number"
},
"YMin": {
"type": "number"
},
"YMax": {
"type": "number"
},
"ZMin": {
"type": "number"
},
"ZMax": {
"type": "number"
}
},

"required": ["XMin", "XMax", "YMin", "YMax", "ZMin", "ZMax"]
}
```
The unit of all components of the clipping cuboid is [m].
The values 
```json
{
  "ClippingBottom": "131",
  "ClippingLeft": "0",
  "ClippingRight": "175",
  "ClippingTop": "0",
}
```
Are read-only parameters and present the calculated values.

graugans, 2018-11-12:
@Semih67 Please keep in mind when playing with those parameters and using RVIZ that the coordinate Frame of the camera and ROS is different. https://github.com/lovepark/ifm3d/issues/108#issuecomment-436297781 sheds some light on this.

Another side-note, the mounting angle of the camera has several advantages or disadvantages depending on your application. I am not an ROS expert, so I do not know if there are any filters within ROS which will allow you to extract the floor somehow and a transform the point cloud. Your robots pitch during acceleration and de acceleration might also have to be taken into account.

We do provide the point cloud the application is in your domain, without more details neither me nor @tpanzarella can assist you. So I recommend to get in touch with @GarrettPlace to discuss the application.

tpanzarella, 2018-11-12:
In follow-up to @graugans comment, simply clipping the point cloud in the camera frame will likely be a very fragile approach to estimate (and eliminate) the ground plane. The way this is typically done is to calibrate the camera so that you can transform the point cloud to the vehicle frame then estimate the ground plane WRT the vehicle (there are several ways to do this). Once you have segmented the ground from the rest of the scene you can filter out the points as you would like. To @graugans point above, this is application-level code and not something we provide in the driver/sensor-interface.

Semih67, 2018-11-13:
` ~ % ifm3d dump | jq ".ifm3d.Apps[0].Imager"
{
  "ExposureTime": "1350",
  "ExposureTimeRatio": "10",
  "FrameRate": "4",
  "MaxAllowedFrameRate": "4.16666666666667",
  "MinimumAmplitude": "60",
  "SpatialFilter": {
    "MaskSize": "0"
  },
  "SpatialFilterType": "1",
  "SymmetryThreshold": "0.4",
  "TemporalFilter": {
    "": ""
  },
  "TemporalFilterType": "0",
  "Type": "upTo30m_moderate"
} `

I can only modify these parameters. I don't have a parameter called ClippingCuboid. The article number of the camera is O3X100.

Semih67, 2018-11-13:
Ok thank you for your help. But I have one more question: How big is the field of view (H x V) of the camera? Do you have a table with all of your articles? And if I don't have oppotunity to change the clipping on the O3X, on which article of yours could I change it?

graugans, 2018-11-13:
@Semih67 I am not aware of such a table, maybe @GarrettPlace is. You can take a look at our official website. There are O3D3xx cameras available with [60x40](https://www.ifm.com/products/ae/ds/O3D301.htm) and [40x30](https://www.ifm.com/products/ae/ds/O3D301.htm)

And as stated above the O3D3xx line of cameras do support the full feature set with a wider range. There are also less restrictions on the framerate.

graugans, 2018-11-16:
@Semih67 do you need more information or can we close this issue?

closed 2018-11-16 by Semih67