ifm/ifm3d #111: O3X : Memory leak and CPU increase...

Arag24, 2018-11-12:
Hi everyone,

From a user experience (@jChaudet) it looks like the following line in `bytebuffer.hpp` (ifm3d 0.11.0 revision / Camera in Trig Mode):

> `if( EXT_OK) else { LOG(WARNING) << "illu temp and exposure times skipped (can't trust extidx)"; } `

is leading to some nasty symptoms, after running ifm3d for a while (more than 15hours)
 1) Memory leakage.
 2) Increase of CPU usage.
 3) Some random missed frame (missed frame increase with running time)

Please see experience using following code:

```
export IFM3D_MASK=3
ifm3d --ip=192.168.1.65 jitter --nframes=10000000

```

**- Time T=0:**
[image]

**- Time T > 15hours:**
[image]

We clearly see increase of Memory & CPU use with htop.

Removing > `if( EXT_OK) else { LOG(WARNING) << "illu temp and exposure times skipped (can't trust extidx)"; } ` is getting O3X really much stable for CPU / Memory / and No miss.
Test was made 

It looks like EXT_OK, is every time wrong with O3X and is leading to this behavior. Is it normal ? Can it be fixed ?

Sincerely Yours,

tpanzarella, 2018-11-12:
The [dev-0.11.2 branch](https://github.com/ifm/ifm3d/tree/dev-0.11.2) has a "fix" implemented. It would be good if it could be tested by you. The approach was to emit the log message at the VLOG severity of `IFM3D_PROTO_DEBUG`. By default, that is off. To turn it on, the `IFM3D_VLOG` environment variable can be set to an integer `>=15`. I chose to use the `VLOG` approach over the `DLOG` approach so that debugging could be done without requiring a recompile.

You should also note that when upgrading to `0.11.1` or better, bad pixels (as inferred from the confidence image) are flagged as `nan` in the image containers rather than simply passed through. So, if your algorithms to do not support it, you need to be ready to see some nan values particularly with the O3X. If using OpenCV (either via our `image` or `opencv` containers) you can see an exemplary nan filter in the unit tests, for example, [here](https://github.com/ifm/ifm3d/blob/dev-0.11.2/modules/image/test/ifm3d-image-tests.cpp#L370). This simply filters the nan values by setting them all to `0` (you may wish to apply a different policy).

graugans, 2018-11-13:
@Arag24 and @jChaudet thank you very much for sharing this information. Related to the memory leak I do have suspicion. Depending on your system this might be caused due to the fact that for O3X the log is spammed with:

```
illu temp and exposure times skipped (can't trust extidx)
```

The default location for ifm3d is to write those log files to the ``/tmp`` folder. If you are running this kind of stuff on an embedded device, which I am pretty sure you do. The ``/tmp`` folder is most likely  mounted as ``tmpfs`` filesystem which means putting any data into this folder will lower your free memory. For example on the camera itself we mounted ``/tmp`` as tmpfs.

```
root@O3D303-40-BB-30:~# readlink /tmp -f
/var/volatile/tmp
root@O3D303-40-BB-30:~# mount | grep -e "/volatile"
tmpfs on /var/volatile type tmpfs (rw,relatime)
```

If I do interpret the output of the ``htop`` images correctly the memory consumption of ifm3d is for ``T0``  4.7% and for ``T > 15 hours`` it is also 4.7% . I do agree we have to eliminate the log SPAM but the ifm3d itself does not leak memory. Unfortunately it seems the logging framework ([glog](https://github.com/google/glog/issues/36)) we are using does not support log rotation.

jChaudet, 2018-11-13:
@graugans, you are right, this is what we have. I came to the same conclusion by looking at the type of CPU usage (red for kernel). This is not related to the camera itself.

I delete last friday the log message in the ifm3d library and I have no more problem since.

@tpanzarella ,
I will test this as soon as possible (during the week).

tpanzarella, 2018-11-13:
@jChaudet Realize there are a few other issues in 0.11.2 that I need to address. Most of it is related to how we are marking bad pixels in the image containers. In its current state, you may have some unit test failures. Once I get in the office (~4 hours from now) I will be looking into to it. None-the-less the log spam should be resolved. I appreciate your willingness to test.

graugans, 2018-11-16:
@Arag24, @jChaudet  after releasing [v.0.11.2](https://github.com/ifm/ifm3d/releases/tag/v0.11.2) this issue can be closed isn't it?

jChaudet, 2018-11-23:
@graugans, @Arag24, 
For the moment, I can not test the v0.11.2. 
I should have news in January.
Sorry for the delay. 
I will contact directly Arag24.

graugans, 2018-11-23:
No Problem @jChaudet take your time.

jChaudet, 2019-01-18:
Hi,

jChaudet, 2019-01-18:
@graugans, @Arag24, @tpanzarella

Hi,
I tested 0.11.2 version and this is functional. There is no longer the  
`illu temp and exposure times skipped (can't trust extidx)` log.
Thanks for that.

But logs are always intrusive for other reasons, for example, a disconnection. A log appears as long as the sensor is not connected.

```
E0713 01:53:21.774899  3022 camera_impl.hpp:259] http://192.168.1.32:80/api/rpc/v1/com.ifm.efector/->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP$
E0713 01:53:21.777318  3022 frame_grabber_impl.hpp:171] Could not get PCIC Port of the camera: Lib: XMLRPC Timeout - can you `ping' the sensor?
E0713 01:53:24.897348  3022 camera_impl.hpp:259] http://192.168.1.32:80/api/rpc/v1/com.ifm.efector/->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP$
E0713 01:53:24.899027  3022 frame_grabber_impl.hpp:171] Could not get PCIC Port of the camera: Lib: XMLRPC Timeout - can you `ping' the sensor?
E0713 01:53:28.014946  3022 camera_impl.hpp:259] http://192.168.1.32:80/api/rpc/v1/com.ifm.efector/->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP$
E0713 01:53:28.018008  3022 frame_grabber_impl.hpp:171] Could not get PCIC Port of the camera: Lib: XMLRPC Timeout - can you `ping' the sensor?
E0713 01:53:31.134858  3022 camera_impl.hpp:259] http://192.168.1.32:80/api/rpc/v1/com.ifm.efector/->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP$
E0713 01:53:31.137365  3022 frame_grabber_impl.hpp:171] Could not get PCIC Port of the camera: Lib: XMLRPC Timeout - can you `ping' the sensor?
E0713 01:53:34.254827  3022 camera_impl.hpp:259] http://192.168.1.32:80/api/rpc/v1/com.ifm.efector/->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP$
E0713 01:53:34.257052  3022 frame_grabber_impl.hpp:171] Could not get PCIC Port of the camera: Lib: XMLRPC Timeout - can you `ping' the sensor?
E0713 01:53:37.377854  3022 camera_impl.hpp:259] http://192.168.1.32:80/api/rpc/v1/com.ifm.efector/->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP$
E0713 01:53:37.379287  3022 frame_grabber_impl.hpp:171] Could not get PCIC Port of the camera: Lib: XMLRPC Timeout - can you `ping' the sensor?
```
In general, this seems fair but for our application, we want to be robust to disconnection. 
Is there a solution to not activate logs by software and so avoid an increase in memory ?

tpanzarella, 2019-01-18:
@jChaudet `ifm3d` employs Google Glog for logging. So, for your particular question here, if you set the environment variable `GLOG_minloglevel=3` for your process, it will silence this message. The side-effect, of course, is that other important messages logged at the `ERROR` (and below) severity levels will also be missed. We will contemplate a better solution for the masses on this topic, but for now, if this is a pressing issue for you, you can do as described above. Perhaps we will employ a `LOG_EVERY_N(ERROR, ...)` macro for this particular situation. Would that be suitable for you? [This page](http://rpg.ifi.uzh.ch/docs/glog.html) explains the options if you are not familiar with `glog`.

jChaudet, 2019-01-18:
@tpanzarella,

Modify an environment variable obliges me to compile again all my distribution if I wish to recover or not the log of the camera.
I think it will be better to limit / define the size of log files. Use a `LOG_EVERY_N(ERROR, ...) macro ` is a good complement.
In my case, I want to be sure that the memory will be not filled by the log files.

Arag24, 2019-05-31:
Closed until we have something new on this(@jChaudet and/or @Arag24 )

closed 2019-05-31 by Arag24